RIOT-OS/RIOT #12200: Arduino MRK WiFi 1010 support

paulrox, 2019-09-11:
Hi all,

I saw that RIOT is already compatible with many Arduino boards, but the MKR WiFi 1010 is missing.
I tried to build and flash a simple "Hello World" following the steps in the quick start guide and using the MKRZERO as board in the Makefile (they are both from the MKR family and they share the same MCU: Atmel SAMD21G18A),

I'm able to successfully build the application in the RIOT VM but I'm stuck on the flash phase.
The `make flash` command gives me the following output:
```
"make" -C /home/vagrant/RIOT/boards/arduino-mkrzero
"make" -C /home/vagrant/RIOT/boards/common/arduino-mkr
"make" -C /home/vagrant/RIOT/core
"make" -C /home/vagrant/RIOT/cpu/samd21
"make" -C /home/vagrant/RIOT/cpu/cortexm_common
"make" -C /home/vagrant/RIOT/cpu/cortexm_common/periph
"make" -C /home/vagrant/RIOT/cpu/sam0_common
"make" -C /home/vagrant/RIOT/cpu/sam0_common/periph
"make" -C /home/vagrant/RIOT/cpu/samd21/periph
"make" -C /home/vagrant/RIOT/drivers
"make" -C /home/vagrant/RIOT/drivers/periph_common
"make" -C /home/vagrant/RIOT/sys
"make" -C /home/vagrant/RIOT/sys/auto_init
"make" -C /home/vagrant/RIOT/sys/newlib_syscalls_default
"make" -C /home/vagrant/RIOT/sys/pm_layered
"make" -C /home/vagrant/RIOT/sys/stdio_uart
   text    data     bss     dec     hex filename
   7080     120    2540    9740    260c /home/vagrant/RIOT/arduino_riot/bin/arduino-mkrzero/foobar.elf
/home/vagrant/RIOT/dist/tools/bossa/bossac -p /dev/ttyACM0 -e -i -w -v -b -R /home/vagrant/RIOT/arduino_riot/bin/arduino-mkrzero/foobar.bin
No device found on /dev/ttyACM0
/home/vagrant/RIOT/arduino_riot/../Makefile.include:554: recipe for target 'flash' failed
make: *** [flash] Error 1
```

I don't think it is a USB problem with the VM because if I use the `make term` command I can see the output of a program which I previously uploaded using the Arduino IDE. I also followed [those instructions](https://github.com/RIOT-OS/RIOT/tree/master/dist/tools/vagrant) for the VM configuration.

Is it a compatibility issue with the board or am I doing something wrong??

### Useful links
https://store.arduino.cc/mkr-wifi-1010

aabadie, 2019-09-11:
Thanks for working on porting a new MKR boards @paulrox !

The problem you have is because the board doesn't switch automatically in bootloader mode. You have to do this manually by double pressing the reset button before reflashing (every time...).

After that, you won't have the STDIO output directly via USB, you'll have to plug an USB to UART converter on the RX/TX pins of the board.

This is not very convenient but with the newly merged USB stack (and CDC ACM profile) it will be possible to end up with something as user-friendly as Arduino is providing for these boards: automatic switch in bootloader mode when flashing + STDIO over USB.

paulrox, 2019-09-11:
Thank you for the fast reply!

I've already tried before to reboot in bootloader mode but I obtained the same result.
After some trials, I made it work by adding the Arduino USB device (when in bootloader mode) to the USB filter in virtual box and by adding the udev rules for it.

I also had to reboot in bootloader mode a couple of times to let the board detected by bossac.

Now I will test if the application is actually working as expected.

Another question, is there any limitation to the RIOT modules compatibility with this particular board ? (or with the MKRZERO in general?) Because I would like to use the RIOT CoAP module with this board.

aabadie, 2019-09-11:
> is there any limitation to the RIOT modules compatibility with this particular board ? (or with the MKRZERO in general?) Because I would like to use the RIOT CoAP module with this board.

The SAMD21 MCU is well supported in RIOT. If you want to use CoAP on this board via the WiFi radio, you will have to write the driver for it with an adaption to netdev.

closed 2020-04-24 by stale[bot]