RIOT-OS/RIOT #12205: toolchain: Unable to compile starting with gcc 9.1.0

miri64, 2019-09-11:
#### Description

When trying to compile for `native` with a rather current Arch Linux with `gcc`, the compilation failed when not set to `WERROR=0`

#### Steps to reproduce the issue
On an Arch Linux (or any system that has `gcc` 9.1 installed) do for `BOARD=native` (or any ARM board if you use Arch's arm-none-eabi toolchain)

```sh
make -C examples/hello-world
```

#### Expected results
Application builds

#### Actual results
<details><summary>Lots of warnings in <code>core/atomic_c11.c</code> that cause errors due to our compiler configuration:</summary>

(sorry, my system is configured in German ATM, it's mostly complaining about some functions expectining `[const ]volatile void *`)

```
make: Verzeichnis „/home/mlenders/Repositories/RIOT-OS/RIOT/examples/hello-world“ wird betreten
Building application "hello-world" for "native" with MCU "native".

"make" -C /home/mlenders/Repositories/RIOT-OS/RIOT/boards/native
"make" -C /home/mlenders/Repositories/RIOT-OS/RIOT/boards/native/drivers
"make" -C /home/mlenders/Repositories/RIOT-OS/RIOT/core
/home/mlenders/Repositories/RIOT-OS/RIOT/core/atomic_c11.c:57:10: Fehler: Unstimmigkeit im Typ von Argument 1 der eingebauten Funktion »__atomic_load_1«; erwartet wurde »const volatile void *« [-Werror=builtin-declaration-mismatch]
   57 |     I##n __atomic_load_##n (I##n *ptr, int memorder) \
      |          ^~~~~~~~~~~~~~
/home/mlenders/Repositories/RIOT-OS/RIOT/core/atomic_c11.c:160:1: Anmerkung: bei Substitution des Makros »TEMPLATE_ATOMIC_LOAD_N«
  160 | TEMPLATE_ATOMIC_LOAD_N(1)                 /* __atomic_load_1 */
      | ^~~~~~~~~~~~~~~~~~~~~~
/home/mlenders/Repositories/RIOT-OS/RIOT/core/atomic_c11.c:57:10: Fehler: Unstimmigkeit im Typ von Argument 1 der eingebauten Funktion »__atomic_load_2«; erwartet wurde »const volatile void *« [-Werror=builtin-declaration-mismatch]
   57 |     I##n __atomic_load_##n (I##n *ptr, int memorder) \
      |          ^~~~~~~~~~~~~~
/home/mlenders/Repositories/RIOT-OS/RIOT/core/atomic_c11.c:161:1: Anmerkung: bei Substitution des Makros »TEMPLATE_ATOMIC_LOAD_N«
  161 | TEMPLATE_ATOMIC_LOAD_N(2)                 /* __atomic_load_2 */
      | ^~~~~~~~~~~~~~~~~~~~~~
/home/mlenders/Repositories/RIOT-OS/RIOT/core/atomic_c11.c:57:10: Fehler: Unstimmigkeit im Typ von Argument 1 der eingebauten Funktion »__atomic_load_4«; erwartet wurde »const volatile void *« [-Werror=builtin-declaration-mismatch]
   57 |     I##n __atomic_load_##n (I##n *ptr, int memorder) \
      |          ^~~~~~~~~~~~~~
/home/mlenders/Repositories/RIOT-OS/RIOT/core/atomic_c11.c:162:1: Anmerkung: bei Substitution des Makros »TEMPLATE_ATOMIC_LOAD_N«
  162 | TEMPLATE_ATOMIC_LOAD_N(4)                 /* __atomic_load_4 */
      | ^~~~~~~~~~~~~~~~~~~~~~
/home/mlenders/Repositories/RIOT-OS/RIOT/core/atomic_c11.c:57:10: Fehler: Unstimmigkeit im Typ von Argument 1 der eingebauten Funktion »__atomic_load_8«; erwartet wurde »const volatile void *« [-Werror=builtin-declaration-mismatch]
   57 |     I##n __atomic_load_##n (I##n *ptr, int memorder) \
      |          ^~~~~~~~~~~~~~
/home/mlenders/Repositories/RIOT-OS/RIOT/core/atomic_c11.c:163:1: Anmerkung: bei Substitution des Makros »TEMPLATE_ATOMIC_LOAD_N«
  163 | TEMPLATE_ATOMIC_LOAD_N(8)                 /* __atomic_load_8 */
      | ^~~~~~~~~~~~~~~~~~~~~~
/home/mlenders/Repositories/RIOT-OS/RIOT/core/atomic_c11.c:72:10: Fehler: Unstimmigkeit im Typ von Argument 1 der eingebauten Funktion »__atomic_store_1«; erwartet wurde »volatile void *« [-Werror=builtin-declaration-mismatch]
   72 |     void __atomic_store_##n (I##n *ptr, I##n val, int memorder) \
      |          ^~~~~~~~~~~~~~~
/home/mlenders/Repositories/RIOT-OS/RIOT/core/atomic_c11.c:165:1: Anmerkung: bei Substitution des Makros »TEMPLATE_ATOMIC_STORE_N«
  165 | TEMPLATE_ATOMIC_STORE_N(1)                /* __atomic_store_1 */
      | ^~~~~~~~~~~~~~~~~~~~~~~
/home/mlenders/Repositories/RIOT-OS/RIOT/core/atomic_c11.c:72:10: Fehler: Unstimmigkeit im Typ von Argument 1 der eingebauten Funktion »__atomic_store_2«; erwartet wurde »volatile void *« [-Werror=builtin-declaration-mismatch]
   72 |     void __atomic_store_##n (I##n *ptr, I##n val, int memorder) \
      |          ^~~~~~~~~~~~~~~
/home/mlenders/Repositories/RIOT-OS/RIOT/core/atomic_c11.c:166:1: Anmerkung: bei Substitution des Makros »TEMPLATE_ATOMIC_STORE_N«
  166 | TEMPLATE_ATOMIC_STORE_N(2)                /* __atomic_store_2 */
      | ^~~~~~~~~~~~~~~~~~~~~~~
/home/mlenders/Repositories/RIOT-OS/RIOT/core/atomic_c11.c:72:10: Fehler: Unstimmigkeit im Typ von Argument 1 der eingebauten Funktion »__atomic_store_4«; erwartet wurde »volatile void *« [-Werror=builtin-declaration-mismatch]
   72 |     void __atomic_store_##n (I##n *ptr, I##n val, int memorder) \
      |          ^~~~~~~~~~~~~~~
/home/mlenders/Repositories/RIOT-OS/RIOT/core/atomic_c11.c:167:1: Anmerkung: bei Substitution des Makros »TEMPLATE_ATOMIC_STORE_N«
  167 | TEMPLATE_ATOMIC_STORE_N(4)                /* __atomic_store_4 */
      | ^~~~~~~~~~~~~~~~~~~~~~~
/home/mlenders/Repositories/RIOT-OS/RIOT/core/atomic_c11.c:72:10: Fehler: Unstimmigkeit im Typ von Argument 1 der eingebauten Funktion »__atomic_store_8«; erwartet wurde »volatile void *« [-Werror=builtin-declaration-mismatch]
   72 |     void __atomic_store_##n (I##n *ptr, I##n val, int memorder) \
[... 366 lines cut ...]
                  clang: clang version 8.0.1 (tags/RELEASE_801/final)

Installed compiler libs
-----------------------
   arm-none-eabi-newlib: "3.1.0"
    mips-mti-elf-newlib: missing
riscv-none-embed-newlib: missing
xtensa-esp32-elf-newlib: missing
xtensa-lx106-elf-newlib: missing
               avr-libc: "2.0.0" ("20150208")

Installed development tools
---------------------------
                  cmake: cmake version 3.15.3
               cppcheck: missing
                doxygen: 1.8.16
                 flake8: 3.6.0 (mccabe: 0.6.1, pycodestyle: 2.4.0, pyflakes: 2.0.0) CPython 3.7.4 on Linux
                    git: git version 2.23.0
                   make: GNU Make 4.2.1
                openocd: Open On-Chip Debugger 0.10.0+dev-00746-g8f518d35 (2019-03-21-09:43)
                 python: Python 3.7.4
                python2: Python 2.7.16
                python3: Python 3.7.4
             coccinelle: missing
```

aabadie, 2020-02-19:
Is this issue fixed ? I can't reproduce it locally (gcc 9.2.1):

<details>

```
make -C examples/hello-world
make: Entering directory '/work/riot/RIOT/examples/hello-world'
Building application "hello-world" for "native" with MCU "native".

"make" -C /work/riot/RIOT/boards/native
"make" -C /work/riot/RIOT/boards/native/drivers
"make" -C /work/riot/RIOT/core
"make" -C /work/riot/RIOT/cpu/native
"make" -C /work/riot/RIOT/cpu/native/periph
"make" -C /work/riot/RIOT/cpu/native/stdio_native
"make" -C /work/riot/RIOT/cpu/native/vfs
"make" -C /work/riot/RIOT/drivers
"make" -C /work/riot/RIOT/drivers/periph_common
"make" -C /work/riot/RIOT/sys
"make" -C /work/riot/RIOT/sys/auto_init
   text	   data	    bss	    dec	    hex	filename
  24816	    568	  47728	  73112	  11d98	/work/riot/RIOT/examples/hello-world/bin/native/hello-world.elf
make: Leaving directory '/work/riot/RIOT/examples/hello-world'
```

</details>

<details><summary>Versions</summary>

```
make print-versions 

Operating System Environment
----------------------------
         Operating System: "Ubuntu" "19.10 (Eoan Ermine)"
                   Kernel: Linux 5.3.0-40-generic x86_64 x86_64
             System shell: /usr/bin/dash (probably dash)
             make's shell: /usr/bin/dash (probably dash)

Installed compiler toolchains
-----------------------------
               native gcc: gcc (Ubuntu 9.2.1-9ubuntu2) 9.2.1 20191008
        arm-none-eabi-gcc: arm-none-eabi-gcc (GNU Tools for Arm Embedded Processors 8-2019-q3-update) 8.3.1 20190703 (release) [gcc-8-branch revision 273027]
                  avr-gcc: avr-gcc (AVR_8_bit_GNU_Toolchain_3.6.2_1759) 5.4.0
         mips-mti-elf-gcc: mips-mti-elf-gcc (Codescape GNU Tools 2016.05-03 for MIPS MTI Bare Metal) 4.9.2
               msp430-gcc: missing
     riscv-none-embed-gcc: riscv-none-embed-gcc (GNU MCU Eclipse RISC-V Embedded GCC, 64-bit) 8.2.0
     xtensa-esp32-elf-gcc: missing
   xtensa-esp8266-elf-gcc: missing
                    clang: clang version 9.0.0-2 (tags/RELEASE_900/final)

Installed compiler libs
-----------------------
     arm-none-eabi-newlib: "3.1.0"
      mips-mti-elf-newlib: "2.1.0"
  riscv-none-embed-newlib: "3.0.0"
  xtensa-esp32-elf-newlib: missing
xtensa-esp8266-elf-newlib: missing
                 avr-libc: "2.0.0" ("20150208")

Installed development tools
---------------------------
                   ccache: ccache version 3.7.3
                    cmake: cmake version 3.13.4
                 cppcheck: Cppcheck 1.88
                  doxygen: 1.8.13
                      git: git version 2.20.1
                     make: GNU Make 4.2.1
                  openocd: Open On-Chip Debugger 0.10.0+dev-01047-g09ac9ab1 (2020-02-05-08:53)
                   python: Python 2.7.17
                  python2: Python 2.7.17
                  python3: Python 3.7.5
                   flake8: 3.7.7 (mccabe: 0.6.1, pycodestyle: 2.5.0, pyflakes: 2.1.1) CPython 3.7.5 on Linux
               coccinelle: spatch version 1.0.4 with Python support and with PCRE support
```

</details>

aabadie, 2020-07-02:
ping @miri64 (I still can't reproduce with gcc 9.3.0).

miri64, 2020-07-02:
I think this was fixed somehow somewhere. I did not have this issue for a while now, so I think we can close :-).

closed 2020-07-02 by miri64