RIOT-OS/RIOT #12219: genconfigheader doesn't like defines with spaces

larseggert, 2019-09-13:
#### Description
genconfigheader.sh chokes on defines of strings with spaces

#### Steps to reproduce the issue
`env BUILD_IN_DOCKER=1 DOCKER_ENVIRONMENT_CMDLINE=$'-e USEMODULE=esp_wifi -e CFLAGS=-DESP_WIFI_SSID="has space"' make BOARD=esp32-wroom-32 -C tests/lwip`

#### Expected results
Expect the compile to not fail.

#### Actual results
```
make: Entering directory '/Users/lars/Documents/Code/riot/tests/lwip'
Launching build container using image "riot/riotbuild:latest".
mkdir -p /Users/lars/Documents/Code/riot/build
docker run --rm -t -u "$(id -u)" \
    -v '/private/var/db/timezone/tz/2019b.1.0/zoneinfo/Europe/Helsinki:/etc/localtime:ro' -v '/Users/lars/Documents/Code/riot:/data/riotbuild/riotbase:delegated' -e 'RIOTBASE=/data/riotbuild/riotbase' -e 'CCACHE_BASEDIR=/data/riotbuild/riotbase' -e 'BUILD_DIR=/data/riotbuild/riotbase/build' -e 'RIOTPROJECT=/data/riotbuild/riotbase' -e 'RIOTCPU=/data/riotbuild/riotbase/cpu' -e 'RIOTBOARD=/data/riotbuild/riotbase/boards' -e 'RIOTMAKE=/data/riotbuild/riotbase/makefiles'     \
    -e USEMODULE=esp_wifi -e CFLAGS=-DESP_WIFI_SSID="has space" -e 'BOARD=esp32-wroom-32' \
    -w '/data/riotbuild/riotbase/tests/lwip/' \
    'riot/riotbuild:latest' make  'BOARD=esp32-wroom-32'
ESP32_SDK_DIR should be defined as /path/to/esp-idf directory
ESP32_SDK_DIR is set by default to /opt/esp/esp-idf
Building application "tests_lwip" for "esp32-wroom-32" with MCU "esp32".

'/data/riotbuild/riotbase/dist/tools/genconfigheader/genconfigheader.sh' -DESP_WIFI_SSID=has space -DDEVELHELP -Werror -DSCHED_PRIO_LEVELS=32 -DSDK_NOT_USED -DCONFIG_FREERTOS_UNICORE=1 -DESP_PLATFORM -DLOG_TAG_IN_BRACKETS -Wno-unused-parameter -Wformat=0 -mlongcalls -mtext-section-literals -fstrict-volatile-bitfields -fdata-sections -ffunction-sections -fzero-initialized-in-bss -DSCHED_PRIO_LEVELS=32 -Os -DFLASH_MODE_DOUT=1 -DFLASH_MODE_DOUT=1 -DFLASH_MODE_DOUT=1 -DFLASH_MODE_DOUT=1 -DFLASH_MODE_DOUT=1 -DFLASH_MODE_DOUT=1 -DRIOT_APPLICATION=\"tests_lwip\" -DBOARD_ESP32_WROOM_32=\"esp32-wroom-32\" -DRIOT_BOARD=BOARD_ESP32_WROOM_32 -DCPU_ESP32=\"esp32\" -DRIOT_CPU=CPU_ESP32 -DMCU_ESP32=\"esp32\" -DRIOT_MCU=MCU_ESP32 -std=c99 -fno-common -Wall -Wextra -Wmissing-include-dirs -fno-delete-null-pointer-checks -fdiagnostics-color -Wstrict-prototypes -Wold-style-definition -Wformat=2 -DSOCK_HAS_IPV6 -DMODULE_AUTO_INIT -DMODULE_BOARD -DMODULE_BOARDS_COMMON_ESP32 -DMODULE_CORE -DMODULE_CORE_MBOX -DMODULE_CORE_MSG -DMODULE_CPU -DMODULE_DIV -DMODULE_ESP_IDF -DMODULE_ESP_IDF_DRIVER -DMODULE_ESP_IDF_ESP32 -DMODULE_ESP_IDF_HEAP -DMODULE_ESP_IDF_NVS_FLASH -DMODULE_ESP_IDF_SOC -DMODULE_ESP_IDF_SPI_FLASH -DMODULE_ESP_IDF_WPA_SUPPLICANT_CRYPTO -DMODULE_ESP_IDF_WPA_SUPPLICANT_PORT -DMODULE_ESP_WIFI -DMODULE_ESP_WIFI_ANY -DMODULE_FMT -DMODULE_GNRC -DMODULE_GNRC_NETAPI -DMODULE_GNRC_NETIF -DMODULE_GNRC_NETIF_ETHERNET -DMODULE_GNRC_NETIF_HDR -DMODULE_GNRC_NETREG -DMODULE_GNRC_PKT -DMODULE_GNRC_PKTBUF -DMODULE_GNRC_PKTBUF_STATIC -DMODULE_IPV6_ADDR -DMODULE_ISRPIPE -DMODULE_L2UTIL -DMODULE_LOG -DMODULE_LUID -DMODULE_LWIP -DMODULE_LWIP_API -DMODULE_LWIP_CONTRIB -DMODULE_LWIP_CORE -DMODULE_LWIP_ETHERNET -DMODULE_LWIP_IPV6 -DMODULE_LWIP_IPV6_AUTOCONFIG -DMODULE_LWIP_NETDEV -DMODULE_LWIP_NETIF -DMODULE_LWIP_RAW -DMODULE_LWIP_SOCK -DMODULE_LWIP_SOCK_IP -DMODULE_LWIP_SOCK_TCP -DMODULE_LWIP_SOCK_UDP -DMODULE_LWIP_TCP -DMODULE_LWIP_UDP -DMODULE_MTD -DMODULE_NETDEV_DEFAULT -DMODULE_NETDEV_ETH -DMODULE_NETIF -DMODULE_NETOPT -DMODULE_NEWLIB -DMODULE_NEWLIB_SYSCALLS_DEFAULT -DMODULE_OD -DMODULE_PERIPH -DMODULE_PERIPH_ADC_CTRL -DMODULE_PERIPH_COMMON -DMODULE_PERIPH_CPUID -DMODULE_PERIPH_FLASH -DMODULE_PERIPH_GPIO -DMODULE_PERIPH_HWRNG -DMODULE_PERIPH_PM -DMODULE_PERIPH_RTC -DMODULE_PERIPH_TIMER -DMODULE_PERIPH_UART -DMODULE_PRNG -DMODULE_PRNG_TINYMT32 -DMODULE_PS -DMODULE_PTHREAD -DMODULE_RANDOM -DMODULE_RIOT_FREERTOS -DMODULE_SEMA -DMODULE_SHELL -DMODULE_SHELL_COMMANDS -DMODULE_SOCK_IP -DMODULE_SOCK_TCP -DMODULE_SOCK_UDP -DMODULE_STDIN -DMODULE_STDIO_UART -DMODULE_STDIO_UART_RX -DMODULE_SYS -DMODULE_TIMEX -DMODULE_TINYMT32 -DMODULE_TSRB -DMODULE_XTENSA -DMODULE_XTIMER -DRIOT_VERSION=\"2019.10-devel-832-g29a17-docker-delegate-volumes\" \
	| '/data/riotbuild/riotbase/dist/tools/lazysponge/lazysponge.py' --verbose '/data/riotbuild/riotbase/tests/lwip/bin/esp32-wroom-32/riotbuild/riotbuild.h'
Keeping old /data/riotbuild/riotbase/tests/lwip/bin/esp32-wroom-32/riotbuild/riotbuild.h (e1ebc8949498830281018da2a3a0b4c3)
make[1]: Entering directory '/data/riotbuild/riotbase/pkg/lwip'
if [ 159e31b689577dbf69cf0683bbaffbd71fa5ee10 != 159e31b689577dbf69cf0683bbaffbd71fa5ee10 ] ; then \
	git -C /data/riotbuild/riotbase/tests/lwip/bin/pkg/esp32-wroom-32/lwip clean -xdff ; \
	git -C /data/riotbuild/riotbase/tests/lwip/bin/pkg/esp32-wroom-32/lwip fetch "https://git.savannah.nongnu.org/git/lwip.git" "159e31b689577dbf69cf0683bbaffbd71fa5ee10" ; \
	git -C /data/riotbuild/riotbase/tests/lwip/bin/pkg/esp32-wroom-32/lwip checkout -f 159e31b689577dbf69cf0683bbaffbd71fa5ee10 ; \
	touch /data/riotbuild/riotbase/tests/lwip/bin/pkg/esp32-wroom-32/lwip/.git-downloaded ; \
fi
make[1]: Leaving directory '/data/riotbuild/riotbase/pkg/lwip'
"make" -C /data/riotbuild/riotbase/pkg/lwip
make[1]: Entering directory '/data/riotbuild/riotbase/pkg/lwip'
"make" -f /data/riotbuild/riotbase/Makefile.base MODULE=lwip_api -C /data/riotbuild/riotbase/tests/lwip/bin/pkg/esp32-wroom-32/lwip/src/api
make[2]: Entering directory '/data/riotbuild/riotbase/tests/lwip/bin/pkg/esp32-wroom-32/lwip/src/api'
xtensa-esp32-elf-gcc \
	-DRIOT_FILE_RELATIVE=\"tests/lwip/bin/pkg/esp32-wroom-32/lwip/src/api/netbuf.c\" \
	-DRIOT_FILE_NOPATH=\"netbuf.c\" \
	space -Werror -Wno-unused-parameter -Wformat=0 -mlongcalls -mtext-section-literals -fstrict-volatile-bitfields -fdata-sections -ffunction-sections -fzero-initialized-in-bss -Os -std=c99 -fno-common -Wall -Wextra -Wmissing-include-dirs -fno-delete-null-pointer-checks -fdiagnostics-color -Wstrict-prototypes -Wold-style-definition -Wformat=2 -include '/data/riotbuild/riotbase/tests/lwip/bin/esp32-wroom-32/riotbuild/riotbuild.h' -Wno-address  -I/data/riotbuild/riotbase/core/include -I/data/riotbuild/riotbase/drivers/include -I/data/riotbuild/riotbase/sys/include -I/data/riotbuild/riotbase/boards/esp32-wroom-32/include -I/data/riotbuild/riotbase/cpu/esp32/include -I/data/riotbuild/riotbase/cpu/esp_common/vendor/ -I/data/riotbuild/riotbase/cpu/esp32/vendor/ -I/data/riotbuild/riotbase/cpu/esp32/vendor/esp-idf/include -I/data/riotbuild/riotbase/cpu/esp32/vendor/esp-idf/include/esp32 -I/data/riotbuild/riotbase/cpu/esp32/vendor/esp-idf/include/heap -I/data/riotbuild/riotbase/cpu/esp32/vendor/esp-idf/include/spi_flash -I/data/riotbuild/riotbase/cpu/esp32/vendor/esp-idf/include/tcpip_adapter -I/opt/esp/esp-idf/components/ -I/opt/esp/esp-idf/components/driver/include -I/opt/esp/esp-idf/components/esp32/include -I/opt/esp/esp-idf/components/heap/include -I/opt/esp/esp-idf/components/soc/esp32/include -I/opt/esp/esp-idf/components/soc/include -I/data/riotbuild/riotbase/boards/common/esp32/include -I/data/riotbuild/riotbase/cpu/esp32 -I/data/riotbuild/riotbase/sys/libc/include -I/data/riotbuild/riotbase/pkg/lwip/include -I/data/riotbuild/riotbase/tests/lwip/bin/pkg/esp32-wroom-32/lwip/src/include -I/data/riotbuild/riotbase/sys/net/gnrc/netif/include -I/data/riotbuild/riotbase/sys/posix/pthread/include -MD -MP -c -o /data/riotbuild/riotbase/tests/lwip/bin/esp32-wroom-32/lwip_api/netbuf.o /data/riotbuild/riotbase/tests/lwip/bin/pkg/esp32-wroom-32/lwip/src/api/netbuf.c
xtensa-esp32-elf-gcc: error: space: No such file or directory
/data/riotbuild/riotbase/Makefile.base:83: recipe for target '/data/riotbuild/riotbase/tests/lwip/bin/esp32-wroom-32/lwip_api/netbuf.o' failed
make[2]: *** [/data/riotbuild/riotbase/tests/lwip/bin/esp32-wroom-32/lwip_api/netbuf.o] Error 1
make[2]: Leaving directory '/data/riotbuild/riotbase/tests/lwip/bin/pkg/esp32-wroom-32/lwip/src/api'
Makefile:24: recipe for target 'lwip_api' failed
make[1]: *** [lwip_api] Error 2
make[1]: Leaving directory '/data/riotbuild/riotbase/pkg/lwip'
/data/riotbuild/riotbase/Makefile.include:526: recipe for target 'pkg-build-lwip' failed
make: *** [pkg-build-lwip] Error 2
make: *** [/Users/lars/Documents/Code/riot/makefiles/docker.inc.mk:276: ..in-docker-container] Error 2
make: Leaving directory '/Users/lars/Documents/Code/riot/tests/lwip'
```

#### Versions
Operating system: Docker under Darwin
Build environment: 
```
Operating System Environment
[... cut ...]
                  cmake: cmake version 3.15.3
               cppcheck: Cppcheck 1.89
                doxygen: 1.8.16
                    git: git version 2.23.0
                   make: GNU Make 4.2.1
                openocd: Open On-Chip Debugger 0.10.0
                 python: Python 2.7.16
                python2: Python 2.7.16
                python3: Python 3.7.4
                 flake8: error: /usr/local/opt/python/bin/python3.7: No module named flake8
             coccinelle: missing
```

cladmi, 2019-09-17:
The issue is not with `genconfigheader.sh` but with these lines https://github.com/RIOT-OS/RIOT/blob/9b4755c3babfce4c6035242f4ab1f7a4935506fc/Makefile.include#L782-L783

The issue was reported in https://github.com/RIOT-OS/RIOT/issues/5776 and have a proposed fix in https://github.com/RIOT-OS/RIOT/issues/9589#issuecomment-408587132

I got the same issue reported by @HendrikVE today and will do the PR for it.

cladmi, 2019-09-17:
I have a fix in https://github.com/RIOT-OS/RIOT/pull/12262 but currently do not manage to define the CFLAGS correctly with `DOCKER_ENVIRONMENT_CMDLINE`.

I fail to escape escaping escape :D

closed 2019-09-28 by aabadie