RIOT-OS/RIOT #12233: Question: Shell commands via telnet

isavitsky, 2019-09-15:
Is there an easy way to redirect standard output to an open gnrc_tcp socket?
I started copying the shell functions I need (_netif_list() etc…) into my telnet shell application but maybe I'm missing a better way to reuse already existing shell code.

Thanks,
Ivan

chrysn, 2019-09-15:
As far as I know there is none at the moment. stdio currently goes to UART (possibly wrapped when Ethernet-over-Serial (ETHOS) is active), the debugger (RTT) or USB serial once #11085 is merged, and all of them provide independent implementations of `stdio_init`, `stdio_read` and `stdio_write`.

(Apart from this telnet approach, there are different approaches circulating around how to best expose shell-like functionality over the network ([a mail on the -devel list](https://lists.riot-os.org/pipermail/devel/2019-September/006296.html) mentions #10624 and #9538; #7429 is related as well), but they go more in the direction of more abstract interactions rather than exposing the shell character by character.)

isavitsky, 2019-09-18:
Thank you!
I think I will take a different approach into that.

Ivan

benemorius, 2019-09-19:
I did the same thing and found that there's an easy way but there's no way that isn't a hack.

So far the best way I found was to hack the scheduler to change `stdin` and `stdout` depending on which thread is running. You can see what I did here: 54e73f125f709d321ed075f074c2e115687fec7b

The telnet server that uses this hack is here: 16cd537c2afdf4b30cd41161f5b04bacc19745ed. 

For me it's been working well in practice.

isavitsky, 2019-09-21:
Thanks for the example with stdio hack!
I'm not yet decided whether I need all the shell stuff for my task or better to implement only the limited subset via API.

jcarrano, 2019-09-24:
@HendrikVE was successful in exposing the shell via bluetooth- I believe he implemented stdio over bluetooth (which would mean it is no longer available via UART).

HendrikVE, 2019-09-24:
@jcarrano Yes, i implemented a new `stdio_nimble` module for this purpose, which replaces the `stdio_uart` module. UART and bluetooth can be used in parallel btw (UART could also be disabled by a flag). At the end the shell simply uses my stdio module and shell output/input is delivered via bluetooth.
@isavitsky You can get some inspiration by my PR #12012. You could implement your module the same way, it is important to declare your dependencies in RIOT's `Makefile.dep`, so that the buildsystem wont include a second stdio module.

miri64, 2019-10-07:
@isavitsky is your question satisfied or do you want to open a feature request for telnet?

closed 2020-04-09 by miri64

chrysn, 2022-03-02:
Just for further reference: A full telnet server is now provided, see #16723 for details.