RIOT-OS/RIOT #12243: ESP32 programming using BUILD_IN_DOCKER still expects

LordTy, 2019-09-16:
#### Description

When trying to program ESP32 boards using riotdocker, the flash command still expects the esptool.py to be located in:
`/opt/esp/esp-idf/components/esptool_py/esptool/esptool.py`
Even if esptool.py should be installed and added to the path. 

#### Steps to reproduce the issue

1. Checkout RIOT (ece313020fae620ff06a3d767073c7a6a36afce6, current master)
2. Install riot/riotbuild in docker according to: 
http://doc.riot-os.org/group__cpu__esp32.html#esp32_manual_toolchain_installation
2. Go to examples/default
3. connect esp32 board.
4. Try to make and flash te example using:
`make BOARD=ESP32-WROOM-32 PORT=/dev/cu.usbmodem1410 BUILD_IN_DOCKER=1 flash term`

#### Expected results

The image is built in the docker container and flashed to the ESP32 using the esptools.py in the path of the host.

#### Actual results

While flashing the following command is executed:
``/opt/esp/esp-idf/components/esptool_py/esptool/esptool.py --chip esp32 elf2image -fm dout -fs 2MB -ff 40m    -o /Users/tim/RIOT/Tutorials/RIOT/examples/default/bin/ESP32-WROOM-32/default.elf.bin /Users/tim/RIOT/Tutorials/RIOT/examples/default/bin/ESP32-WROOM-32/default.elf; echo "" > /Users/tim/RIOT/Tutorials/RIOT/examples/default/bin/ESP32-WROOM-32/partitions.csv; echo "nvs, data, nvs, 0x9000, 0x6000" >> /Users/tim/RIOT/Tutorials/RIOT/examples/default/bin/ESP32-WROOM-32/partitions.csv; echo "phy_init, data, phy, 0xf000, 0x1000" >> /Users/tim/RIOT/Tutorials/RIOT/examples/default/bin/ESP32-WROOM-32/partitions.csv; echo -n "factory, app, factory, 0x10000, " >> /Users/tim/RIOT/Tutorials/RIOT/examples/default/bin/ESP32-WROOM-32/partitions.csv; ls -l /Users/tim/RIOT/Tutorials/RIOT/examples/default/bin/ESP32-WROOM-32/default.elf.bin | awk '{ print $5 }' >> /Users/tim/RIOT/Tutorials/RIOT/examples/default/bin/ESP32-WROOM-32/partitions.csv; python /Users/tim/RIOT/Tutorials/RIOT/cpu/esp32/gen_esp32part.py --disable-sha256sum --verify /Users/tim/RIOT/Tutorials/RIOT/examples/default/bin/ESP32-WROOM-32/partitions.csv /Users/tim/RIOT/Tutorials/RIOT/examples/default/bin/ESP32-WROOM-32/partitions.bin
``
Which expects the esp-idf to be installed on the host:
''/opt/esp/esp-idf/components/esptool_py/esptool/esptool.py''
Which is not installed.
This results in the error:
`/bin/sh: /opt/esp/esp-idf/components/esptool_py/esptool/esptool.py: No such file or directory
ls: /Users/tim/RIOT/Tutorials/RIOT/examples/default/bin/ESP32-WROOM-32/default.elf.bin: No such file or directory`

Adding the argument ESPTOOL=esptool.py fixes the issue, but this should not be required. (It is not for esp8266).

#### Versions

Operating System Environment
-----------------------------
       Operating System: Mac OS X 10.14.5
                 Kernel: Darwin 18.6.0 x86_64 i386

Installed compiler toolchains
-----------------------------
             native gcc: Configured with: --prefix=/Applications/Xcode.app/Contents/Developer/usr --with-gxx-include-dir=/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.14.sdk/usr/include/c++/4.2.1
      arm-none-eabi-gcc: missing
                avr-gcc: avr-gcc (GCC) 4.8.1
       mips-mti-elf-gcc: missing
             msp430-gcc: missing
      riscv-none-embed-gcc: missing
      xtensa-esp32-elf-gcc: missing
      xtensa-lx106-elf-gcc: missing
                  clang: Apple LLVM version 10.0.1 (clang-1001.0.46.4)

Installed compiler libs
-----------------------
      arm-none-eabi-newlib: missing
      mips-mti-elf-newlib: missing
      riscv-none-embed-newlib: missing
      xtensa-esp32-elf-newlib: missing
      xtensa-lx106-elf-newlib: missing
      avr-libc: "1.8.0svn" ("20111229")

Installed development tools
---------------------------
                  cmake: cmake version 3.15.3
               cppcheck: Cppcheck 1.82
                doxygen: missing
                    git: git version 2.16.2
                   make: GNU Make 3.81
                openocd: Open On-Chip Debugger 0.10.0
                 python: Python 2.7.15
                python2: Python 2.7.15
                python3: Python 3.7.4
                 flake8: error: /usr/local/opt/python/bin/python3.7: No module named flake8
             coccinelle: missing

gschorcht, 2019-11-04:
There is a PR #12028 waiting for merge that should fix the problem.

aabadie, 2020-01-03:
#12750 should have fixed this issue. @LordTy, @gschorcht do you confirm ?

gschorcht, 2020-01-03:
Yes, I confirm that #12750 have fixed it.

aabadie, 2020-01-03:
Thanks,  then let's close!

closed 2020-01-03 by aabadie

gschorcht, 2020-01-03:
@aabadie Thanks for pointing that out.