RIOT-OS/RIOT #12244: Flashing on OSX uses /bin/sh instead of bash, resulting in problems with echo -n

LordTy, 2019-09-16:
#### Description

When programming ESP32 boards on OSX, the flashing command uses echo -n in bin/sh.
This does not work resulting in a corrupt csv partition table.

#### Steps to reproduce the issue

1. Checkout RIOT (ece313020fae620ff06a3d767073c7a6a36afce6, current master)
2. Install riot/riotbuild in docker according to: 
http://doc.riot-os.org/group__cpu__esp32.html#esp32_manual_toolchain_installation
2. Go to examples/default
3. connect esp32 board.
4. Try to make and flash te example using:
`make BOARD=ESP32-WROOM-32 PORT=/dev/cu.usbmodem1410 BUILD_IN_DOCKER=1 ESPTOOL=esptool.py flash term`
#### Expected results

Correctly programmed ESP32 with a correct `bin/ESP32-WROOM-32/partitions.csv`
Which should contain:

    nvs, data, nvs, 0x9000, 0x6000
    phy_init, data, phy, 0xf000, 0x1000
    factory, app, factory, 0x10000, 117664

#### Actual results

`bin/ESP32-WROOM-32/partitions.csv` contains:

    nvs, data, nvs, 0x9000, 0x6000
    phy_init, data, phy, 0xf000, 0x1000
    -n factory, app, factory, 0x10000, 
    117664

Resulting in the programmin error:

    Parsing CSV input...
    Error at line 4: Size field can't be empty

This happen because /bin/sh does not understands `echo -n`
Adding  SHELL=/bin/bash to use bash instead of sh fixes the issue.

#### Versions

Operating System Environment
-----------------------------
       Operating System: Mac OS X 10.14.5
                 Kernel: Darwin 18.6.0 x86_64 i386

Installed compiler toolchains
-----------------------------
             native gcc: Configured with: --prefix=/Applications/Xcode.app/Contents/Developer/usr --with-gxx-include-dir=/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.14.sdk/usr/include/c++/4.2.1
      arm-none-eabi-gcc: missing
                avr-gcc: avr-gcc (GCC) 4.8.1
       mips-mti-elf-gcc: missing
             msp430-gcc: missing
      riscv-none-embed-gcc: missing
      xtensa-esp32-elf-gcc: missing
      xtensa-lx106-elf-gcc: missing
                  clang: Apple LLVM version 10.0.1 (clang-1001.0.46.4)

Installed compiler libs
-----------------------
      arm-none-eabi-newlib: missing
      mips-mti-elf-newlib: missing
      riscv-none-embed-newlib: missing
      xtensa-esp32-elf-newlib: missing
      xtensa-lx106-elf-newlib: missing
      avr-libc: "1.8.0svn" ("20111229")

Installed development tools
---------------------------
                  cmake: cmake version 3.15.3
               cppcheck: Cppcheck 1.82
                doxygen: missing
                    git: git version 2.16.2
                   make: GNU Make 3.81
                openocd: Open On-Chip Debugger 0.10.0
                 python: Python 2.7.15
                python2: Python 2.7.15
                python3: Python 3.7.4
                 flake8: error: /usr/local/opt/python/bin/python3.7: No module named flake8
             coccinelle: missing

cladmi, 2019-09-19:
The solution is to never use `echo`, but `printf` instead.

cladmi, 2019-09-20:
Damn I was doing it at the same time https://github.com/RIOT-OS/RIOT/pull/12283

LordTy, 2019-09-20:
Wow, that was close only 3 seconds apart.

PR #12282 (2019-09-24): cpu/esp32: Use awk/printf instead of echo -n when flashing esp32

### Contribution description

This changes the Makefile of the esp32 cpu to use `awk` and `printf` to make the partitions csv file, instead of the `echo (-n)`

### Testing procedure

Execute the steps of #12244 without the SHELL=/bin/bash.
No error should occur during flashing.

### Issues/PRs references
fixes #12244
closed 2019-09-24 by fjmolinas