RIOT-OS/RIOT #12247: esp8266: make flash fails

benpicco, 2019-09-16:
#### Description
When following the instructions from [the documentation](http://doc.riot-os.org/group__cpu__esp8266.html#esp8266_precompiled_toolchain), esptool.py v2.5.1 gets installed.

422644bd3a switched to a notation introduced by esptool.py v2.6 that the old esptool does not understand.

#### Steps to reproduce the issue
##### from the documentation
```
sudo git clone https://github.com/gschorcht/RIOT-Xtensa-ESP8266-toolchain.git /opt/esp
export PATH=$PATH:/opt/esp/esp-open-sdk/xtensa-lx106-elf/bin
```
##### trying to flash something
```
make -C examples/hello-world BOARD=esp8266-esp-12x flash
```

#### Expected results
Flashing works.

#### Actual results
```
esptool.py elf2image -fs 1MB /home/benpicco/dev/RIOT/examples/hello-world/bin/esp8266-esp-12x/hello-world.elf
usage: esptool elf2image [-h] [--output OUTPUT] [--version {1,2}]
                         [--flash_freq {40m,26m,20m,80m}]
                         [--flash_mode {qio,qout,dio,dout}]
                         [--flash_size {4m,2m,8m,16m,32m,16m-c1,32m-c1,32m-c2}]
                         input
esptool elf2image: error: argument --flash_size/-fs: invalid choice: '1mb' (choose from '4m', '2m', '8m', '16m', '32m', '16m-c1', '32m-c1', '32m-c2')
make: *** [/home/benpicco/dev/RIOT/examples/hello-world/../../Makefile.include:605: preflash] Error 2
```

#### Versions
Riot master

cladmi, 2019-09-16:
Indeed, this is working when following this documentation http://doc.riot-os.org/group__cpu__esp8266.html#esp8266_riot_docker_toolchain which is the one I used.

cladmi, 2019-09-16:
I think the quick fix is to revert that commit, it is only a warning anyway. I never tested with the version from the toolchain as I always build `esp` with docker using pip version of `esptool.py`.

cladmi, 2019-09-16:
I created the PR to revert it. It was, as the commit said it, a deprecation warning. It was just done too early, and both 2.6 and the current 2.7 work with the old option.

gschorcht, 2019-09-16:
@benpicco @cladmi Sorry when I'm answering too late or not at all these days. I'm on vacation for hiking all the day and have often only very limited internet access. Furthermore, I don't have the hardware with me to check all the problems.

Unfortunately, the issue is a bit more complicated. The ESP8266 toolchain uses [esp-open-sdk](https://github.com/pfalcon/esp-open-sdk.git) which already includes `esptool.py`, independent on whether you [compile it by yourself](https://doc.riot-os.org/group__cpu__esp8266.html#esp8266_manual_toolchain_installation) or you use the [precompiled version](https://doc.riot-os.org/group__cpu__esp8266.html#esp8266_precompiled_toolchain). The version of this `esptool.py` is not 2.5 but version 1.2. That's why it doesn't understand the new arguments. As documented for ESP32, `esptool.py` version 2.4 is required to understand the new arguments.

I could prepare a special version of the toolchain which overwrites the included `esptool.py` with a newer version. But if the user compiles the toolchain by himself, he would also have to install `esptool.py` with `pip install esptool`. Furthermore, he would have to ensure that the newer version is found in his `PATH` before the version of `esptool.py` provided by the `esp-open-sdk`.

@cladmi Thank you for PR #12252 and reverting the change for the moment.

closed 2019-09-16 by benpicco