RIOT-OS/RIOT #12258: Rebootloop when using esp_now on the ESP32

LordTy, 2019-09-17:
#### Description

When using the esp_now network module on the ESP32, the processor keeps rebooting and the system is unusable.
This does not happen when using the esp_now module on an esp8266.

#### Steps to reproduce the issue

1. Checkout the RIOT tutorial according to the instructions
2. Install riot/riotbuild in docker according to: 
http://doc.riot-os.org/group__cpu__esp32.html#esp32_manual_toolchain_installation
2. Go to task-05
2. Add USEMODULE+=esp_now
3. connect esp32 board.
4. Try to make and flash and connect to the example using:
`make BOARD=ESP32-WROOM-32 PORT=/dev/cu.usbmodem1410 BUILD_IN_DOCKER=1 WERROR=0 ESPTOOL=esptool.py flash term`

#### Expected results

You can connect to the esp32 and use the shell.

#### Actual results

The shell does not respond and the ESP32 keeps rebooting.
This reboot gives the following information:

```
2019-09-17 10:42:21,912 - INFO # rst:0x7 (TG0WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
2019-09-17 10:42:21,915 - INFO # configsip: 0, SPIWP:0xee
2019-09-17 10:42:21,920 - INFO # clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
2019-09-17 10:42:21,923 - INFO # mode:DOUT, clock div:2
2019-09-17 10:42:21,926 - INFO # load:0x3fff0018,len:4
2019-09-17 10:42:21,928 - INFO # load:0x3fff001c,len:5632
2019-09-17 10:42:21,929 - INFO # ho 0 tail 12 room 4
2019-09-17 10:42:21,932 - INFO # load:0x40078000,len:0
2019-09-17 10:42:21,936 - INFO # load:0x40078000,len:14868
2019-09-17 10:42:21,937 - INFO # entry 0x40078628
2019-09-17 10:42:21,942 - INFO # W (30) boot: PRO CPU has been reset by WDT.
2019-09-17 10:42:21,947 - INFO # W (31) boot: WDT reset info: PRO CPU PC=0x400803c0
2019-09-17 10:42:21,953 - INFO # W (31) boot: WDT reset info: APP CPU PC=0xef9f98ac
2019-09-17 10:42:21,959 - INFO # I (37) boot: ESP-IDF v3.1-dev-961-ga255622 2nd stage bootloader
2019-09-17 10:42:21,964 - INFO # I (43) boot: compile time 12:38:02
2019-09-17 10:42:21,970 - INFO # I (47) boot: Enabling RNG early entropy source...
2019-09-17 10:42:21,973 - INFO # I (53) boot: SPI Speed      : 40MHz
2019-09-17 10:42:21,978 - INFO # I (57) boot: SPI Mode       : DOUT
2019-09-17 10:42:21,981 - INFO # I (61) boot: SPI Flash Size : 4MB
2019-09-17 10:42:21,986 - INFO # I (65) boot: Partition Table:
2019-09-17 10:42:21,992 - INFO # I (69) boot: ## Label            Usage          Type ST Offset   Length
2019-09-17 10:42:22,000 - INFO # I (76) boot:  0 nvs              WiFi data        01 02 00009000 00006000
2019-09-17 10:42:22,008 - INFO # I (83) boot:  1 phy_init         RF data          01 01 0000f000 00001000
2019-09-17 10:42:22,014 - INFO # I (91) boot:  2 factory          factory app      00 00 00010000 00063120
2019-09-17 10:42:22,020 - INFO # I (98) boot: End of partition table
2019-09-17 10:42:22,034 - INFO # I (103) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x0b3bc ( 46012) map
2019-09-17 10:42:22,042 - INFO # I (128) esp_image: segment 1: paddr=0x0001b3e4 vaddr=0x3ffb0000 size=0x0213c (  8508) load
2019-09-17 10:42:22,051 - INFO # I (132) esp_image: segment 2: paddr=0x0001d528 vaddr=0x40080000 size=0x00400 (  1024) load
2019-09-17 10:42:22,061 - INFO # I (136) esp_image: segment 3: paddr=0x0001d930 vaddr=0x40080400 size=0x026e0 (  9952) load
2019-09-17 10:42:22,090 - INFO # I (148) esp_image: segment 4: paddr=0x00020018 vaddr=0x400d0018 size=0x4a678 (304760) map
2019-09-17 10:42:22,192 - INFO # I (263) esp_image: segment 5: paddr=0x0006a698 vaddr=0x40082ae0 size=0x08a60 ( 35424) load
2019-09-17 10:42:22,197 - INFO # I (286) boot: Loaded app from partition at offset 0x10000
2019-09-17 10:42:22,203 - INFO # I (286) boot: Disabling RNG early entropy source...
2019-09-17 10:42:22,203 - INFO # 
2019-09-17 10:42:22,206 - INFO # Starting ESP32 with ID: 55a4cf1225650c
2019-09-17 10:42:22,206 - INFO # 
2019-09-17 10:42:22,214 - INFO # Current clocks in Hz: CPU=80000000 APB=80000000 XTAL=40000000 SLOW=150000
2019-09-17 10:42:22,217 - INFO # PRO cpu is up (single core mode, only PRO cpu is used)
2019-09-17 10:42:22,225 - INFO # I (0) [ heap_init]: Initializing. RAM available for dynamic allocation:
2019-09-17 10:42:22,230 - INFO # I (0) [ heap_init]: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
2019-09-17 10:42:22,234 - INFO # I (0) [ heap_init]: At 3FFBACB8 len 00025348 (148 KiB): DRAM
2019-09-17 10:42:22,239 - INFO # I (0) [ heap_init]: At 3FFE0440 len 00003BC0 (14 KiB): D/IRAM
2019-09-17 10:42:22,245 - INFO # I (0) [ heap_init]: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
2019-09-17 10:42:22,250 - INFO # I (0) [ heap_init]: At 4008B540 len 00014AC0 (82 KiB): IRAM
2019-09-17 10:42:22,253 - INFO # Heap free: 158588 byte
2019-09-17 10:42:22,258 - INFO # PRO cpu starts user code
2019-09-17 10:42:22,266 - INFO # Used clocks in Hz: CPU=80000000 APB=80000000 XTAL=40000000 FAST=8000000 SLOW=150000
2019-09-17 10:42:22,269 - INFO # XTAL calibration value: 3462093
2019-09-17 10:42:22,269 - INFO # Heap free: 158544 bytes
2019-09-17 10:42:22,294 - INFO # System time: 1970-01-01 00:00:32
```
Showing that somewhere the processor is stuck and is rebooted by the WDT.

#### Versions

Operating System Environment
-----------------------------
       Operating System: Mac OS X 10.14.5
                 Kernel: Darwin 18.6.0 x86_64 i386

Installed compiler toolchains
-----------------------------
             native gcc: Configured with: --prefix=/Applications/Xcode.app/Contents/Developer/usr --with-gxx-include-dir=/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.14.sdk/usr/include/c++/4.2.1
      arm-none-eabi-gcc: missing
                avr-gcc: avr-gcc (GCC) 4.8.1
       mips-mti-elf-gcc: missing
             msp430-gcc: missing
      riscv-none-embed-gcc: missing
      xtensa-esp32-elf-gcc: missing
      xtensa-lx106-elf-gcc: missing
                  clang: Apple LLVM version 10.0.1 (clang-1001.0.46.4)

Installed compiler libs
-----------------------
      arm-none-eabi-newlib: missing
      mips-mti-elf-newlib: missing
      riscv-none-embed-newlib: missing
      xtensa-esp32-elf-newlib: missing
      xtensa-lx106-elf-newlib: missing
      avr-libc: "1.8.0svn" ("20111229")

Installed development tools
---------------------------
                  cmake: cmake version 3.15.3
               cppcheck: Cppcheck 1.82
                doxygen: missing
                    git: git version 2.16.2
                   make: GNU Make 3.81
                openocd: Open On-Chip Debugger 0.10.0
                 python: Python 2.7.15
                python2: Python 2.7.15
                python3: Python 3.7.4
                 flake8: error: /usr/local/opt/python/bin/python3.7: No module named flake8
             coccinelle: missing

gschorcht, 2019-09-18:
Really strange. What I could figure out is that the problem arose 3 days ago by merging PR #10195, which apparently has nothing to do with ESP-NOW.

When you checkout commit 95a2081cfa61fd829d740d1346c41454380b42e1 it works as exepected.
But, when you checkout commit 65b71026088f5c2ef7b99fab1e00ca5779f83127, the problem occurs.

@benpicco Would you have some time to check whether the changes of compile options of PR #10195 affect the compilation of ESP32 code? Maybe it is just a side effect and the real problem is something else. You could use 
```
USEMODULE=esp_now make BOARD=esp32-wroom-32 -C examples/gnrc_networking flash
```
which produces the same problem but makes the test easier since it doesn't require the fix in PR #12263.

@jcarrano Might changed compile options produce problems on certain architectures?

gschorcht, 2019-09-19:
I spent some time for further investigation of the problem. Indeed, the problem is caused by the very small change in PR #10195: 
```diff
--- a/makefiles/cflags.inc.mk
+++ b/makefiles/cflags.inc.mk
@@ -65,5 +65,5 @@ CFLAGS += $(filter-out $(OPTIONAL_CFLAGS_BLACKLIST),$(OPTIONAL_CFLAGS))
 # Default ARFLAGS for platforms which do not specify it.
 # Note: make by default provides ARFLAGS=rv which we want to override
 ifeq ($(origin ARFLAGS),default)
-  ARFLAGS = rcs
+  ARFLAGS = rcTs
 endif
```
@jcarrano Do you have any idea why it could be a problem? `ar` man page just says
```
-T       Make the specified archive a thin archive.  If it already exists and is a regular archive, the
           existing members must be present in the same directory as archive.
```

benpicco, 2019-09-23:
The reboot loop also happens with `esp_wifi`. Changing `rcTs` back to `rcs` fixes the problem.

I'm afraid we'll have to revert #10195 :disappointed:

cladmi, 2019-09-24:
Why globally? Just change the `ARFLAGS` in the `Makefile` of the modules that cause issues.

Or if it is an `esp` `ar` specific issue, just define it in the cpu Makefile.

It creates the archive without actually putting the files inside the archive, just links to the objects files. Not really sure why it could cause issues though…

cladmi, 2019-09-24:
> Or if it is an `esp` `ar` specific issue, just define it in the cpu Makefile.

I just noticed the PR doing it.

PR #12292 (2019-09-27): cpu/esp32: disable thin archives if esp_wifi is used
### Contribution description
For a yet unknown reason esp32 gets stuck in a reboot loop when thin archives are used.

As a workaround, disable thin archives for now if esp_wifi is used.

### Testing procedure
Compile `examples/gnrc_networking` with `USEMODULE += esp_wifi`

On master, the device will be stuck in a boot loop.

With this applied RIOT starts up normally and connects to the WiFi.

### Issues/PRs references

fixes #12258
bug introduced by #10195
commit 111fe80e (2019-09-27): cpu/esp: disable thin archives if esp_wifi is used

For a yet unknown reason, both esp8266 and esp32 get stuck in a
reboot loop when thin archives are used.

As a workaround, disable thin archives for now if esp_wifi is used.

fixes #12258
closed 2019-09-27 by benpicco