RIOT-OS/RIOT #12311: riotboot: irq not enabled after jumping to slot

basilfx, 2019-09-26:
#### Description
While working on #11940, I noticed that all EFM32 boards I own work, except the SLTB001a.

This board has a I2C power interrupt controller that is used during `board_init()` after `cpu_init()`. I2C is using interrupts in combination with `__WFE()`. I noticed that `tests/riotboot` did not work, unless I skipped the I2C part. After a lot of debugging and testing, I added `irq_enable()` before [this line](https://github.com/RIOT-OS/RIOT/blob/master/cpu/efm32/periph/i2c.c#L60), and it worked!

My hypothesis is:
- when jumping to the slot by the bootloader, `irq_disable()` is invoked
- it is not enabled/reset during startup of the actual slot
- the transfer uses `irq_disable()` and `irq_restore()`, so it stays disabled
- therefore, the slot boots differently then without riotboot.

This is a bit out of my league, and I don't have a good idea to solve this. Maybe my driver is wrong? Maybe the startup needs to be changed?

#### Steps to reproduce the issue
Take the SLTB001a and use #11940. Try to run `tests/riotboot`.

#### Expected results
The test results should appear, and you can interact with the shell.

#### Actual results
No test results appear, and no interaction is possible.

#### Versions
```
Operating System Environment
-----------------------------
       Operating System: Mac OS X 10.14.6
                 Kernel: Darwin 18.7.0 x86_64 i386

Installed compiler toolchains
-----------------------------
             native gcc: Configured with: --prefix=/Library/Developer/CommandLineTools/usr --with-gxx-include-dir=/Library/Developer/CommandLineTools/SDKs/MacOSX10.14.sdk/usr/include/c++/4.2.1
      arm-none-eabi-gcc: arm-none-eabi-gcc (GNU Tools for Arm Embedded Processors 8-2019-q3-update) 8.3.1 20190703 (release) [gcc-8-branch revision 273027]
                avr-gcc: missing
       mips-mti-elf-gcc: missing
             msp430-gcc: missing
   riscv-none-embed-gcc: missing
   xtensa-esp32-elf-gcc: missing
   xtensa-lx106-elf-gcc: missing
                  clang: Apple LLVM version 10.0.1 (clang-1001.0.46.4)

Installed compiler libs
-----------------------
   arm-none-eabi-newlib: "3.1.0"
    mips-mti-elf-newlib: missing
riscv-none-embed-newlib: missing
xtensa-esp32-elf-newlib: missing
xtensa-lx106-elf-newlib: missing
               avr-libc: missing (missing)

Installed development tools
---------------------------
                  cmake: cmake version 3.15.3
               cppcheck: missing
                doxygen: missing
                    git: git version 2.23.0
                   make: GNU Make 3.81
                openocd: missing
                 python: Python 2.7.16
                python2: Python 2.7.16
                python3: Python 3.7.4
                 flake8: error: /usr/local/opt/python/bin/python3.7: No module named flake8
             coccinelle: missing
```

kaspar030, 2019-09-26:
Hm, [here](https://github.com/kaspar030/RIOT/blob/ac9e80cfd0e8715aa451da9e20443b641a9c5278/cpu/cortexm_common/include/cpu.h#L203) irqs are actually disabled. The comment is pretty awesome (I know I checked it in but I'm sure I didn't write that. :wink:).

Removing that line should also fix your problem. Could you try?
I don't think interrupts need to be disabled there, but need to run more tests.

basilfx, 2019-09-26:
Yes, that works as well.

PR #12312 (2019-10-08): cpu/cortexm: don't disable IRQs in cpu_jump_to_image()

### Contribution description

```cpu_jump_to_image()``` disables IRQs. As the function is supposed to be called from bootloaders, this shouldn't be necessary.

### Testing procedure

Primarily all users of riotboot must still work.

### Issues/PRs references

Fixes #12311. *edit* fixed reference
closed 2019-10-08 by maribu