RIOT-OS/RIOT #12320: Particle boards: Support DFU mode for flashing

chrysn, 2019-09-27:
#### Description

The Particle family has, in its default firmware, a `dfu-util` based bootloader that can be entered by keeping the "mode" button pressed at power-up (or after any other reset situation) and releasing it when the LED flashes yellow.

I couldn't find out yet what memory area is available for flashing while this bootloader is in place, and whether it's an "actual" bootloader (can preempt booting of anything flashed) or just a fall-back mode that needs the application's assistance to enter (in which case it'd need something in the board startup script to stay usable for more than one time), but at any rate I think it'd be good to have the programming possibility available.

#### Context

I see this as part of a larger development: While many dev kits have had programmers and UART adapters for quite some time (and they're not too complex, just another M3 wired to the debug ports), that does contribute to board complexity and thus cost – and with bootloaders becoming more available, my assumption is that we'll see more and more boards without programmers. Sure, they're no good options for low-level development, but RIOT is mature enough that developers more on the "user of RIOT" side can come a long way without ever getting into situations where getting a debugger's view of the system becomes necessary.

### Useful links

* Description of Particle Xenon modes: https://docs.particle.io/tutorials/device-os/led/xenon/

#### Next steps

I'll have a deeper look into the memory layout involved and the dfu-util invocations required. A deep look into how memory regions are configured (and thus how the information about an involved bootloader must be passed in) will help as well.

As outcome of ensuing discussion I hope to have a PR that allows something like `make BOARD=particle-xenon USING_BOOTLOADER=particle flash` to program a board as fresh out of the box.

#### Alternatives

Buy a .1"-to-.05" programmer adapter cable. Naaah…

chrysn, 2019-09-27:
Some research notes:

* The [local build guide](https://docs.particle.io/support/particle-tools-faq/local-build/) contains instructions of how to get to a `make program-dfu` step; that seemed to be the easiest way to getting a memory map used in whatever gets flashed via dfu (as in their system of modules and applications I'm not sure what gets built usually).
* What gets flashed in the end is a a system-part1 and a user part with the linker script in `modules/xenon/…-part/linker.ld` generated at build time. The system part1 is flashed with `dfu-util -a 0 -s 0x30000 -D`, and the user part with `dfu-util -a 0 -s 0xD4000:leave -D`. Oddly, the files passed in are named `.dfu` (`build/target/…-part/platform-14-m/…-part.dfu`) which dfu-util says not to do with `-s`, but given (at least the system) part is definitely a Cortex vector (first 32 bit word is a 0x2003f000 RAM address, then several that are either around 0x0004… or 0), so I guess they just use an exotic file extension for what's usually called `.bin`.
* Looking into the `modules/xenon/*part*/linker.ld` scripts, this uses the nRF softdevice and puts its system code in the area between APP_CODE_BASE (something the softdevice currently defines as 192k) and the a point (128k + 48k) from the 1M flash head (per calculation in the linker script) and the user code in the 128k below the top 48k per calculated-out numbers (I suppose linker script arithmetic would have been too complicated here).
So the memory layout is 192k soft-device at the bottom, then system+user, then 48k bootloader at the top end of the 1M.
* nm'ing `build/target/system-part1/platform-14-m/system-part1.elf` confirms that it has its __isr_vector at 192k, objdump shows code starting at 192k+something.
* That's all consistent with the dfu-util invocation putting things right at 192k (0x30000) in raw mode.

(That's around the point where I found the [memory map documentation](https://docs.particle.io/datasheets/mesh/xenon-datasheet/#memory-map) that'd have told me the same thing.)

The common theme around the [nRF softdevice-involving boot process](https://devzone.nordicsemi.com/f/nordic-q-a/27599/boot-sequence-for-nrf52832) seems to be that there's a NRF_APP_IMAGE_START_ADDR and a NRF_UICR_BOOTLOADER_START_ADDR, and whatever wants to keep the bootloader intact better play inside there (as the bootloader may also check the softdevice). So better keep it all in place while we're using that bootloader.

With those limits, I could build a linker script for working inside that bootloader; however, the only effect it had is that the Xenon now always comes up in the DFU mode (LED flashing yellow).

chrysn, 2019-09-28:
For the spark core board (a predecessor of the current particle boards if I'm not mistaken), exactly what I'm asking for here has already been working for years, and was built up to [in the particle forums](https://community.particle.io/t/replace-core-firmware-with-riot-os/7007/11) with disabling the watchdog the bootloader sets up. As in this current case the behavior is different (immediate bootloader), I suspect that there is some form of firmware validity checking going on, [asking there](https://community.particle.io/t/replace-xenon-co-firmware-with-riot-os/52382) for assistance.

dpgeorge, 2019-09-28:
We recently added support for the Xenon board in MicroPython, and it currently just overwrites the entire flash, requiring an external programmer.  But I have made progress supporting the built-in Particle bootloader, which you can see in this PR: https://github.com/micropython/micropython/pull/5158.  This may prove useful to you. 
 It needs a special structure at 0x200, and a CRC at the end.

chrysn, 2019-09-30:
Wow, that is really helpful, thanks.

With that, I could flash a monofirmware that would at least not get rejected immediately. (Observation from in-between: If the image end is set to zero, the bootloader enters some strange loop, which may hint at that the checksum is not checked in that case).

The program runs only for a fraction of a second (probably until a short preconfigured watchdog bites), but hey, great success thanks to you, @dpgeorge!

Next steps: Put the watchdog to sleep, publish branch, find out whether I can get the final binary length right from the linker (as I'm trying for a C-populated module_info_t).

chrysn, 2019-09-30:
Funny, it's not actually the hardware watchdog that's biting, it's a timer. After the bootloader NVIC->ISER has interrupts 0 and 6 (POWER_CLOCK and GPIOTE) enabled, clearing them made the programs run without apparent limit. (It's fortunate that it's not a hardware timer; those can't be turned off without a hard reset). Thus I can run applications as complex as the USB or the nimble GATT example.

My changes are in [my particle-xenon-native-bootloader branch](https://github.com/chrysn-pull-requests/RIOT/tree/particle-xenon-native-bootloader), currently at https://github.com/chrysn-pull-requests/RIOT/commit/65c46a7ec12ccfecef336e2c0f9ce41c1bb9e130.

Unlike the micropython code I'm trying to do as much as possible in C by having a module_info_t at the right location (but as I later found I can't easily get the complete bin file size from C, it's now an ugly hybrid), leaving only verification and checksumming (and, because ugly hybrid, setting the final size value) to the Makefile (which calls a Python one-liner for checksumming).

dpgeorge, 2019-09-30:
> Funny, it's not actually the hardware watchdog that's biting, it's a timer

You might need to set VTOR.

chrysn, 2019-09-30:
> You might need to set VTOR
I did, but my handler for the POWER_CLOCK timer was a default (probably fault) handler b/c RIOT doesn't use that interrupt. Honestly I don't quite understand their approach here (either you jump into someone else's code via a VTOR, then you're behaving like a true bootloader, and should disable all NVIC interrupts and fully set the VTOR -- or you're jumping into an application, in which case you don't need to have something VTOR'ish in the front but just get a jump address from your container format), but hey it's working now :-)

(And not to do particle unjustice, a) the above is just my view of bootloaders, and b) they may even do it that way, and how I'm using is just a crooked way into their ecosystem).

dpgeorge, 2019-10-01:
> I'm trying to do as much as possible in C by having a module_info_t at the right location

You might be able to put in a fake, fixed size, even a fake BASE value, and then you'll know the fixed CRC value.  This can then all be encoded in C, no need for additional steps to inject the size and checksum at the end.

chrysn, 2019-11-18:
I've tried out various fake fixed values, and claiming that the firmware is only a few hundred bytes (eg. up to and including the module_info_t) works if I place the checksum right after that. Unfortunately, changing module_start_address appears to break the boot process (probably the loader tries to follow the reset vector at the indicated location and fails the sanity check), so a zero-sized checksummed piece would have its checksum right at the image start address from where the code is executed -- so I can't go all the way and have everything constant. (There might still be a possibility of making it zero-sized if `0` (the checksum of the empty length) is accepted by the bootloader as a valid stack pointer address as that's what comes first in the vector table, and RIOT will (or could) set it at startup, but that's a tad too deep for me to seriously consider right now).

Still, I'd use this suggestion by only checksumming up to and including the module_info_t block. This allows applications to freely use the flash page driver on their remaining memory.

---

The branch has been updated to include some documentation and cleaner makefile handling; when #12344 is merged, I'll create a PR.

closed 2020-11-06 by chrysn