RIOT-OS/RIOT #12356: Question: stackless transmission

tinstructor, 2019-10-02:
I was wondering if there's currently an easy way to bypass the networking stack. More specifically, I want to trigger packet transmission wherein the only header is PHY dependent (and probably implemented in hardware) and with a PHY payload that is completely of my own choosing. This means no routing, 6lowpan, nor MAC headers. For example, based on the work of @benpicco in #12128 would it be possible to do something like [this paper](https://www.mdpi.com/2079-9292/8/9/1045/pdf-vor). In order for such tests to succeed and be repeatable, there can be no routing traffic traffic (nor any other overhead) and medium access should be user-triggered only (i.e., no carrier sensing or anything else of that nature).

miri64, 2019-10-02:
First of all, be aware that @jia200x is currently doing some remodeling in that area. However, currently the communication you want to do is possible via [`netdev`](https://doc.riot-os.org/group__drivers__netdev__api.html).

tinstructor, 2019-10-02:
@miri64 I assume one would need to implement support for such a thing in the RF device driver?

kaspar030, 2019-10-02:
> @miri64 I assume one would need to implement support for such a thing in the RF device driver?

Currently, the netdev->send() function expects a correctly formed L2 PDU. You'd need to create that somehow.

benpicco, 2019-10-02:
@tinstructor try adding your openmote-b to the `BOARD_PROVIDES_NETIF` in the Makefile in `examples/default` (it should have been there all along).

This should give you a netif example with only layer 2.

tinstructor, 2019-10-02:
> Currently, the netdev->send() function expects a correctly formed L2 PDU. You'd need to create that somehow.

At first sight I'd think that's a problem since #4858 is currently not implemented and I would rather not use an old MAC frame format. However, for the purpose of sending a generic data frame, I can't actually think of anything that would change. Therefore, I'll resort to including a L2 header and just pad it by putting bogus data in the MAC payload in order to get the required PHY payload size.

tinstructor, 2019-10-02:
> This should give you a netif example with only layer 2.

@benpicco I saw that CSMA is configured by default? How would I bypass this?

benpicco, 2019-10-02:
`ifconfig 4 -csma` should do.

tinstructor, 2019-10-02:
> `ifconfig 4 -csma` should do.

Can I also turn of automatic ACKs and unset the ACK requirement field in the MHR?

benpicco, 2019-10-02:
> Can I also turn of automatic ACKs and unset the ACK requirement field in the MHR?

```
ifconfig 4 -autoack
ifconfig 4 -ack_req
```

tinstructor, 2019-10-02:
> > Can I also turn of automatic ACKs and unset the ACK requirement field in the MHR?
> 
> ```
> ifconfig 4 -autoack
> ifconfig 4 -ack_req
> ```

How is the unsetting of such options implemented exactly? This seems interesting to me and I haven't encountered it in the course materials and tutorials I've walked through to this point?

benpicco, 2019-10-02:
You can check the source of `ifconfig`: Those [options](https://github.com/RIOT-OS/RIOT/blob/master/sys/shell/commands/sc_gnrc_netif.c#L55) are translated into their corresponding `NETOPT_` enum and then sent to the netif thread by [`gnrc_netapi_set`](https://riot-os.org/api/group__net__gnrc__netapi.html#ga88f71db0621dc9ce78d14664f5e23f4b).

The driver then has a large [switch-case block](https://github.com/RIOT-OS/RIOT/blob/8f27f7a290dcf8428b76de98f2822e522a177722/drivers/at86rf215/at86rf215_netdev.c#L419) where it performs the appropriate action to execute the option change.

tinstructor, 2019-10-02:
@benpicco So where can I choose which options are set/unset by default?

benpicco, 2019-10-02:
I'd say you add a few calls to `gnrc_netapi_set()` to your code.

tinstructor, 2019-10-02:
> I'd say you add a few calls to `gnrc_netapi_set()` to your code.

Which example is best suited to get me up to speed with this quickly? After some digging, a call to `gnrc_netapi_set()`  is not straightforward at all.

tinstructor, 2019-10-02:
Something like this?
```c
#include <string.h>

#include "thread.h"
#include "shell.h"
#include "shell_commands.h"

#ifdef MODULE_NETIF
#include "net/gnrc/pktdump.h"
#include "net/gnrc.h"

static gnrc_netif_t *netif = NULL;
#endif

int main(void)
{
#ifdef MODULE_NETIF
    gnrc_netreg_entry_t dump = GNRC_NETREG_ENTRY_INIT_PID(GNRC_NETREG_DEMUX_CTX_ALL,
                                                          gnrc_pktdump_pid);
    gnrc_netreg_register(GNRC_NETTYPE_UNDEF, &dump);
    netif = gnrc_netif_iter(netif);
    gnrc_netapi_set(netif->pid, NETOPT_AUTOACK, 0, NETOPT_DISABLE, 1);
#endif

    (void) puts("Welcome to RIOT!");

    char line_buf[SHELL_DEFAULT_BUFSIZE];
    shell_run(NULL, line_buf, SHELL_DEFAULT_BUFSIZE);

    return 0;
}
```

benpicco, 2019-10-02:
Say you want to disable auto ACK on all interfaces:

```C
for (gnrc_netif_t *netif = gnrc_netif_iter(NULL); netif; netif = gnrc_netif_iter(netif)) {
    netopt_enable_t set = NETOPT_DISABLE;
    gnrc_netapi_set(netif, NETOPT_AUTOACK, 0, &set, sizeof(set));
}
```

(untested)

closed 2019-10-02 by tinstructor

benpicco, 2019-11-07:
btw.: I'm also building a little range testing application to try out the different modulations.

It just uses UDP and I didn't configure all modulations yet (also better start with the most resilient one, otherwise you might get no handshake yet still have a better modulation that would work - although that's a bit catch 22 since the goal of this test is to find the best modulation... hmm)

It might not be the prettiest, but maybe you find it useful: https://github.com/benpicco/RIOT/tree/at86rf215-range_test/examples/range_test