RIOT-OS/RIOT #12371: fail to send data to can bus

jhx2002, 2019-10-04:
1. hardware: stm32f4  and sn65hvd230  can transceiver
2. test code:   tests/conn_can/
3. use command: test_can send
4. sampling with oscilloscope, I can see regular square wave on TX pin of transceiver. But
there is no regular wave on CANH or CANL pin of transceiver .

Supern00dle, 2019-10-24:
Hello jhx2002,

did you add a CAN Termination? the Tranceiver cannot send something on the physical bus without a 60 Ohm termination.

jhx2002, 2019-10-26:
> Hello jhx2002,
> 
> did you add a CAN Termination? the Tranceiver cannot send something on the physical bus without a 60 Ohm termination.

Yes I did, I add 120 Ohm terminaiton.

vincent-d, 2019-10-28:
@jhx2002 did you check the configuration of your transceiver (most transceivers have a low power mode where they don't emit anything)? Is there another node up on the bus that can acknowledge the frames (if frames are not acknowledged, the controller will try to re-emit endlessly)?

jhx2002, 2019-10-28:
There is a can-analyzer on the bus, analyzer did not receive frame.

Supern00dle, 2019-11-01:
@jhx2002  Many CAN analyzers do not send acks on the bus so for a communication you need a second CAN device.
As there is data on the tx line of the micro controller this seems not an OS issue.

Please provide a schematic of your implementation and detailed configuration of your settings in the micro controller.

fjmolinas, 2020-01-10:
@jhx2002, is this still an issue for you?

jhx2002, 2020-01-15:
Sorry I have no time to trace this issue recently. I'm not sure.

wosym, 2021-11-20:
Two year old issue that has been added and removed from Release milestones over and over again without any more input. Can this be closed? Since it was most likely a setup-problem and not a driver issue. Also: the CAN system in RIOT has changed quite substantially since then.