RIOT-OS/RIOT #12373: Leds not working on stm32 based boards

toonst, 2019-10-04:
#### Description

Since https://github.com/RIOT-OS/RIOT/pull/11758, gpio's are initialized from `cpu_init`. Since some boards eg. stm32f429i-disc1 contain led initialization before `cpu_init`, the led gpio initialization will be overridden and leds might not work on those boards.

#### Steps to reproduce the issue
Run `tests/leds` on stm32 based boards.

#### Expected results
Leds should blink

#### Actual results
Leds don't blink

#### Versions
Operating System Environment
-----------------------------
       Operating System: "Ubuntu" "18.04.3 LTS (Bionic Beaver)"
                 Kernel: Linux 4.15.0-64-generic x86_64 x86_64

Installed compiler toolchains
-----------------------------
             native gcc: gcc (Ubuntu 7.4.0-1ubuntu1~18.04.1) 7.4.0
      arm-none-eabi-gcc: arm-none-eabi-gcc (GNU Tools for Arm Embedded Processors 8-2018-q4-major) 8.2.1 20181213 (release) [gcc-8-branch revision 267074]
                avr-gcc: avr-gcc (GCC) 5.4.0
       mips-mti-elf-gcc: missing
             msp430-gcc: missing
   riscv-none-embed-gcc: missing
   xtensa-esp32-elf-gcc: missing
   xtensa-lx106-elf-gcc: missing
                  clang: clang version 6.0.0-1ubuntu2 (tags/RELEASE_600/final)

Installed compiler libs
-----------------------
   arm-none-eabi-newlib: "3.0.0"
    mips-mti-elf-newlib: missing
riscv-none-embed-newlib: missing
xtensa-esp32-elf-newlib: missing
xtensa-lx106-elf-newlib: missing
               avr-libc: "2.0.0" ("20150208")

Installed development tools
---------------------------
                  cmake: cmake version 3.15.3
               cppcheck: Cppcheck 1.82
                doxygen: 1.8.13
                    git: git version 2.17.1
                   make: GNU Make 4.1
                openocd: Open On-Chip Debugger 0.10.0+dev-g28d6b019 (2018-05-25-11:10)
                 python: Python 2.7.15+
                python2: Python 2.7.15+
                python3: Python 3.6.8
                 flake8: error: /usr/bin/python3: No module named flake8
             coccinelle: missing

aabadie, 2019-10-04:
I tested on a couple of boards and it's working. But those boards are initializing the leds after `cpu_init` is called. The stm32-f429i-disc1 doesn't work and led pins are initialized before.
So this is a bug on the stm32 with pins initialized before `cpu_init`.

Let's try to list these ones here.

aabadie, 2019-10-04:
Boards impacted:
- [ ] stm32f0discovery
- [ ] stm32f3discovery
- [ ] stm32f4discovery
- [ ] stm32f428i-disc1
- [ ] stm32l476g-disco
- [ ] stm32l0538-disco

aabadie, 2019-10-04:
@toonst, can you confirm that #12374 fixes your issue ?

aabadie, 2019-10-04:
To be more precise, L0 and L4 based boards are working in master because there's no need to initialize them to AIN for power consumption by default. But for consistency, they are also updated in #12374.

closed 2019-10-04 by fjmolinas

toonst, 2019-10-04:
Wow quick response. Good job!

PR #12374 (2019-10-04): boards/stm32: initialize board gpios after cpu

### Contribution description

This update all STM32 based boards where gpio are initialized after cpu in `board_init`. Since #11758, `tests/leds` is broken for these boards as reported in #12373

### Testing procedure

Run `tests/leds` on modified boards, see https://github.com/RIOT-OS/RIOT/issues/12373#issuecomment-538433121, and verify the leds are blinking.
On master they are not blinking at all.

### Issues/PRs references

fixes #12373