RIOT-OS/RIOT #12384: CDC-ACM (serial console) loses characters

ant9000, 2019-10-07:
#### Description
I am using newly merged PR #11085 on a SAMR34-Xpro board, with USB support added as per #12135 .

The driver works mostly fine, but after boot some characters are lost. The first few lines of output do arrive perfectly, then there is a gap in trasmission, and then things are smooth again. I've tried adding an fflush() call after each output, but it does not make any difference. Same output under UART works flawlessly.

The same behaviour is exhibited by pyterm, screen, and even cat.

#### Steps to reproduce the issue

I'm using [my SAMR34 branch](https://github.com/ant9000/RIOT/tree/samr34), but AFAICT any SAM0 based board should exhibit the same behaviour since they share a common USB driver. Possibly the same bug is not even SAM0 specific - can't tell, since I only have this device at hand.

This simple change to tests/usbus_cdc_acm_stdio is enough to trigger the problem:

```
diff --git a/tests/usbus_cdc_acm_stdio/main.c b/tests/usbus_cdc_acm_stdio/main.c
index e342013af..1b974c4d0 100644
--- a/tests/usbus_cdc_acm_stdio/main.c
+++ b/tests/usbus_cdc_acm_stdio/main.c
@@ -25,6 +25,11 @@
 int main(void)
 {
     (void) puts("RIOT USB CDC ACM shell test");
+    (void) puts("1 RIOT USB CDC ACM shell test");
+    (void) puts("2 RIOT USB CDC ACM shell test");
+    (void) puts("3 RIOT USB CDC ACM shell test");
+    (void) puts("4 RIOT USB CDC ACM shell test");
+    (void) puts("5 RIOT USB CDC ACM shell test");
 
     char line_buf[SHELL_DEFAULT_BUFSIZE];
     shell_run(NULL, line_buf, SHELL_DEFAULT_BUFSIZE);
```

#### Expected results

All lines in puts() should appear in the output.

#### Actual results

After the standard RIOT banner, only the first two lines are output. If I type help, the text I get back starts like this:

```
2 RIhelp
```

so definitely a good part of the initial output characters has been lost. Subsequent output (with ps or help) seem to be fine.

#### Versions

```
Operating System Environment
-----------------------------
       Operating System: "Debian GNU/Linux" "10 (buster)"
                 Kernel: Linux 4.19.0-6-amd64 x86_64 unknown

Installed compiler toolchains
-----------------------------
             native gcc: gcc (Debian 8.3.0-6) 8.3.0
      arm-none-eabi-gcc: arm-none-eabi-gcc (15:7-2018-q2-6) 7.3.1 20180622 (release) [ARM/embedded-7-branch revision 261907]
                avr-gcc: missing
       mips-mti-elf-gcc: missing
             msp430-gcc: missing
   riscv-none-embed-gcc: missing
   xtensa-esp32-elf-gcc: missing
   xtensa-lx106-elf-gcc: missing
                  clang: clang version 7.0.1-8 (tags/RELEASE_701/final)

Installed compiler libs
-----------------------
   arm-none-eabi-newlib: "3.1.0"
    mips-mti-elf-newlib: missing
riscv-none-embed-newlib: missing
xtensa-esp32-elf-newlib: missing
xtensa-lx106-elf-newlib: missing
               avr-libc: missing (missing)

Installed development tools
---------------------------
                  cmake: cmake version 3.13.4
               cppcheck: missing
                doxygen: 1.8.13
                    git: git version 2.20.1
                   make: GNU Make 4.2.1
                openocd: Open On-Chip Debugger 0.10.0
                 python: Python 2.7.16
                python2: Python 2.7.16
                python3: Python 3.7.3
                 flake8: error: /usr/bin/python3: No module named flake8
             coccinelle: missing
```

bergzand, 2019-10-08:
Thank you for the issue. This is a design issue with the current implementation of CDC ACM. With CDC ACM RIOT depends on the host computer to request bytes from the USB endpoint. This endpoint, being a bulk type endpoint, doesn't give any bandwidth guarantees. To keep things predictable in delay, the CDC ACM module buffers the stdio output until the host requested and received the data.

[This define](https://github.com/RIOT-OS/RIOT/blob/master/sys/include/usb/usbus/cdc/acm.h#L45) configures the default size for the buffer (must be a power of 2), increasing it to 256 bytes, your example shouldn't lose any characters.

I'm thinking of adding a few defines to add a blocking mode and to add a mode where the last `USBUS_CDC_ACM_STDIO_BUF_SIZE` bytes are kept in the buffer (instead of the first `USBUS_CDC_ACM_STDIO_BUF_SIZE` bytes).

The other issue with the blocking mode is that if there is no host connected threads will start blocking on stdio after the buffer is full.

dylad, 2019-10-08:
@bergzand I tried to increase this define to 1024. It doesn't help.

output with pyterm:

```
$ dist/tools/pyterm/pyterm -p /dev/ttyACM0
Twisted not available, please install it if you want to use pyterm's JSON capabilities
2019-10-08 20:52:17,394 # Connect to serial port /dev/ttyACM0
Welcome to pyterm!
Type '/exit' to exit.
2019-10-08 20:52:22,280 # Serial port disconnected, waiting to get reconnected...
2019-10-08 20:52:23,282 # Try to reconnect to /dev/ttyACM0 again...
2019-10-08 20:52:23,283 # Reconnected to serial port /dev/ttyACM0
2019-10-08 20:52:27,306 # Serial port disconnected, waiting to get reconnected...
2019-10-08 20:52:28,308 # Try to reconnect to /dev/ttyACM0 again...
2019-10-08 20:52:28,309 # Reconnected to serial port /dev/ttyACM0
2019-10-08 20:52:28,310 # main(): This is RIOT! (Version: 2019.07-devel-2276-g866ec-devel/usbus_msc)
2019-10-08 20:52:28,311 # RIOT USB CDC ACM shell test
> 2019-10-08 20:52:32,066 # Serial port disconnected, waiting to get reconnected...
2019-10-08 20:52:33,070 # Try to reconnect to /dev/ttyACM0 again...
2019-10-08 20:52:33,071 # Serial port disconnected, waiting to get reconnected...
2019-10-08 20:52:34,072 # Try to reconnect to /dev/ttyACM0 again...
2019-10-08 20:52:34,073 # Reconnected to serial port /dev/ttyACM0
2019-10-08 20:52:34,076 #  main(): This is RIOT! (Version: 2019.07-devel-2276-g866ec-devel/usbus_msc)
2019-10-08 20:52:34,077 # RIOT USB CDC ACM shell test
> 2019-10-08 20:52:37,111 # Serial port disconnected, waiting to get reconnected...
2019-10-08 20:52:38,112 # Serial port disconnected, waiting to get reconnected...
2019-10-08 20:52:39,114 # Try to reconnect to /dev/ttyACM0 again...
2019-10-08 20:52:39,116 # Reconnected to serial port /dev/ttyACM0
2019-10-08 20:52:42,244 # Serial port disconnected, waiting to get reconnected...
2019-10-08 20:52:43,246 # Serial port disconnected, waiting to get reconnected...
2019-10-08 20:52:44,248 # Try to reconnect to /dev/ttyACM0 again...
2019-10-08 20:52:44,249 # Reconnected to serial port /dev/ttyACM0
2019-10-08 20:52:47,828 # Serial port disconnected, waiting to get reconnected...
2019-10-08 20:52:48,831 # Serial port disconnected, waiting to get reconnected...
2019-10-08 20:52:49,832 # Try to reconnect to /dev/ttyACM0 again...
2019-10-08 20:52:49,834 # Reconnected to serial port /dev/ttyACM0
2019-10-08 20:52:55,808 # Serial port disconnected, waiting to get reconnected...
2019-10-08 20:52:56,810 # Try to reconnect to /dev/ttyACM0 again...
2019-10-08 20:52:56,812 # Reconnected to serial port /dev/ttyACM0
2019-10-08 20:52:58,665 # Serial port disconnected, waiting to get reconnected...
2019-10-08 20:52:59,667 # Try to reconnect to /dev/ttyACM0 again...
2019-10-08 20:52:59,669 # Reconnected to serial port /dev/ttyACM0
2019-10-08 20:52:59,674 #  main(): This is RIOT! (Version: 2019.07-devel-2276-g866ec-devel/usbus_msc)
2019-10-08 20:52:59,676 # RIOT USB CDC ACM shell test
> 

```

Restart app multiple time with the reset button on my board only.

ant9000, 2019-10-09:
> [This define](https://github.com/RIOT-OS/RIOT/blob/master/sys/include/usb/usbus/cdc/acm.h#L45) configures the default size for the buffer (must be a power of 2), increasing it to 256 bytes, your example shouldn't lose any characters.

Thank you very much for your explanation: things make a lot more sense, now. I've tried tweaking USBUS_CDC_ACM_STDIO_BUF_SIZE and can confirm that the modified test works fine with it.
 
> I'm thinking of adding a few defines to add a blocking mode and to add a mode where the last `USBUS_CDC_ACM_STDIO_BUF_SIZE` bytes are kept in the buffer (instead of the first [...]

I was thinking that using a ringbuffer in non blocking mode could make more sense for my specific application, but I can imagine there are use cases also for the other behaviours you mention.

When using my own application in verbose mode I can easily flood even a 1024 chars buffer. I could make it even bigger, of course, but in principle the problem remains.

It is actually only limited to the initial transfer, because Linux takes some time to settle down and establish the serial connection - but once connected, everything works fine.

Do you think that keeping a larger ringbuffer until connection and then switching to a smaller one to save RAM could make sense? I guess it would be reasonable to assume that the PC can sustain STDIO_UART_BAUDRATE, once connected.

Again thanks for the explanation - and for the code as well, of course!

ant9000, 2019-10-09:
I have been digging into that a bit more. This change does make things work better:

```
diff --git a/sys/usb/usbus/cdc/acm/cdc_acm_stdio.c b/sys/usb/usbus/cdc/acm/cdc_acm_stdio.c
index c61af0b22..a5e2ed31a 100644
--- a/sys/usb/usbus/cdc/acm/cdc_acm_stdio.c
+++ b/sys/usb/usbus/cdc/acm/cdc_acm_stdio.c
@@ -52,9 +52,12 @@ ssize_t stdio_read(void* buffer, size_t len)
 
 ssize_t stdio_write(const void* buffer, size_t len)
 {
-    usbus_cdc_acm_submit(&cdcacm, buffer, len);
-    usbus_cdc_acm_flush(&cdcacm);
-    /* Use tsrb and flush */
+    size_t n = 0;
+    do {
+        n += usbus_cdc_acm_submit(&cdcacm, buffer, len);
+        usbus_cdc_acm_flush(&cdcacm);
+        /* Use tsrb and flush */
+    } while (n < len);
     return len;
 }
```
Now the buffer gets flushed as expected, but I still have a bug. Here is the output:

```
$ dd if=/dev/ttyACM1
main(): This is RIOT! (Version: 2019.10-devel-1359-gef2d4-samr34)
0 RIOT USB CDC ACM shell test
1 RIOT USB CDC ACM shell test
2 2 RIOT USB CDC ACM shell test
3 RIOT USB CDC ACM shell test
4 RIOT USB CDC ACM shell test
5 RIOT USB CDC ACM shell test
0+7 records in
0+1 records out
248 bytes copied, 1.62583 s, 0.2 kB/s
```
As you can see, the line "2 2 RIOT" shows that the two initial characters "2 " have actually been written out twice.

ant9000, 2019-10-09:
I guess my fix implements precisely what you call blocking mode. Not sure what the correct behaviour should be when the cable is unplugged - discarding everything looks the most sensible behaviour to me. What do you think?

ant9000, 2019-10-11:
Reworked #12402 - it should resolve the problem. Waiting for review, if approved this bug can be closed.

benpicco, 2019-10-19:
If you include a 
> fixes #12349 

in your commit message, the issue will be closed automatically on merge.

miri64, 2019-10-19:
> If you include a
> 
> > fixes #12384
> 
> in your commit message, the issue will be closed automatically on merge.

I think a different issue was meant ;-).

commit d1cc563c (2020-01-15): USBUS CDC ACM STDIO: test for the buffering code

fixes #12384
closed 2020-01-15 by bergzand