RIOT-OS/RIOT #12422: Can't ping wireless nodes with global address

kb2ma, 2019-10-11:
#### Description
Can't ping wireless nodes with global address using gnrc_border_router and gnrc_networking examples

#### Steps to reproduce the issue
As I described in a [comment](https://github.com/RIOT-OS/RIOT/pull/12408#issuecomment-540646993) to PR 12408, I have been unable to setup a wireless network as expected using the gnrc_border_router and gnrc_networking examples. The main issue is that I do not see the "automatic" setup of global addresses, the 2001:db8 network.

Here is what I experience with a pair of samr21-xpro boards, with the border router connected to an Ubuntu laptop.

I start the gnrc_border_router node with `make term`. I see ethos and uhcpd startup, and the command below runs:

   > sudo sh /home/kbee/dev/riot/repo/dist/tools/ethos/start_network.sh  /dev/ttyACM0 tap0 2001:db8::/64

I also start the gnrc_networking node with `make term`.

Returning to the border router node, `ifconfig` shows the wireless and wired interfaces. I do not see a global address, so I add:

   > ifconfig 6 add 2001:db8::1/64

Running `nib route` shows:

   > 2001:db8::/64 dev #6
   > default* via fe80::1 dev #7

Returning to the gnrc_networking node, `ifconfig` does not show a 2001:db8 address. Shouldn't I expect this address to be generated automatically at this point?

From the gnrc_border_router node I can ping the gnrc_networking node using its link local address and the interface specifier (%6).

At any rate, below is what I have to do to create the global address on the gnrc_networking node.

Create the global address:

   > ifconfig 7 add 2001:db8::7b79:6576:137d:5482/64

Running `nib route` shows:

   > 2001:db8::/64 dev #7

At this point I still cannot ping either node using the global addresses. So, on the gnrc_border_router node, I add a route to the gnrc_networking node:

   > nib route add 6 2001:db8::7b79:6576:137d:5482/128 fe80::7b79:6576:137d:5482

Now I can ping the gnrc_networking node. Returning to the gnrc_networking_node, now I also can ping the gnrc_border_router node.

#### Versions
I am running from the newly minted '2019.10-branch'.

miri64, 2019-10-11:
I can't reproduce that _nothing_ works (neither UHCP which provides the BR with a prefix and route, nor the RA which provides the downstream node), but from what I can gather that you did not set a default route. Have you tried restarting the border router? I noticed that for me at least sometimes UHCP is not bootstrapping properly.

miri64, 2019-10-11:
With https://github.com/RIOT-OS/RIOT/pull/12414#issuecomment-541032518 in mind: Might UHCP also not work because of your firewall?

kb2ma, 2019-10-11:
> Might UHCP also not work because of your firewall?

I'll experiment with that. However, when I use RPL, it all works as expected. In fact, I looked at the nib routes in the RPL setup to get the non-RPL use of global addresses to work.

miri64, 2019-10-11:
Over one hop it should work without RPL. RPL also doesn't use UDP, but ICMPv6 and isn't exposed to the host, so the firewall wouldn't make a problem here. UHCP in contrast interacts with the host computer, as it opens a server that communicates with the client on the border router node.

kb2ma, 2019-10-12:
Yes, the firewall was the problem. I set it to allow incoming from the tap interface.

Now when the border router starts up, it automatically gets the 2001:db8::1 address. And when I startup a gnrc_networking node, it also gets an address with this prefix. I am able to ping between wireless nodes, as well as ping between my workstation and the gnrc_networking node.

Thanks, @miri64!

closed 2019-10-12 by kb2ma