RIOT-OS/RIOT #12445: driver/hts221: Temperature and Humidity readings incorrect

MrKevinWeiss, 2019-10-14:
#### Description
While testing the hts221 i2c driver on the `thingy52` I was reading incorrect temperatures and humidity.  This may be something to do with calibration but it was pretty incorrect on what I think the factory values were.  Discovered in #10793.

#### Steps to reproduce the issue
`make BOARD=thingy52 -C tests/driver_hts221 flash term`

#### Expected results
Reading room temperature and reading not desert like humidity.

#### Actual results
```
2019-10-14 10:11:05,685 # Init HTS221 on I2C_DEV(0)
2019-10-14 10:11:05,685 # H: 0.0%, T: 8.9°C
2019-10-14 10:11:05,685 # H: 0.0%, T: 8.2°C
2019-10-14 10:11:07,657 # H: 0.0%, T: 8.2°C
```

#### Versions
<details><summary>print_toolchain_versions.sh</summary>

Operating System Environment
-----------------------------
       Operating System: "Ubuntu" "16.04.6 LTS (Xenial Xerus)"
                 Kernel: Linux 4.15.0-65-generic x86_64 x86_64

Installed compiler toolchains
-----------------------------
             native gcc: gcc (Ubuntu 5.4.0-6ubuntu1~16.04.11) 5.4.0 20160609
      arm-none-eabi-gcc: arm-none-eabi-gcc (GNU Tools for Arm Embedded Processors 7-2017-q4-major) 7.2.1 20170904 (release) [ARM/embedded-7-branch revision 255204]
                avr-gcc: avr-gcc (GCC) 4.9.2
       mips-mti-elf-gcc: missing
             msp430-gcc: missing
   riscv-none-embed-gcc: missing
   xtensa-esp32-elf-gcc: xtensa-esp32-elf-gcc (crosstool-NG crosstool-ng-1.22.0-80-g6c4433a) 5.2.0
   xtensa-lx106-elf-gcc: xtensa-lx106-elf-gcc (crosstool-NG crosstool-ng-1.22.0-60-g37b07f6) 4.8.5
                  clang: clang version 5.0.0-3~16.04.1 (tags/RELEASE_500/final)

Installed compiler libs
-----------------------
   arm-none-eabi-newlib: "2.5.0"
    mips-mti-elf-newlib: missing
riscv-none-embed-newlib: missing
xtensa-esp32-elf-newlib: "2.2.0"
xtensa-lx106-elf-newlib: "2.0.0"
               avr-libc: "1.8.0svn" ("20111229")

Installed development tools
---------------------------
                  cmake: cmake version 3.12.3
               cppcheck: Cppcheck 1.72
                doxygen: 1.8.11
                    git: git version 2.7.4
                   make: GNU Make 4.1
                openocd: GNU MCU Eclipse OpenOCD, 64-bitOpen On-Chip Debugger 0.10.0+dev-00593-g23ad80df4 (2019-04-22-20:18)
                 python: Python 2.7.12
                python2: Python 2.7.12
                python3: Python 3.6.3
                 flake8: 3.6.0 (mccabe: 0.6.1, pycodestyle: 2.4.0, pyflakes: 2.0.0) CPython 3.6.3 on Linux
             coccinelle: missing

</details>

aabadie, 2019-10-14:
> MrKevinWeiss assigned haukepetersen, aabadie and MrKevinWeiss 4 hours ago

I would rather assign @smlng, since he is the original author of the driver :)

smlng, 2019-10-14:
mhm, I wrote this way back I think for some STM sensor board or so where several sensors are on. It did worked, looking at my old PR https://github.com/RIOT-OS/RIOT/pull/7280 its seems that calibration is key here.

isaackc, 2021-03-22:
Hi, was this issue solved? I am also getting the wrong temperatures on my hts221, and have installed multiples of them. In comparing the calibration factors, they are very different from each other.

MrKevinWeiss, 2021-03-23:
@isaackc, I can try to get access to a board tomorrow and do some testing.

MrKevinWeiss, 2021-03-24:
Hmmm, it seems that it is even worst condition than before.  I am not getting any output.  I will look into it further.

crasbe, 2024-07-08:
I happened to find an X-NUCLEO-IKS01A2 board, which features (among other things) an HTS221 sensor.
For the test it is plugged into a NUCLEO-L452RE and it runs `tests/drivers/hts221` on the latest master branch.

First I put it into the hot fan exhaust stream of my ThinkPad, then I hit it with ice spray and put it back into the exhaust stream. 100% humidity was because of the condensation on the PCB after the ice spray.
It is unknown if the chip has added user calibration, just from my feeling I'd say it reads about 2-3 degrees C too hot, but that's within expected margins I guess.

So it works for me...

```
2024-07-08 15:22:24,277 # main(): This is RIOT! (Version: 2024.07-devel-509-ge1e5f)
2024-07-08 15:22:24,279 # Init HTS221 on I2C_DEV(0)
2024-07-08 15:22:24,290 # H: 45.4%, T: 31.5°C
2024-07-08 15:22:26,293 # H: 44.9%, T: 31.3°C
2024-07-08 15:22:28,298 # H: 44.3%, T: 31.2°C
2024-07-08 15:22:30,301 # H: 43.9%, T: 31.2°C
2024-07-08 15:22:32,304 # H: 43.5%, T: 31.1°C
2024-07-08 15:22:34,308 # H: 43.2%, T: 31.2°C
2024-07-08 15:22:36,312 # H: 42.8%, T: 31.2°C
2024-07-08 15:22:38,315 # H: 42.5%, T: 31.2°C
2024-07-08 15:22:40,319 # H: 42.1%, T: 31.1°C
2024-07-08 15:22:42,323 # H: 42.3%, T: 30.9°C
2024-07-08 15:22:44,326 # H: 42.7%, T: 30.6°C
2024-07-08 15:22:46,330 # H: 43.1%, T: 30.0°C
2024-07-08 15:22:48,334 # H: 42.8%, T: 30.0°C
2024-07-08 15:22:50,336 # H: 42.7%, T: 30.0°C
2024-07-08 15:22:52,340 # H: 42.5%, T: 30.0°C
2024-07-08 15:22:54,344 # H: 42.3%, T: 30.0°C
2024-07-08 15:22:56,348 # H: 42.1%, T: 30.0°C
2024-07-08 15:22:58,351 # H: 42.0%, T: 29.9°C
2024-07-08 15:23:00,355 # H: 41.9%, T: 29.9°C
2024-07-08 15:23:02,359 # H: 41.8%, T: 29.9°C
2024-07-08 15:23:04,362 # H: 41.6%, T: 29.8°C
2024-07-08 15:23:06,366 # H: 41.6%, T: 29.8°C
2024-07-08 15:23:08,370 # H: 41.6%, T: 29.7°C
2024-07-08 15:23:10,373 # H: 41.5%, T: 29.7°C
2024-07-08 15:23:12,377 # H: 41.4%, T: 29.7°C
2024-07-08 15:23:14,381 # H: 41.4%, T: 29.6°C
2024-07-08 15:23:16,384 # H: 41.3%, T: 29.3°C
2024-07-08 15:23:18,388 # H: 41.2%, T: 23.4°C
2024-07-08 15:23:20,392 # H: 42.0%, T: 23.7°C
2024-07-08 15:23:22,395 # H: 41.8%, T: 19.1°C
2024-07-08 15:23:24,399 # H: 42.8%, T: 20.0°C
2024-07-08 15:23:26,402 # H: 43.8%, T: 20.3°C
2024-07-08 15:23:28,406 # H: 44.4%, T: 20.5°C
2024-07-08 15:23:30,410 # H: 45.1%, T: 20.6°C
2024-07-08 15:23:32,414 # H: 45.8%, T: 20.7°C
2024-07-08 15:23:34,417 # H: 45.7%, T: 16.2°C
2024-07-08 15:23:36,420 # H: 49.0%, T: 12.2°C
2024-07-08 15:23:38,424 # H: 48.5%, T: 12.0°C
2024-07-08 15:23:40,428 # H: 51.0%, T: 12.9°C
2024-07-08 15:23:42,432 # H: 52.5%, T: 14.1°C
2024-07-08 15:23:44,435 # H: 53.3%, T: 14.6°C
2024-07-08 15:23:46,439 # H: 53.5%, T: 13.8°C
2024-07-08 15:23:48,443 # H: 54.2%, T: 15.4°C
2024-07-08 15:23:50,446 # H: 55.0%, T: 15.6°C
2024-07-08 15:23:52,450 # H: 55.7%, T: 15.8°C
2024-07-08 15:23:54,453 # H: 56.3%, T: 16.7°C
2024-07-08 15:23:56,456 # H: 56.3%, T: 14.2°C
2024-07-08 15:23:58,461 # H: 57.0%, T: 15.9°C
2024-07-08 15:24:00,464 # H: 57.5%, T: 17.1°C
2024-07-08 15:24:02,468 # H: 57.7%, T: 16.3°C
2024-07-08 15:24:04,471 # H: 58.4%, T: 16.5°C
2024-07-08 15:24:06,475 # H: 58.1%, T: 13.3°C
2024-07-08 15:24:08,479 # H: 58.0%, T: 11.7°C
2024-07-08 15:24:10,482 # H: 59.0%, T: 13.5°C
2024-07-08 15:24:12,486 # H: 59.1%, T: 8.7°C
2024-07-08 15:24:14,490 # H: 100.0%, T: -44.2°C
2024-07-08 15:24:16,494 # H: 100.0%, T: -39.5°C
2024-07-08 15:24:18,497 # H: 74.1%, T: -24.3°C
2024-07-08 15:24:20,501 # H: 74.6%, T: -17.1°C
[... 50 lines cut ...]
2024-07-08 15:26:02,688 # H: 64.0%, T: 27.7°C
2024-07-08 15:26:04,692 # H: 62.6%, T: 27.9°C
2024-07-08 15:26:06,696 # H: 61.4%, T: 28.1°C
2024-07-08 15:26:08,700 # H: 60.2%, T: 28.3°C
2024-07-08 15:26:10,703 # H: 59.2%, T: 28.6°C
2024-07-08 15:26:12,707 # H: 58.2%, T: 28.9°C
2024-07-08 15:26:14,711 # H: 57.2%, T: 29.1°C
2024-07-08 15:26:16,714 # H: 56.2%, T: 29.3°C
2024-07-08 15:26:18,717 # H: 55.4%, T: 29.5°C
2024-07-08 15:26:20,722 # H: 54.6%, T: 29.7°C
2024-07-08 15:26:22,724 # H: 53.9%, T: 29.9°C
2024-07-08 15:26:24,729 # H: 53.1%, T: 30.2°C
2024-07-08 15:26:26,733 # H: 52.5%, T: 30.4°C
2024-07-08 15:26:28,736 # H: 51.7%, T: 30.6°C
2024-07-08 15:26:30,739 # H: 51.1%, T: 30.8°C
2024-07-08 15:26:32,744 # H: 50.5%, T: 31.0°C
2024-07-08 15:26:34,748 # H: 50.0%, T: 31.0°C
2024-07-08 15:26:36,750 # H: 49.4%, T: 31.2°C
2024-07-08 15:26:38,754 # H: 48.8%, T: 31.4°C
2024-07-08 15:26:40,758 # H: 48.4%, T: 31.5°C
2024-07-08 15:26:42,761 # H: 48.1%, T: 31.6°C
2024-07-08 15:26:44,765 # H: 47.6%, T: 31.7°C
2024-07-08 15:26:46,769 # H: 47.3%, T: 31.9°C
2024-07-08 15:26:48,772 # H: 46.8%, T: 32.1°C
```

crasbe, 2024-07-09:
But the HTS221 is obsolete anyway now and no longer for sale.. https://www.st.com/en/mems-and-sensors/hts221.html

closed 2024-07-23 by MrKevinWeiss