RIOT-OS/RIOT #12448: tests: run tests on all boards.

fjmolinas, 2019-10-14:
## Description

For testing to be performed it must be easy. `/dist/toolscompile_and_test_for_board` was a stepping stone towards this. 

Many boards can now run all tests in a row reliably. But there exception for boards that:

- `make reset` does not work
    e.g.: `arduino`, `native`, `z1` (all passing by a BSL)
- `make reset` blocks
   e.g.: `esp32`
- `make reset` causes serial disconnect
    e.g.: `atmega2560, mkrwan`
- `make flash` fails without manual intervention
    e.g.: `ccs2538dk`,
- port with no `stdin` or `stdout`
   e.g: all `mips`

This issue wants to track the needed efforts to be able to achieve this. The following list is not exhaustive at it only involves the issues I'm aware off

## Steps

- `make reset` does not work
   - [ ] use `test_interactive_sync_util` in all possible test, i.e. test not disabling input/output
   - [ ] test application should run in a loop for tests that cant use test_interactive_sync_util
- `make reset` causes serial disconnect
    - [ ] use sync functions and disable reset on term
    - [ ] wait for terminal to become available
    - [ ] do not reset
- `make reset` blocks
    - [ ] not reset when terminal is open
- `make flash` fails without manual intervention
    - [ ] use usb-bootloader 
    - [ ] blacklist tests that cause hardfaults (`tests/ssp`)

miri64, 2019-10-14:
>     * `make reset` does not work

I wonder how we would do this on `native`. It is effectively a process running on the host system, so we somehow need to restart that without effecting `make term`.

miri64, 2019-10-14:
The other question: is it necessary for `native`?

kaspar030, 2019-10-14:
> I wonder how we would do this on `native`. It is effectively a process running on the host system, so we somehow need to restart that without effecting `make term`.

We could trigger native's "reboot" (which does a self-exec) via signal. For that, there needs to be a way to export the pid, maybe through a CLI argument setting a pid filename...

miri64, 2019-10-14:
Signaling and rebootstrapping (not necessarilly via `reboot`) was also my thought. But saving the PID, especially with multiple `native` sessions (which especially considering some of our networking test cases is desirable), seems like a challenge.

kaspar030, 2019-10-14:
> But saving the PID, especially with multiple `native` sessions (which especially considering some of our networking test cases is desirable), seems like a challenge.

How are multi-node native tests usually run?

miri64, 2019-10-14:
On multiple terminals, currently. But having some kind of automation integration should be possible for future use cases.

miri64, 2019-10-14:
Have a look at https://github.com/RIOT-OS/Release-Specs/pull/79 for specifics e. g.

miri64, 2019-10-20:
See #12517 for a PoC for `native` reset.

MrKevinWeiss, 2019-11-05:
We should also try to get rid of the unneeded sleeps, with the sync tests should run much faster.

fjmolinas, 2019-12-03:
> The other question: is it necessary for native?

Coming back to this it isn't necessary, but make native resemble more to normal `BOARD` which also means the test running on native will be much more like the one running on a physical `BOARD`.

> >     * `make reset` does not work
> 
> I wonder how we would do this on `native`. It is effectively a process running on the host system, so we somehow need to restart that without effecting `make term`.

But `make reset` will still not work on native even while there isn't a process running (previous `make term`), which is not the case for a "physical" board where I can `reset`, and the call `term`. This difference needs to be handled.

fjmolinas, 2019-12-03:
> But make reset will still not work on native even while there isn't a process running (previous make term), which is not the case for a "physical" board where I can reset, and the call term. This difference needs to be handled.

Thinking of it better, it would be the same as for `z1`, where it just does nothing.

MrKevinWeiss, 2019-12-03:
> * `make reset` blocks
>   e.g.: `esp32`

#12752 I think fixed that.  Is it still the case?

aabadie, 2021-05-27:
With test-on-ryot and test-on-iotlab github actions being merged, can we consider that this issue is completed ?