RIOT-OS/RIOT #12540: STM32F4:  nucleo-f401re: printf carriege return issue

Ciusss89, 2019-10-22:
#### Description
`printf` has bad aligiment, seems that the carriage return `\r` is missing.

#### Steps to reproduce the issue
Compile the following code con the current board:

Hardware: nucleo-f401re
Code:
1. Makefile:
```
~/mcu/_riot_app $ cat Makefile 
# name of your application
APPLICATION = first_app

# If no BOARD is found in the environment, use this default:
BOARD ?= nucleo-f401re

# This has to be the absolute path to the RIOT base directory:
RIOTBASE ?= $(CURDIR)/../RIOT/

# Comment this out to disable code in RIOT that does safety checking
# which is not needed in a production environment but helps in the
# development process:
CFLAGS += -DDEVELHELP

# Change this to 0 show compiler invocation lines by default:
QUIET ?= 1

# Modules to include:
USEMODULE += shell
USEMODULE += shell_commands
USEMODULE += ps
```

2. main:
```
~/mcu/_riot_app $ cat main.c 
#include <stdio.h>
#include <string.h>

#include "shell.h"

int main(void)
{
    printf("RIOT on a(n) %s board, MCU %s\n\r", RIOT_BOARD, RIOT_MCU);

    char line_buf[SHELL_DEFAULT_BUFSIZE];
    shell_run(NULL, line_buf, SHELL_DEFAULT_BUFSIZE);

    return 0;
}
```
 
#### Expected results

#### Actual results
```
> reboot
�main(): This is RIOT! (Version: 2020.01-devel-16-gd9684-2019.10-branch)
                                                                        RIOT on a(n) nucleo-f401re board, MCU stm32f4
> ps
	pid | name                 | state    Q | pri | stack  ( used) | base addr  | current     
                                                                                                  	  - | isr_stack            | -        - |   - |    512 (  116) | 0x20000000 | 0x200001c8
                                                                                                                                                                                                	  1 | idle                 | pending  Q |  15 |    256 (  128) | 0x200003cc | 0x2000044c 
                                                   	  2 | main                 | running  Q |   7 |   1536 (  632) | 0x200004cc | 0x2000093c 
                                                                                                                                                 	    | SUM                  |            |     |   2304 (  876)
                                                                                                                                                                                                                      > 
> 
         
```

#### Versions
`2020.01-devel-16-gd9684-2019.10-branch`

kaspar030, 2019-10-22:
> `printf` has bad aligiment, seems that the carriage return `\r` is missing.

This is intentional, RIOT sends "\n" for newlines, or better, it doesn't interfere with what the application writes to the serial port.

How do you access the serial port on the host side?

kaspar030, 2019-10-24:
> To connect the shell, I use `picocom -b 115200 /dev/ttyACM0`

I use picocom, too. :wink:
You'll need to add `--imap lfcrlf`.

> 
> Ok, but that can make difficult the development, I don't understand if:
> 
>     1. that is not a RIOT's issue, but it's an issue of the stm32f4's driver (uart?).
> 
>     2. if that is a limit of the current RIOT implementation.

Well, actually just sending LF is pretty standard for serial terminals.

> btw the output of `ps` command is not readable

I guess that's a seperate issue? Can you provide some output?

Ciusss89, 2019-10-24:
> You'll need to add --imap lfcrlf

The output adding `--imap` setup:

```
> reboot
�main(): This is RIOT! (Version: 2020.01-devel-16-gd9684-2019.10-branch)
RIOT on a(n) nucleo-f401re board, MCU stm32f4
> help
Command              Description
---------------------------------------
reboot               Reboot the node
ps                   Prints information about running threads.
> 	        
> ps
	pid | name                 | state    Q | pri | stack  ( used) | base addr  | current     
	  - | isr_stack            | -        - |   - |    512 (  116) | 0x20000000 | 0x200001c8
	  1 | idle                 | pending  Q |  15 |    256 (  128) | 0x200003cc | 0x2000044c 
	  2 | main                 | running  Q |   7 |   1536 (  632) | 0x200004cc | 0x2000093c 
	    | SUM                  |            |     |   2304 (  876)
> 
```

The issue has been fixed.  :beer: :beer: 
Thanks

kaspar030, 2019-10-24:
> The issue has been fixed. beer beer

Awesome. Have fun with RIOT!

closed 2019-10-24 by kaspar030