RIOT-OS/RIOT #12545: No answers to Neighbor Solicitation when 6lo and non-6lo interfaces are mixed

chrysn, 2019-10-22:
In setups that have both a 6lo and a non-6lo interface mixed, the RIOT device does not answer incoming Neighbor Solicitation messages on the non-6lo device unless it is configured as a border router. Such a setup can come up when USB CDC-ECM is added to any 6lo-enabled device.

### Reproducing

In the gcoap example modified to include a `CFLAGS += -DGNRC_IPV6_NIB_CONF_SLAAC` line as per the warning from #12513, run

    $ GNRC_NETIF_NUMOF=2 USEMODULE="" TERMFLAGS="tap0" make all term

and try pinging from the host -- in my setup (where several tap devices are bridged together) that's

    $ ping fe80::18a9:39ff:fe2b:f3ff%tapbr0

which succeedes, and

    $ ping ff02::1%tapbr0

(which is a handy reference both for "networking is actually up" and for copy-pasting out the IP address) shows responses both from the host and the device.

When adding a 6lo interface to the mix like this:

    $ GNRC_NETIF_NUMOF=2 USEMODULE="socket_zep" TERMFLAGS="-z [::]:17755 tap0" make all term

(nothing needs to actually run on :17755 for the effect to be visible), the multicast pings will still show responses from both the device and the host, but the unicasts won't. (You may need to take the interface down and up again *after starting the second program* -- see below for what's odd here).

Only when enabling the border router module:

    $ GNRC_NETIF_NUMOF=2 USEMODULE="socket_zep gnrc_sixlowpan_border_router_default" TERMFLAGS="-z [::]:17755 tap0" make all term

the unicast address becomes reachable again.

#### Oddities

When the interface was taken down and up again between tests (ie. before the second program runs), the unicast pings succeed. The information about the link-layer address of the RIOT device seems to be transported in the Router Solicitation message it sends when coming up -- the host remembers that address. That was only observed with native tap-based setup, the USB-ECM setup (below) behaved more consistently in it always ran into soliciting addresses and not getting a response.

#### Other setups

I've originally observed this on an nrf52840-dongle that uses the usbus_cdc_ecm and nrf802154 modules -- same symptoms.

### Observations on the wire

In the working scenarios, Neighbor Solicitations for the device's address are answered promptly with solicited target link-layer address messages. In the non-working scenario, those stay unanswered.

### Workarounds & directions

For the time being, enabling the border router is a viable option in those scenarios to keep testing. A more desirable outcome of this issue would be to have a settable flag with a warning like that from #12513, or even to create suitable configurations out of the box.

It may be that this scenario is exotic enough to not warrant dedicated support (and #12513, which I think targets just those cases, was just easy enough to fix). If so, that should be made explicit, possibly right in the SLAAC warning message (but preferably at build time).

I'll dig around a bit, maybe understanding the nib.c debug messages will get me somewhere.

chrysn, 2019-10-22:
The NS messages that are being ignored are sent to ff02::1:tail-of:address (the "solicited node addresses"). In the working setups, that group is joined as indicated in `ifconfig` (in the BR case, one each is joined on both interfaces), in the two-interfaces-but-not-BR case it is not.

That being found, I'm looking on to when and why they are joined. (It's not for lack of GNRC_NETIF_IPV6_GROUPS_NUMOF space.)

chrysn, 2019-10-22:
Defining `GNRC_IPV6_NIB_CONF_ARSM` is a sufficient workaround, functionality-wise. I suppose the criterion for applying this workaround should be when (and only when) the #12513 override is recommended.

From the user experience point of view, overriding those is not ideal. The need to override stems from the way the top part of sys/include/net/gnrc/ipv6/nib/conf.h is structured: It checks for whether the 6LBR, 6LR or 6LN modules are active and changes the later defaults. I think an approach where the individual features are pulled in by pseudomodules that indicate them, but not pushed out due to their presence, would be preferable. Could we have a gnrc_ipv6_nib_non6lo ("traditional"? "wired"?) pseudomodule pulled in by all non-6lo netdevs? The presence of that could then enable SLAAC and ARSM, and in absence of such an interface they could stay off.

miri64, 2019-10-22:
Activating the address resolution state-machine (ARSM) really seems like more of a workaround to me. An NS should always be answered, may there be a state machine to hold the state provided by the NS or not. Because of that I'm against implementing a dedicated pseudo-module, as this is most definitely a bug.

chrysn, 2019-10-22:
If there's a need for some devices to join the solicited node groups (or might they not need joining at all if all NS are answered?), I figure there'll still be some distinction to be made at build time on whether or not to add that joining code -- but sure, if ARSM is not sharp enough a workaround yet, let's try to nail that down first.

Two things I've observed trying to understand the role of ARSM:
* If I define `GNRC_IPV6_NIB_CONF_ARSM=0` on a single-role ethernet board (like plain native), the groups are not joined either, and NS to those addresses are not answered. (As above, it takes taking down and up again the interface in my tap/bridge setup).
* Running through invocations of `ipv6_addr_set_solicited_nodes` (which calculates a solicited node address from any other address), only the `#if GNRC_IPV6_NIB_CONF_ARSM` code in gnrc_netif.c appears to do that in conjunction with joining groups; all other uses are with sending soliciations.

miri64, 2019-10-22:
Ahhh so the NS are _not_ unicast, I somehow thought they were... In this case yes activating ARSM solves the problem, because only then the node joins the multicast group, as per [RFC 6775](https://tools.ietf.org/html/rfc6775#section-5.2), more because the state-machine is not assumed.

https://github.com/RIOT-OS/RIOT/blob/master/sys/net/gnrc/netif/gnrc_netif.c#L602-L616

Maybe another IPv6 option JOIN_SOL_NODES is necessary to better reflect RFC 6775, section 5.2 is needed?

miri64, 2019-10-22:
Ah wait lets go one step backwards. It's the ethernet interface, that isn't pingable, right? The 6LoWPAN-side still would join a LoWPAN as a 6LN, not a border router.

chrysn, 2019-10-22:
Yes, the ethernet interface isn't pingable. The 6lo side probably works, I did not try much there.

miri64, 2019-10-22:
Ok, yeah then the ARSM needs to be activated by hand. ^^"

miri64, 2019-10-22:
> #### Oddities
>
> When the interface was taken down and up again between tests (ie. before the second program runs), the unicast pings succeed. The information about the link-layer address of the RIOT [...]

The unicast/multicast thing explains also that: as per RFC 4861, both Linux and RIOT store the source address of an RS as STALE (if a SLLAO is provided with the RS), with that then follow-up NS to unstale the address are sent unicast.

miri64, 2019-10-22:
See #12547 for the addition of a warning #12513-style ;-)

chrysn, 2019-10-23:
Thanks, #12547 takes the pressure off this to the point it can be closed.

From the fix I also see that it's not as easy with the pseudomodules (as there is frequent distinction between 6ln/6lr/6lbr, and also because there's the loopback interface), so that couldn't be a sharp criterion but a hint at "there are non-6lo interfaces expected to be around" at best.

closed 2019-10-24 by cgundogan