RIOT-OS/RIOT #12565: gnrc_icmpv6_echo: flood-pinging another node leads to leaks in own packet buffer

miri64, 2019-10-24:
#### Description
When flood-pinging another node, some received echo replies might get stuck in the packet buffer

#### Steps to reproduce the issue
Start two native instances of `tests/gnrc_udp` (for `gnrc_pktbuf_cmd`) and flood-ping one from the other. See https://github.com/RIOT-OS/Release-Specs/issues/142#issuecomment-546019666 for a detailed analysis

```
$ PORT=tap0 make -C tests/gnrc_udp all term
> ifconfig
--- different terminal
$ PORT=tap1 make -C tests/gnrc_udp all term
ping6 -s 1452 -i 0 -c 500000 '<link-local address of tap0 instance>"
```

#### Expected results

The packet buffer should be empty on both nodes afterwards.

#### Actual results
<details><summary>The packet buffer on the <code>tap1</code> instance is filled with echo replies</summary>

```
> pktbuf
pktbuf
packet buffer: first byte: 0x565b9d60, last byte: 0x565bbd60 (size: 8192)
  position of last byte used: 8000
=========== chunk   0 (0x565b9d60 size:   96) ===========
00000000  A8  9D  5B  56  E8  AF  5B  56  B4  05  00  00  01  00  00  00
00000010  03  00  00  00  00  00  00  00  00  00  00  00  90  9D  5B  56
00000020  14  00  00  00  01  00  00  00  FF  FF  FF  FF  3A  40  FE  80
00000030  06  06  05  00  00  00  00  00  5A  ED  47  3B  43  7F  4E  58
00000040  F0  A5  07  5E  00  00  4C  58  78  9D  5B  56  C0  AF  5B  56
00000050  28  00  00  00  01  00  00  00  01  00  00  00  53  53  53  53
~ unused: 0x565b9dc0 (next: 0x565ba3a0, size: 1456) ~
=========== chunk   1 (0x565ba370 size:   48) ===========
00000000  88  A3  5B  56  08  AA  5B  56  B4  05  00  00  01  00  00  00
00000010  03  00  00  00  53  53  53  53  B0  A3  5B  56  E0  A9  5B  56
00000020  28  00  00  00  01  00  00  00  01  00  00  00  3A  40  FE  80
~ unused: 0x565ba3a0 (next: 0x565ba3c8, size:   16) ~
=========== chunk   2 (0x565ba3b0 size:   24) ===========
00000000  00  00  00  00  F0  A3  5B  56  14  00  00  00  01  00  00  00
00000010  FF  FF  FF  FF  00  00  00  00
~ unused: 0x565ba3c8 (next: 0x565ba468, size:   40) ~
=========== chunk   3 (0x565ba3f0 size:  120) ===========
00000000  06  06  05  00  00  00  00  00  5A  ED  47  3B  43  7F  4E  58
00000010  F0  A5  07  5E  3A  40  FE  80  20  A4  5B  56  D8  B5  5B  56
00000020  B4  05  00  00  01  00  00  00  03  00  00  00  00  00  4C  58
00000030  38  A4  5B  56  B0  B5  5B  56  28  00  00  00  01  00  00  00
00000040  01  00  00  00  53  53  53  53  00  00  00  00  50  A4  5B  56
00000050  14  00  00  00  01  00  00  00  FF  FF  FF  FF  53  53  53  53
00000060  06  06  05  00  00  00  00  00  5A  ED  47  3B  43  7F  4E  58
00000070  F0  A5  07  5E  53  53  53  53
~ unused: 0x565ba468 (next: 0x565bb5a0, size: 1400) ~
=========== chunk   4 (0x565ba9e0 size: 3008) ===========
00000000  60  00  00  00  05  B4  3A  40  FE  80  00  00  00  00  00  00
00000010  58  ED  47  FF  FE  3B  43  7F  FE  80  00  00  00  00  00  00
00000020  4C  58  F0  FF  FE  A5  07  5E  81  00  C3  8A  F1  FD  26  DC
00000030  92  E7  3F  17  53  53  53  53  53  53  53  53  53  53  53  53
00000040  53  53  53  53  53  53  53  53  53  53  53  53  53  53  53  53
00000050  53  53  53  53  53  53  53  53  53  53  53  53  53  53  53  53
00000060  53  53  53  53  53  53  53  53  53  53  53  53  53  53  53  53
00000070  53  53  53  53  53  53  53  53  53  53  53  53  53  53  53  53
00000080  53  53  53  53  53  53  53  53  53  53  53  53  53  53  53  53
00000090  53  53  53  53  53  53  53  53  53  53  53  53  53  53  53  53
000000A0  53  53  53  53  53  53  53  53  53  53  53  53  53  53  53  53
000000B0  53  53  53  53  53  53  53  53  53  53  53  53  53  53  53  53
000000C0  53  53  53  53  53  53  53  53  53  53  53  53  53  53  53  53
000000D0  53  53  53  53  53  53  53  53  53  53  53  53  53  53  53  53
000000E0  53  53  53  53  53  53  53  53  53  53  53  53  53  53  53  53
000000F0  53  53  53  53  53  53  53  53  53  53  53  53  53  53  53  53
[... 250 lines cut ...]
000004C0  53  53  53  53  53  53  53  53  53  53  53  53  53  53  53  53
000004D0  53  53  53  53  53  53  53  53  53  53  53  53  53  53  53  53
000004E0  53  53  53  53  53  53  53  53  53  53  53  53  53  53  53  53
000004F0  53  53  53  53  53  53  53  53  53  53  53  53  53  53  53  53
00000500  53  53  53  53  53  53  53  53  53  53  53  53  53  53  53  53
00000510  53  53  53  53  53  53  53  53  53  53  53  53  53  53  53  53
00000520  53  53  53  53  53  53  53  53  53  53  53  53  53  53  53  53
00000530  53  53  53  53  53  53  53  53  53  53  53  53  53  53  53  53
00000540  53  53  53  53  53  53  53  53  53  53  53  53  53  53  53  53
00000550  53  53  53  53  53  53  53  53  53  53  53  53  53  53  53  53
00000560  53  53  53  53  53  53  53  53  53  53  53  53  53  53  53  53
00000570  53  53  53  53  53  53  53  53  53  53  53  53  53  53  53  53
00000580  53  53  53  53  53  53  53  53  53  53  53  53  53  53  53  53
00000590  53  53  53  53  53  53  53  53  53  53  53  53  53  53  53  53
000005A0  53  53  53  53  53  53  53  53  53  53  53  53  53  53  53  53
000005B0  53  53  53  53  53  53  53  53  53  53  53  53  53  53  53  53
000005C0  53  53  53  53  53  53  53  53  53  53  53  53  53  53  53  53
000005D0  53  53  53  53  53  53  53  53  53  53  53  53  53  53  53  53
~ unused: 0x565bbb90 (next: (nil), size:  464) ~
```

</summary>

#### Versions
Tested with `2019.10-devel`, `2019.10-RC2`, and https://github.com/RIOT-OS/RIOT/commit/45ec7668710de1f282cd1ceca8463357428ec5ae (the commit that introduced the new `ping6` implementation. All show this behavior, so I believe this is also behaving like this on master.

miri64, 2019-10-24:
BTW: when flood pinging from a `native` node to the Linux host, I was reliably able to freeze up my Linux machine if I were other things on the side (did not test without doing anything) :flushed:

fjmolinas, 2020-01-10:
@miri64 have there been any updates on this issue?

miri64, 2020-01-10:
No.

Chamaeleon-, 2020-06-30:
Interesting side effect:
This bug seams to produce an underflow on packet loss possibly due to the leftover packets in the buffer.
```
2020-06-30 15:41:31,042 # ping6 ff02::1 -c 5
[...]
2020-06-30 15:41:34,069 # --- ff02::1 PING statistics ---
2020-06-30 15:41:34,069 # 4 packets transmitted, 5 packets received, 59 duplicates, 1073741799% packet loss
2020-06-30 15:41:34,070 # round-trip min/avg/max = 0.987/440.742/9363.049 ms
```
(3rd ping of this kind in a row in a mesh with 25 native nodes made by desvirt)

fjmolinas, 2020-06-30:
> Interesting side effect:
> This bug seams to produce an underflow on packet loss possibly due to the leftover packets in the buffer.
>
> ```
> 2020-06-30 15:41:31,042 # ping6 ff02::1 -c 5
> [...]
> 2020-06-30 [...]

That seems like an easier way of replicating the bug as well.

Chamaeleon-, 2020-06-30:
The precise network setup is made with desvirts topologycreator running on a Ubuntu 18.04 LTS Notebook.
`./topology_creator -s 25 -t mesh -n riot_native -e ./gnrc_networking.elf`
Seams to me like it triggers the bug every time?

miri64, 2020-07-16:
After some more debugging I was able to discern that the packets getting lost this way are stuck in the neighbor address resolution packet queue and get stuck there because of the neighbor fails this check every time somehow.

https://github.com/RIOT-OS/RIOT/blob/f722022708bfcb8054daea262f6550aab073c9e1/sys/net/gnrc/network_layer/ipv6/nib/_nib-arsm.c#L278-L283

Replacing the `>` with a `>=` improves the situation, but in rare cases still packets get stuck + I don't believe this is the correct solution.

Teufelchen1, 2025-07-23:
Is this still an issue? I replicated miri64s instructions but could not reproduce the erroneous behavior.