RIOT-OS/RIOT #12578: ATmega256RFR2 / ATmega128RFA1: `periph/cpu_id` provided without HW support

maribu, 2019-10-26:
#### Description

##### TL;DR

The ATmega MCUs have no unique hardware identifier. Still, we have a `periph/cpu_id` implementation for ATmega256RFR2 and ATmega128RFA1 claiming to provide one.

##### In Detail

The ATmega128RFA1 and ATmega256RFR2 do have a manufacturer ID, a part number, a JTAG ID etc. Out of these a CPU ID is generated that differs at least between different ATmega MCU product lines. But any MCU of the same product line (e.g. all ATmega256RFR2 of the same revision) will have the exact same values here. In addition a region of the flash is dedicated as "User Signature Data", which can only be accessed via JTAG or High Voltage Parallel Programming. It contains no data and therefore cannot be used to uniquely identify the chip, unless the user writes unique IDs there. The current implementation of `periph/cpu_id` uses both manufacturer ID and part numbers as well as the user signature for the CPU ID. Without the user explicitly writing a (locally) unique user signature via JTAG or parallel programming, the CPU ID is therefore the very same for every MCU of the same model and revision.

To me, `periph/*` are features that the MCU provides. Not something that is added later on to the MCU. With the current implementation, the unique ID is clearly added later on, so it should not be a `periph/cpu_id` in my opinion. The general approach to let the user supply a unique ID later on stored at a special location could also be implemented as a more generic driver. E.g. an eeprom backed flavor of this could be used very generically.

#### Steps to reproduce the issue

Get two (or more) ATmega256RFR2 and/or ATmega128RFA1 boards, flash and run `tests/periph_cpuid`.

#### Expected results

The CPU IDs differ

#### Actual results

The CPU IDs on the same boards are completely identical.

#### Versions

Since support for the MCUs were added.

maribu, 2019-10-26:
My conclusion would be to simple remove the `periph/cpu_id` implementation for the ATmega MCUs.

benpicco, 2019-10-26:
I wonder if the [RC Oscillator Calibration Byte](http://ww1.microchip.com/downloads/en/DeviceDoc/Atmel-8266-MCU_Wireless-ATmega128RFA1_Datasheet.pdf#page=463) could serve as a CPU ID, but I only have one ATmega128RFA1.

For me, `usr_sign_0` and `usr_sign_1` are always equal to `signature_0` and `signature_1`.

Is this the expected behavior? How was this supposed to be used @Josar?

maribu, 2019-10-30:
On first Microduino CoreRF:
`CPUID: 0x1e 0xa7 0x01 0x1f 0x83 0x04 0x1e 0xa7 0xb5`

On second one:
`CPUID: 0x1e 0xa7 0x01 0x1f 0x83 0x04 0x1e 0xa7 0xa9`

So it does indeed increase the hamming distance of CPU-IDs and makes life with L2 addresses generated via `luid_get()` easier, if no user signature is added by the user.

maribu, 2019-10-30:
See https://github.com/RIOT-OS/RIOT/pull/12609

maribu, 2019-11-11:
Fixed with https://github.com/RIOT-OS/RIOT/pull/12617

closed 2019-11-11 by maribu