RIOT-OS/RIOT #12621: Potential race condition in compile_and_test_for_board.py

OYTIS, 2019-10-31:
While working on https://github.com/RIOT-OS/RIOT/pull/9323 I've run into numerous failures of `compile_and_test_for_board.py`, that seemed to be caused by the board starting before the terminal on the host machine starts. While the test expected to see certain output from the board, it was sometimes missing, and sometimes the beginning was truncated, while flashing an test manually would show correct behaviour.

The error can be reproduced at least with cc2650-launchpad and cc1352-launchpad. I suspect that any board using XDS110 may be affected as well. Probably hardening the test runner so that the terminal always starts before the image is flashed should help.

aabadie, 2020-07-02:
@fjmolinas I think the work done with `test_utils_interactive_sync` improved the situation here. @OYTIS maybe you could add a delay before the test actually starts. This is what is done with board where stdio is provided via USB.

https://github.com/RIOT-OS/RIOT/blob/4c5cf2b1bb5682948c7eb5787f65a6cf2a2a9209/boards/adafruit-clue/Makefile.include#L25-L26

You can define the `TERM_DELAY` with an integer in second.

fjmolinas, 2020-07-15:
> @fjmolinas I think the work done with `test_utils_interactive_sync` improved the situation here. @OYTIS maybe you could add a delay before the test actually starts.

I think this is indeed fixed for most cases, it might still happen for some boards with specific issues, but if that is the case we should open specific issues. I currently have a cc2650-launchpad at hand, I'll run `compile_and_test_for_boards` and see if its fixed for this board.