RIOT-OS/RIOT #12631: boards/hifive1b: SiFive HiFive1 rev B no I2C & SPI

mrusme, 2019-11-02:
#### Description
Even though (according to the specs) the HiFive1 revB does provide I2C and SPI, they don't seem to be implemented and enabled yet in RIOT.

### Useful links
- [Specs](https://www.sifive.com/boards/hifive1-rev-b)
- [Zephyr repo](https://github.com/zephyrproject-rtos/zephyr/tree/master/boards/riscv/hifive1_revb)

mrusme, 2019-11-02:
@kenrabold do you happen to know about the implementation of I2C and SPI on the HiFive? Maybe you can help out here quickly? :-)

kenrabold, 2019-11-02:
I actually have a PR ready for SPI and PWM. I just need to test it out a little more.
I'm also working on I2C, but that will take a little longer.
So, those drivers are coming soon

mrusme, 2019-11-03:
@kenrabold need help on testing? If so, point me to the PR and I can test it as well. SPI is something I could use right away. Could also test PWM with a couple of motors.

kenrabold, 2019-11-03:
I tested the PWM driver with the built in tri-color LED on the HiFive 1B, so I'm pretty confident that works. I also started getting the SPI interface to the ESP32-SOLO WiFi module on the HiFive1 1B going with commands and responses, but I don't have the complete handshake working yet. I wanted to connect up a different SPI module (RF reader), but haven't had the time to implement and fully test it out.

PWM and SPI driver PR:  https://github.com/kenrabold/RIOT/tree/spi

closed 2020-01-22 by fjmolinas