RIOT-OS/RIOT #12636: nina b112 - hex file

tcpipchip, 2019-11-03:
Hi, one question

I compiled the HELLO-WORLD example to NRF52832-DK and i created a HEX file! I used VIRTUAL BOX (UBUNTU) but i am having problems to burn the HEX (VIRTUAL BOX didnt recognize the SEGGER).

Then i want to burn on WINDOWS with SEGGER.

Must i burn the BOOTLOADER too ?

How the instruction to run it ? I burn only the HEX with j-LINK

[image]

dylad, 2019-11-04:
Hello @tcpipchip 
First of all,how do you generate the .hex file ? RIOT generates .bin and .elf 
Second question, what is the bootloader you are talking about ? You're only suppose to flash the generated firmware by RIOT
Finally, could you try to update your J-Flash software ? 6.20i is two years old already and newer software contains some fixes for nrf52.

tcpipchip, 2019-11-04:
@dylad 
Hello! First of all! Thanks to anwer the question.
Well, to compile i executed
make BOARD=NRF52832-DK and an "elf" and "hex" file were created. I cant burn because i had problems with VIRTUAL BOX that cant recognized my ST-V2 LINK and J-LINK SEGGER, i dont know why!
Then i uploaded the hex the file to the cloud to download it to my Windows real machine and burn it with J-FLASH. I changed the TXD and RXD to be compatible with the U-BLOX NINA B112 module.
Looks that nothing happens, no HELLO WORLD on TXD.

I supposed that the HEX file must to be burn on 00000000H address, right ?

I cant see that bin file, just hex file created. Does it creates d BIN ?

"Finally, could you try to update your J-Flash software ? 6.20i is two years old already and newer software contains some fixes for nrf52."
-Wow, i will do that!

dylad, 2019-11-05:
> I cant see that bin file, just hex file created. Does it creates d BIN ?

By default, make BOARD=nrf52dk -C examples/hello-world will produce a .bin file and an .elf file. Not a .hex. These files should be located under examples/hello-world/bin/nrf52dk/
Which version of RIOT are you using ?
Could you paste the output of `dist/tools/ci/print_toolchain_versions.sh` ?

tcpipchip, 2019-11-05:
Hello!!!!
Look
osboxes@osboxes:~/BLE-environment-sensor/examples/hello-world$ make BOARD=nrf52832-mdk
Building application "hello-world" for "nrf52832-mdk" with MCU "nrf52".

"make" -C /home/osboxes/BLE-environment-sensor/boards/nrf52832-mdk
"make" -C /home/osboxes/BLE-environment-sensor/core
"make" -C /home/osboxes/BLE-environment-sensor/cpu/nrf52
"make" -C /home/osboxes/BLE-environment-sensor/cpu/cortexm_common
"make" -C /home/osboxes/BLE-environment-sensor/cpu/cortexm_common/periph
"make" -C /home/osboxes/BLE-environment-sensor/cpu/nrf52/periph
"make" -C /home/osboxes/BLE-environment-sensor/cpu/nrf5x_common
"make" -C /home/osboxes/BLE-environment-sensor/cpu/nrf5x_common/periph
"make" -C /home/osboxes/BLE-environment-sensor/drivers
"make" -C /home/osboxes/BLE-environment-sensor/drivers/periph_common
"make" -C /home/osboxes/BLE-environment-sensor/sys
"make" -C /home/osboxes/BLE-environment-sensor/sys/auto_init
"make" -C /home/osboxes/BLE-environment-sensor/sys/div
"make" -C /home/osboxes/BLE-environment-sensor/sys/newlib_syscalls_default
"make" -C /home/osboxes/BLE-environment-sensor/sys/stdio_uart
"make" -C /home/osboxes/BLE-environment-sensor/sys/xtimer
   text	   data	    bss	    dec	    hex	filename
  11896	    116	   2584	  14596	   3904	/home/osboxes/BLE-environment-sensor/examples/hello-world/bin/nrf52832-mdk/hello-world.elf
osboxes@osboxes:~/BLE-environment-sensor/examples/hello-world$ cd bin
osboxes@osboxes:~/BLE-environment-sensor/examples/hello-world/bin$ cd nrf52832-mdk/
osboxes@osboxes:~/BLE-environment-sensor/examples/hello-world/bin/nrf52832-mdk$ ls
application_hello-world    cortexm_common_periph    **hello-world.elf**            periph_common.a
application_hello-world.a  cortexm_common_periph.a  **hello-world.hex**            riotbuild
auto_init                  cpu                      **hello-world.map**            stdio_uart
auto_init.a                cpu.a                    newlib_syscalls_default    stdio_uart.a
board                      cpu_common               newlib_syscalls_default.a  sys
board.a                    cpu_common.a             nrf5x_common_periph        sys.a
core                       div                      nrf5x_common_periph.a      xtimer
core.a                     div.a                    periph                     xtimer.a
cortexm_common             drivers                  periph.a
cortexm_common.a           drivers.a                periph_common
osboxes@osboxes:~/BLE-environment-sensor/examples/hello-world/bin/nrf52832-mdk$

and see
`osboxes@osboxes:~/BLE-environment-sensor/dist/tools/ci$ cat print_toolchain_versions.sh 
#!/bin/sh

get_cmd_version() {
    if [ -z "$1" ]; then
        return
    fi

    VERSION_RAW=$( ($@ --version) 2>&1)
    ERR=$?
    VERSION=$(echo "$VERSION_RAW" | head -n 1)

    if [ $ERR -eq 127 ] ; then # 127 means command not found
        VERSION="missing"
    elif [ $ERR -ne 0 ] ; then
        VERSION="error: ${VERSION}"
    fi

    printf "%s" "$VERSION"
}

get_define() {
    local cc="$1"
    local line=
    if command -v "$cc" 2>&1 >/dev/null; then
        line=$(echo "$3" | "$cc" -x c -include "$2" -E -o - - 2>&1 | sed -e '/^[   ]*#/d' -e '/^[  ]*$/d')
    fi
    if [ -z "$line" ]; then
        line=missing
    fi
    printf "%s" "$line"
}

get_kernel_info() {
    uname -mprs
}

get_os_info() {
    local os="$(uname -s)"
    local osname="unknown"
    local osvers="unknown"
    if [ "$os" = "Linux" ]; then
        osname="$(cat /etc/os-release | grep ^NAME= | awk -F'=' '{print $2}')"
        osvers="$(cat /etc/os-release | grep ^VERSION= | awk -F'=' '{print $2}')"
    elif [ "$os" = "Darwin" ]; then
        osname="$(sw_vers -productName)"
        osvers="$(sw_vers -productVersion)"
    fi
    printf "%s %s" "$osname" "$osvers"
}

newlib_version() {
    if [ -z "$1" ]; then
        printf "%s" "error"
    else
        local cc="$1"
        printf "%s" "$(get_define "$cc" newlib.h _NEWLIB_VERSION)"
    fi
}

avr_libc_version() {
    if [ -z "$1" ]; then
        printf "%s" "error"
    else
        local cc="$1"
        printf "%s (%s)" "$(get_define "$cc" avr/version.h __AVR_LIBC_VERSION_STRING__)" "$(get_define "$cc" avr/version.h __AVR_LIBC_DATE_STRING__)"
    fi
}

printf "\n"
# print operating system information
printf "%s\n" "Operating System Environment"
printf "%s\n" "-----------------------------"
printf "%23s: %s\n" "Operating System" "$(get_os_info)"
printf "%23s: %s\n" "Kernel" "$(get_kernel_info)"
printf "\n"

printf "%s\n" "Installed compiler toolchains"
printf "%s\n" "-----------------------------"
printf "%23s: %s\n" "native gcc" "$(get_cmd_version gcc)"
for p in \
         arm-none-eabi \
         avr mips-mti-elf \
         msp430 \
         riscv-none-embed \
         xtensa-esp32-elf \
         xtensa-lx106-elf \
         ; do
    printf "%23s: %s\n" "$p-gcc" "$(get_cmd_version ${p}-gcc)"
done
printf "%23s: %s\n" "clang" "$(get_cmd_version clang)"
printf "\n"
printf "%s\n" "Installed compiler libs"
printf "%s\n" "-----------------------"
# platform specific newlib version
for p in \
         arm-none-eabi \
         mips-mti-elf \
         riscv-none-embed \
         xtensa-esp32-elf \
         xtensa-lx106-elf \
         ; do
    printf "%23s: %s\n" "$p-newlib" "$(newlib_version ${p}-gcc)"
done
# avr libc version
printf "%23s: %s\n" "avr-libc" "$(avr_libc_version avr-gcc)"
# tools
printf "\n"
printf "%s\n" "Installed development tools"
printf "%s\n" "---------------------------"
for c in \
         cmake \
         cppcheck \
         doxygen \
         git \
         make \
         openocd \
         python \
         python2 \
         python3 \
         ; do
    printf "%23s: %s\n" "$c" "$(get_cmd_version "${c}")"
done
printf "%23s: %s\n" "flake8" "$(get_cmd_version "python3 -Wignore -m flake8")"
printf "%23s: %s\n" "coccinelle" "$(get_cmd_version spatch)"

exit 0
`

dylad, 2019-11-05:
> cat print_toolchain_versions.sh

Please run this script instead, and copy/paste the output. (the content of the script isn't useful so don't use cat)

`./print_toolchain_versions.sh`

> about the version, maybe this the problem to
> i got IOT from
> git clone https://github.com/iota-community/

I don't know what you downloaded there, but I don't see any RIOT repo from iota-community.

You should give a try to the official RIOT 
`git clone https://github.com/RIOT-OS/RIOT.git`

Could you share your development branch to a public repo somewhere ? It might be easier to debug if I can see some code.

aabadie, 2019-11-05:
In order to print your toolchain versions, it's easier to run `make print-versions` from the RIOT directory.

tcpipchip, 2019-11-05:
All running fine now on NINA B112.
Used the correct GIT and used the new J-FLASH
But no bin created, only HEX
Thanks for all team!

https://riot-nina-b112.blogspot.com/2019/11/riot-para-nina-b.html

closed 2019-11-05 by tcpipchip