RIOT-OS/RIOT #12651: Failing tests on AVR (tested with atmega256rfr2-xpro)

aabadie, 2019-11-06:
#### Description

Opening this issue for tracking the problems with tests found in #12649. I remove the problems related to interaction with the UART because they should be fixed #12461 and the problem related to network setup with GNRC tests.

Here is the list of failing tests:
- [ ] <b>tests/bitarithm_timings</b>: looks like a context switching issue
  <details>
  
  ```
  make RIOT_CI_BUILD=1 CC_NOCOLOR=1 --no-print-directory -C ./tests/bitarithm_timings test
  /work/riot/RIOT/dist/tools/pyterm/pyterm -p "/dev/ttyACM0" -b "115200" --noprefix --no-repeat-command-on-empty-line
  Connect to serial port /dev/ttyACM0
  Welcome to pyterm!
  Type '/exit' to exit.
  
  
  main(): This is RIOT! (Version: buildtest)
  Start.
  + bitarithm_msb: 102096 iterations per second
  Timeout in expect script at "child.expect('\+ bitarithm_lsb: \d+ iterations per second')" (tests/bitarithm_timings/tests/01-run.py:16)
  ```
  
  </details>

- [ ] <b>tests/event_wait_timeout</b>: looks like a context switching issue

  <details>
  
  ```
  make RIOT_CI_BUILD=1 CC_NOCOLOR=1 --no-print-directory -C ./tests/event_wait_timeout test
  /work/riot/RIOT/dist/tools/pyterm/pyterm -p "/dev/ttyACM0" -b "115200" --noprefix --no-repeat-command-on-empty-line
  Connect to serial port /dev/ttyACM0
  Welcome to pyterm!
  Type '/exit' to exit.
  finished: 4/4 events and 1/4 timeouts recorded
  [FAILED]

  main(): This is RIOT! (Version: buildtest)
  [START] event_wait_timeout test application.

  finished: 4/4 events and 1/4 timeouts recorded
  [FAILED]
  Timeout in expect script at "child.expect_exact(u"[SUCCESS]")" (tests/event_wait_timeout/tests/01-run.py:15)
  ```

  </details>

- [ ] <b>tests/gnrc_ipv6_fwd_w_sub</b>: maybe again a missing network configuration ?

  <details>
  
  ```
  make RIOT_CI_BUILD=1 CC_NOCOLOR=1 --no-print-directory -C ./tests/gnrc_ipv6_fwd_w_sub test
  run_test
  /work/riot/RIOT/dist/tools/pyterm/pyterm -p "/dev/ttyACM0" -b "115200" --noprefix --no-repeat-command-on-empty-line
  Connect to serial port /dev/ttyACM0
  Welcome to pyterm!
  Type '/exit' to exit.
  
  
  main(): This is RIOT! (Version: buildtest)
  > Timeout in expect script at "child.expect(r"Forwarded Ethernet frame:")" (tests/gnrc_ipv6_fwd_w_sub/tests/01-run.py:16)
  ```
  
  </details>

- [ ] <b>tests/gnrc_ndp</b>: no idea of the problems

  <details>
  
[... 271 lines cut ...]
  
  Timeout in expect script at "child.expect('>')" (tests/ps_schedstatistics/tests/01-run.py:43)
  ```
  
  </details>

#### Steps to reproduce the issue

Run the following command:

```
$ BUILD_IN_DOCKER=1 ./dist/tools/compile_and_test_for_board/compile_and_test_for_board.py . atmega256rfr2-xpro --jobs=4 --with-test-only
```

#### Expected results

All tests succeed. Unsupported tests (for good reasons) are skipped.

#### Actual results

Some tests are failing.

#### Versions

Versions from riotbuild docker image.

aabadie, 2019-11-06:
For `tests/pkg_lora-serialization`, there's a portability problem with 8bit architectures. https://github.com/leandrolanzieri/lora-serialization/pull/1 is not completely solving the failing test (because there are apparently also floating point precision issues) but is already an improvement I think.
The floating point issue can be solved by using an adapted expected test array in the case where AVR GCC is used.

maribu, 2019-11-23:
I was able to fix `tests/ps_schedstatistics` via this:

```patch
diff --git a/tests/ps_schedstatistics/main.c b/tests/ps_schedstatistics/main.c
index 7d2f9de6d3..ce95fad7b5 100644
--- a/tests/ps_schedstatistics/main.c
+++ b/tests/ps_schedstatistics/main.c
@@ -45,7 +45,7 @@ static void *_thread_fn(void *arg)
         for (int i = 0; i < (10 * (next + 1)); ++i) {
             _xtimer_now64();
         }
-        xtimer_usleep(XTIMER_BACKOFF * 3);
+        xtimer_usleep(XTIMER_BACKOFF * 30);
         msg_send(&m2, pids[next]);
     }
 ```

Those threads just created too much load for an ATmega to handle, so that the main thread was starving. I have the feeling that just generally reducing the load will not be ideal for benchmarking ARM boards. (I checked this on top of the fix in #12790, but I believe that this is unrelated really just caused by the load.)

fjmolinas, 2019-11-25:
I think the reason `tests/ps_schedstatistics` is failing is because of un-proper `XTIMER_BACKOFF` , see #11522 and #11546. It is also the case for `frdm-kw41z`

maribu, 2019-11-25:
I'm not so sure that this is the sole reason. `_xtimer_now64()` is **way** more expensive in terma of CPU cycles on ATmega.

I guess that a value for `XTIMER_BACKOFF` that would allow the test to pass would be so high, that we get regressions elsewhere. E.g. some apps might have to threads which both periodically get executed with a short period. If `XTIMER_BACKOFF` gets to high, the other thread might strave.

But I'll happily give it a try to see by how much the backoff needs to be increased to get the test working. If the value doesn't get super high, this would indeed be the better fix.

aabadie, 2019-11-26:
@miri64, do you have any idea of what could be the problem with `tests/gnrc_ndp` on AVR boards (see details in the 4th point of this issue) ?

miri64, 2019-11-26:
No, not really. The test is working as expected on ARM Cortex :-/ and I don't really have an AVR board at hand right now nor the time to investigate this. For the `gnrc_pktbuf_is_empty()` fails I would at least use `gnrc_pktbuf_stats()` first to check what is the packet stuck in the packet buffer, for the others (if available) I would use the debugger.

aabadie, 2019-11-28:
Regarding `ps_schedstatistics`, even when changing XTIMER_BACKOFF from 3 to 10, it's working on `atmega256rfr2-xpro` and `arduino-mega2560`. On a samr21-xpro, I can also see some load on the different threads. So maybe increasing the XTIMER_BACKOFF value to 10 would be enough ?

aabadie, 2019-11-28:
@maribu, I opened #12832 because I'd like to move forward with `tests/ps_schedstatistics` on AVR. Threre are other problems with xtimer but I think that we should not block the test to succeed on AVR because of this.

fjmolinas, 2020-07-22:
I have another failing test on avr, `tests/thread_msg_bus`, tested on `arduino-mega2560` and `atmega256rfr2-xpro`:

```
main(): This is RIOT! (Version: 2020.10-devel-18-gab072f-HEAD)
THREAD 1 start
THREAD 2 start
THREAD 3 start
THREADS CREATED
Posted event 22 to 0 threads
Posted event 23 to 0 threads
Posted event 24 to 0 threads
SUCCESS
Timeout in expect script at "child.expect_exact('T1 recv: Hello Threads! (type=23)')" (tests/thread_msg_bus/tests/01-run.py:15)
```

@benpicco can you take a look?

fjmolinas, 2020-07-22:
I updates the issue.

fjmolinas, 2020-07-22:
I have another failing test on avr, `tests/pkg_libb2`, tested on `arduino-mega2560` and `atmega256rfr2-xpro`:

```
2020-07-22 15:07:30,503 # Help: Press s to start test, r to print it is ready
s
2020-07-22 15:07:31,824 # START
2020-07-22 15:07:31,830 # main(): This is RIOT! (Version: 2020.10-devel-18-gab072f-HEAD)
2020-07-22 15:07:31,837 # .
2020-07-22 15:07:31,845 # blake2_tests.test_blake2s (tests/pkg_libb2/main.c 56) memcmp(b2s, hash, sizeof hash) == 0
2020-07-22 15:07:31,865 # .

2020-07-22 15:07:31,870 # Help: Press s to start test, r to print it is ready
```

Updated list again.

fjmolinas, 2020-07-22:
`tests/ps_schedstatistics` is failing again as well.

chudov, 2021-02-18:
I have successful test `tests/thread_msg_bus` on `derfmega256`:

    main(): This is RIOT! (Version: 2021.04-devel-661-g6b8cb)
    THREAD 1 start
    THREAD 2 start
    THREAD 3 start
    THREADS CREATED
    Posted event 22 to 0 threads
    T1 recv: Hello Threads! (type=23)
    T3 recv: Hello Threads! (type=23)
    Posted event 23 to 2 threads
    T2 recv: Hello Threads! (type=24)
    Posted event 24 to 1 threads
    SUCCESS

chudov, 2021-02-18:
Regarding `pkg_libb2`: 8bit architectures are not supported by libb2. 
@fjmolinas Seems it make sense to exclude `tests/pkg_libb2` from the list.

fjmolinas, 2021-02-19:
> I have successful test `tests/thread_msg_bus` on `derfmega256`:
>
> ```
> main(): This is RIOT! (Version: 2021.04-devel-661-g6b8cb)
> THREAD 1 start
> THREAD 2 start
> THREAD 3 start
> THREADS CREATED
> Posted [...]

Yes, I have the same result seems this stopped failing somewhere along the way.

fjmolinas, 2021-02-19:
> Yes, I have the same result seems this stopped failing somewhere along the way.

Actually the PR is linked the list was never updated, #14582

fjmolinas, 2021-02-19:
> pkg_libb2

Yes, that would make sense, do you have a pointer to where the supported arch are specified?

aabadie, 2021-02-19:
Regarging libb2, this was "fixed" in #15513 (because indeed this package doesn't support 8bit)

aabadie, 2021-02-19:
So there's only `tests/gnrc_ndp` that's still failing and TBH I have no idea what is the cause.

fjmolinas, 2021-02-19:
> Regarging libb2, this was "fixed" in #15513 (because indeed this package doesn't support 8bit)

Ah thanks!

closed 2021-02-19 by miri64

fjmolinas, 2021-02-19:
Nice!

aabadie, 2021-02-19:
Thanks for the last fix @miri64 !