RIOT-OS/RIOT #12658: how to find out which line  failed assertion

jhx2002, 2019-11-07:
Failed assertion, but where ?  such as following log. 
Is there any way to figure out the line failed.? 
```
2019-11-07 13:18:11,894 # [drive_motor_ctl]recv: /controller/cmd 1526,-103 
2019-11-07 13:18:11,990 # 0x800ac8b
2019-11-07 13:18:11,992 # *** RIOT kernel panic:
2019-11-07 13:18:11,993 # FAILED ASSERTION.
2019-11-07 13:18:11,993 # 
2019-11-07 13:18:12,001 # 	pid | name                 | state    Q | pri | stack  ( used) | base addr  | current     
2019-11-07 13:18:12,009 # 	  - | isr_stack            | -        - |   - |    512 (  144) | 0x20000000 | 0x200001c8
2019-11-07 13:18:12,018 # 	  1 | idle                 | pending  Q |  15 |    256 (  156) | 0x200005c8 | 0x20000644 
2019-11-07 13:18:12,023 # 	  2 | main                 | bl mutex _ |   7 |   1536 ( 1196) | 0x200006c8 | 0x200008d4 
2019-11-07 13:18:12,032 # 	  3 | ipv6                 | bl rx    _ |   4 |   1024 (  564) | 0x20001134 | 0x2000142c 
2019-11-07 13:18:12,040 # 	  4 | udp                  | bl rx    _ |   5 |   1024 (  308) | 0x20003200 | 0x20003504 
2019-11-07 13:18:12,049 # 	  5 | w5100                | bl rx    _ |   2 |   1024 (  496) | 0x20000d34 | 0x20000fec 
2019-11-07 13:18:12,057 # 	  6 | emcute_task          | running  Q |   7 |   1024 ( 1020) | 0x2000362c | 0x200038ac 
2019-11-07 13:18:12,065 # 	  7 | data_collector_task  | pending  Q |   7 |   1024 ( 1020) | 0x20003a2c | 0x20003c5c 
2019-11-07 13:18:12,073 # 	  8 | drive_motor_ctl_task | pending  Q |   7 |   1024 ( 1020) | 0x20003e40 | 0x200040d4 
2019-11-07 13:18:12,081 # 	  9 | steering_motor_ctl_task | pending  Q |   7 |   1024 ( 1020) | 0x2000424c | 0x200044c4 
2019-11-07 13:18:12,085 # 	    | SUM                  |            |     |   9472 ( 6944)

```

aabadie, 2019-11-07:
Just add `CFLAGS=-DDEBUG_ASSERT_VERBOSE` to the make command line and the failed assertion will report the line.

jhx2002, 2019-11-07:
Thanks @aabadie

miri64, 2019-11-07:
Alternatively, you can use the `addr2line` tool of the toolchain used to translate the address here
> ```
> 2019-11-07 13:18:11,894 # [drive_motor_ctl]recv: /controller/cmd 1526,-103 
> 2019-11-07 13:18:11,990 # 0x800ac8b
> 2019-11-07 13:18:11,992 # *** RIOT kernel panic:
> ```

to the line.

closed 2019-11-07 by miri64