RIOT-OS/RIOT #12674: vagrant: Update to Ubuntu 18.04

miri64, 2019-11-08:
Our Vagrantfile currently points to the Ubuntu 16.04 release, while our Docker points to 18.04 and our buildsystem also assumes more up-to-date toolchains. Should we update that image to Ubuntu 18.04?

aabadie, 2019-11-08:
I had a look at that 3 weeks ago because the build of the vagrant image is broken: the ubuntu 16.04 iso is no longer available for download. But I had other issues during the build. Maybe my packer version is not compatible or there's something broken no my local setup.
I gave up in the end.

aabadie, 2019-11-08:
Otherwise +1 for updating the image of Ubuntu of course (at least to fix the broken build).

cgundogan, 2019-11-29:
The installation process in ubuntu completely changed so that the shell scripts need a major face-lift. Also, a lot of things (like which toolchains to install) are heavily duplicated. Sharing common scripts between the docker setup and vagrant setup might be really handy, here. @smlng do you have anything in that regard?

closed 2020-02-19 by cgundogan