RIOT-OS/RIOT #12680: RFC: How to go ahead with pointer based (GNRC) netif and sock?

miri64, 2019-11-10:
Now that `gnrc_netif_t` and `netif_t` go more into the direction of a pointer-based system we have somewhat of a problem with `sock`, where [`uint16_t` is used to identify an interface](https://github.com/RIOT-OS/RIOT/blob/master/sys/include/net/sock.h#L207). How do we want to address this in the future? Have a `uint16_t netif_get_id(netif_t *netif)` function that converts however the stack identifies the interface internally to a global ID that `sock` can use?

The usage now is also not very portable btw. Up until now, the PID is used, which only applies for GNRC.

jia200x, 2019-11-11:
I'm adding @leandrolanzieri since he also found this problem while working on LWM2M on top of sock

jia200x, 2019-11-11:
On first thoughts, we where thinking of aiming to pointers instead of `uint16_t` for the netif representation.

Most of the stacks I've been working with have a direct or indirect pointer representation. Keeping an ID representation means that we would need to add some workarounds here and there.

miri64, 2019-11-11:
This issue isn't and shouldn't be about how network interfaces are represented within RIOT. It is about how we deal with RIOT's network interface representation with regards to `sock` *without* changing `sock`.

miri64, 2019-11-11:
> Most of the stacks I've been working with have a direct or indirect pointer representation. Keeping an ID representation means that we would need to add some workarounds here and there.

Not true btw. I have never seen an interface pointer in Linux. There it is names and file descriptors ;-).

miri64, 2019-11-11:
> Keeping an ID representation means that we would need to add some workarounds here and there.

The 16-bit ID could be generated from the pointer if need be e.g.

jia200x, 2019-11-11:
> Not true btw. I have never seen an interface pointer in Linux. There it is names and file descriptors ;-).

Don't get me wrong with this, I mean there's always a way to get the pointer directly or indirectly.
E.g we have `netif_get_by_name`, or the function you are proposing (`netif_get_by_id`).
This is true at least for OpenThread, LWIP, GNRC, etc

leandrolanzieri, 2019-11-11:
> It is about how we deal with RIOT's network interface representation with regards to `sock` _without_ changing `sock`.

@miri64 with this do you mean that we should keep the 16-bit representation for the interfaces?

leandrolanzieri, 2019-11-18:
>Have a uint16_t netif_get_id(netif_t *netif) function that converts however the stack identifies the interface internally to a global ID that sock can use?

What about extending sock with a function to assign an interface? something like `sock_<type>_set_netif(sock_<type>_ep_t *ep, const netif_t *netif)`, that must be implemented by every network stack.

If we then decide on switching to have a `netif_t *netif` in the endpoint structure, this can be replaced by a simple inline function (`ep->netif = netif`).

miri64, 2019-11-18:
This is a contradiction in itself. You can't have both: a network stack specific implementation and a static inline function, as they have different signatures.

Also, `netif_t`  is a RIOT-specific type while `sock` was designed to be OS-independent, so it rather should go into `netif.h` as I proposed originally.

miri64, 2019-11-18:
Sorry, hit the wrong button on my phone

miri64, 2019-11-18:
How about this? 

```c
static inline int uint16_t netif_get_id(netif_t *netif)
{
     intptr_t netif_ptr = (intptr_t)netif;
     return (uint16_t)(netif & UINT16_MAX);
}
```

miri64, 2019-11-18:
Likewise then a `netif_get_by_pid()` function that reverses the operation.

leandrolanzieri, 2019-11-18:
> This is a contradiction in itself. You can't have both: a network stack specific implementation and a static inline function, as they have different signatures.

I was thinking more on having the stack specific version for now (and *then* if we change sock, we could have the inline version). So GNRC could have:

```C
void sock_udp_set_netif(sock_udp_ep_t *ep, const netif_t *netif)
{
    const gnrc_netif_t * _netif = (gnrc_netif_t *)netif;
    ep->netif = (uint16_t)_netif->pid;
}
```

miri64, 2019-11-18:
Again, I'm against adding this to `sock`, as it adds OS-specific types to an OS-independent API.

kaspar030, 2019-11-18:
We're talking about a small number of network interfaces, wight? Like, less than 5?

IMO, in that case, it is acceptable to have an array of pointers to netif, with the array index being the interface integer id for sock. For the reverse (netif_t to id), either netif_t needs a field.

kaspar030, 2019-11-18:
> Again, I'm against adding this to `sock`, as it adds OS-specific types to an OS-independent API.

+1!

miri64, 2019-11-18:
I made a proposal in #12738.

miri64, 2020-07-06:
#12738 was merged, so this can be closed. We still should port all instances of `netif->pid` to `netif_get_id`, though.

closed 2020-07-06 by miri64