RIOT-OS/RIOT #12688: RFC: netif rework

jia200x, 2019-11-11:
## Description
I'm opening this issue to synchronize and keep track of the netif rework.
Here are some bullet points that summarize the motivation of such a rework. 

### Improve support of software MAC layers
Currently the support of software MACs is limited to `gnrc_netif`, so other network stacks cannot reuse whatever is there. 

Also most network stacks relay on MAC functionalities implemented in hardware (auto-ACK, retransmissions, etc). If a radio doesn't provide that cap (or the driver doesn't implement it) then the performance is not optimal.

### Define stack independent network interfaces
If netifs are stack independent, all network stacks can reuse the auto initialization procedures, reuse OS events and write portable applications.

### Separate transceiver, PHY and MAC logic
If we can provide interfaces between this components we could benefit of:
-  A lot of code re utilization.
- Simpler network device drivers (e.g that only write transceiver logic instead of transceiver+PHY+MAC logic).
- A better interface with external code that already include a PHY or MAC layer
    - E.g see https://github.com/RIOT-OS/RIOT/blob/03c467af75c7257fa9597a9f0539055a8dc3044e/cpu/esp32/esp-wifi/esp_wifi_netdev.c#L169. The pkt is already there but a transceiver function has to be called in order to pass the packet to the network interface.
- A better testing infrastructure for network devices and their layers.

### Remove the need of allocating one thread per interface
It's needed to allocate one `gnrc_netif_t` thread per network interface. Besides allocating more resources than needed, this is sometimes problematic for platforms that have several transceivers.

## Proposed roadmap
This whole rework can be done in the following phases
### 1. Rework and extend the `netif ` API
This step is required in order to provide the basis 

- [x] Make `netif_t` a pointer instead of a stack defined type (#11879)
- [ ] Define an interface for stack independent packet allocation, data handling and passing data up to the stack
- [ ] Extend the `netif` API to add send/recv operations (analog to `sock`, but intended to be used from network stacks or applications that send data via an interface)
- [ ] Migrate common code from `gnrc_netif` to `netif`
    - [ ] Remove GNRC dependencies in MAC layers (e.g `gnrc_pktbuf` dependencies in `gnrc_netif_xxx` functions)
    - [ ] Move `gnrc_netif` events to external event handlers + callbacks.
    - [ ] Move `gnrc_netif_ops_t` ops into `netif_ops_t`

### 2. Improve Link Layer support
- [ ] Add support for radio caps (https://github.com/RIOT-OS/RIOT/pull/11473)
- [ ] Add support for common link layer features
    - [ ] Auto-ACK
    - [ ] L2 retransmissions
    - [ ] L2 security

### 3. Add required interfaces PHY and transceiver interfaces
TBD

I will open a Github project to be able synchronize better.

### Useful links
https://github.com/RIOT-OS/RIOT/issues/11483
#11879 
https://github.com/RIOT-OS/RIOT/pull/11473

jia200x, 2019-11-11:
@RIOT-OS/maintainers there have been some offline conversation with some maintainers (@haukepetersen, @bergzand, @kaspar030, @miri64, @leandrolanzieri, @PeterKietzmann).

Unfortunately we didn't have time to discuss this during the RIOT summit, so I open this issue for syncing.

All kind of feedback on the fundamentals or roadmap is more than welcome!

maribu, 2019-11-11:
Also related to this: https://github.com/RIOT-OS/RIOT/issues/12469

jia200x, 2019-11-11:
@maribu added to the road map

miri64, 2019-11-12:
>  Define an interface for stack independent packet allocation, data handling and passing data up to the stack
> […]
> * [ ]  Remove GNRC dependencies in MAC layers (e.g `gnrc_pktbuf` dependencies in `gnrc_netif_xxx` functions)

I thought we figured last week, that this makes things more complicated than it need be…

jia200x, 2019-11-12:
> I thought we figured last week, that this makes things more complicated than it need be…

What I meant with that was: make the MAC layer component independent of GNRC

jia200x, 2019-11-12:
@miri64 rephrased :)

roberthartung, 2019-11-13:
#12128 would probably benefit from this as well.

jia200x, 2020-01-20:
I renamed it to "lower network stack" rework, since the network interface are only one of the components involved

jia200x, 2020-02-17:
I would like to share some thoughts about the "Write network stack independent network interfaces" part:

After some time, I think it's different to distinguish these 2 elements:
1. The OS representation of a network interface
2. The glue code between a network stack and the OS.

For 1., it's clear that the representation of a network interface should be agnostic to the network stack (e.g running `ifconfig` should work with all network stacks, same as `sock`). We already have most of this functions (' netif_iter`, `netif_get_opt`, etc).

However, 2. doesn't necessarily need a unique entry point (e.g `netdev_recv` and `netif_send` function) or a global allocator (`netbuf`). Here are some of the reasons:
1. Some network stacks require access to the radio (OpenWSN), some other to the PHY/SubMAC layer (OpenThread, LWIP) or even on top of the Link Layer. So, not all network stacks would make use of the `netif_recv` and `netif_send` function. Only access to a dedicated layer is needed.
2. Some MAC layers and drivers already have Framebuffers. Those who don't, it's always possible to design MAC layer allocators (e.g for Ethernet) or simply write one function per stack to allocate a packet (in the RECV function of the HAL).
3. Device ISR handling can be done by an (optional) `event_thread` module (#12474 ), and this logic can be used by all Network Stacks. This would simplify much more the integration of network stacks

And last but not least, solving the `netif` problem first doesn't solve the problems of radios with different caps and lack of MAC layers, so I would probably give more priority to the devices/link layer rework.

Any comments?

jia200x, 2020-03-31:
To make this easier to follow, I will split this rework in 2 independent reworks:
1. Improve GNRC Netif
2. Network stack independent link layers (this also includes HALs and mechanisms to process IRQs)

jia200x, 2020-03-31:
this is reorganized in #13771 and #13763 , so I will close the issue.

closed 2020-03-31 by jia200x