RIOT-OS/RIOT #12700: Cannot delete files in SD card. And I can't read more bytes than I ever wrote in tests/pkg_fatfs.

ouyangkan, 2019-11-13:
#### Description

I use [tests/pkg_fatfs](https://github.com/RIOT-OS/RIOT/tree/master/tests/pkg_fatfs).But there is no file deletion.When I read more data than I had ever written, the system stopped working.

ouyangkan, 2019-11-13:
**1.I added a new delete function:**

> int sd_shell_delete(int argc, char **argv)
 {
     char *filename;
     if(argc == 2)
     {
        filename = argv[1];
     }
    else  {
        return 0;
    }
    printf("usage: delete <%s> \n", filename);
     if(filename != NULL)
    {
        f_unlink(filename);
    }
    return 0;
}

**2.I learned that f_size can read file sizes.So I can use read file to read everything.**

MichelRottleuthner, 2019-11-21:
Thanks for opening this issue adding a delete function to the test could definitely be helpful. 
Can you please elaborate/give an example on how the problem you had with the read function exactly looked like? Do you mean that giving a too big `len` parameter to the `read` call breaks the example application?

ouyangkan, 2019-11-21:
I use `read filename` to [read](https://github.com/RIOT-OS/RIOT/blob/8f90da6040aedace5ef390f28cd19cf1ecd08baa/tests/pkg_fatfs/main.c#L188) all the content of filename.Because the filesize will acquired from the f_size function.

ouyangkan, 2019-11-21:
> Thanks for opening this issue adding a delete function to the test could definitely be helpful.
> Can you please elaborate/give an example on how the problem you had with the read function exactly [...]

I use `read filename` to [read](https://github.com/RIOT-OS/RIOT/blob/8f90da6040aedace5ef390f28cd19cf1ecd08baa/tests/pkg_fatfs/main.c#L188) all the content of filename.Because the filesize will acquired from the f_size function.

MichelRottleuthner, 2019-11-21:
>I use read filename to read all the content of filename.Because the filesize will acquired from the f_size >function.

Ok, and that doesn't work as expected? Is so, how does the error look like?

ouyangkan, 2019-11-21:
> > I use read filename to read all the content of filename.Because the filesize will acquired from the f_size >function. 
> 
> Ok, and that doesn't work as expected? If so, how does the error look like?

ouyangkan, 2019-11-21:
> > > I use read filename to read all the content of filename.Because the filesize will acquired from the f_size >function.
> > 
> > 
> > Ok, and that doesn't work as expected? If so, how does the error look like?

Sorry,  I found no error due to my misunderstanding.

closed 2019-11-21 by ouyangkan