RIOT-OS/RIOT #12715: sdcard_spi_write_blocks Write time is too long in sdcard_spi.c.How can I speed him up?

ouyangkan, 2019-11-15:
#### Description

I use stmf4 and class 10 microSD card, but when i use [sdcard_spi_write_blocks ](https://github.com/RIOT-OS/RIOT/blob/1018a6fa674591299c25405761ccaa15008258de/drivers/sdcard_spi/sdcard_spi.c#L813).It USES my 3ms.He doesn't have the speed of an sd card.How can I speed him up.

### Useful links

1.I use [RIOT/tests/driver_sdcard_spi/](https://github.com/RIOT-OS/RIOT/tree/master/tests/driver_sdcard_spi) to test it.

2.Add xtimer to calculate run time in [_write:](https://github.com/RIOT-OS/RIOT/blob/1018a6fa674591299c25405761ccaa15008258de/tests/driver_sdcard_spi/main.c#L265)
    

> uint32_t now = xtimer_now_usec();
    sd_rw_response_t state;
    int chunks_written = sdcard_spi_write_blocks(card, bladdr, write_buffer, SD_HC_BLOCK_SIZE, 1, &state);
    uint32_t later = xtimer_now_usec();

3.This is results(it run 2.75ms):
>write 0 adsf 4
> 2019-11-15 07:30:16,946 #  write 0 adsf 4
> 2019-11-15 07:30:16,952 # will write 'adsf' (4 chars) at start of block 0
> 2019-11-15 07:30:16,954 # the rest of the block will be filled with zeros
> 2019-11-15 07:30:16,960 # time1 sdf_creatnew is 2751
> 2019-11-15 07:30:16,961 # write block 0 [OK]

ouyangkan, 2019-11-18:
1.Change [spi_acquice](https://github.com/RIOT-OS/RIOT/blob/3b8cbec6c540b75173fe2bb11ac71512c35742c7/cpu/stm32_common/periph/spi.c#L152) in spi.c in 42MHz.(It is 37.5MHz Max from datasheet):
>    uint8_t br = spi_divtable[spi_config[bus].apbbus][clk];
    if(br == 1){
        br = 0;
    }
    dev(bus)->CR1 = ((br << BR_SHIFT) | mode | SPI_CR1_MSTR);

2,Change [_write_blocks](https://github.com/RIOT-OS/RIOT/blob/3b8cbec6c540b75173fe2bb11ac71512c35742c7/drivers/sdcard_spi/sdcard_spi.c#L773) in sdcard_spi.c.In my opinion, the "wait for busy"function should be put before using "SPI", rather than after it.Because takes too long to run the "wait for busy" function:
> // if (!_wait_for_not_busy(card, SD_WAIT_FOR_NOT_BUSY_CNT)) {
 //     DEBUG("_write_blocks: _wait_for_not_busy: [FAILED]\n");
 //     _unselect_card_spi(card);
 //     *state = SD_RW_TIMEOUT;
 //     return written;
 // }

3. Add periph_dma in the Makefile.

**others(when i use fatfs):**
4.Change FF_FS_TINY 0 was in ffconf.h(bin/pkg/stm32f4discovery/fatfs) when I use fatfs.Which make time unstable.

5.Creat sdf_open, sdf_write, sdf_close to use f_open ... (change it in [RIOT/tests/pkg_fatfs/)](https://github.com/RIOT-OS/RIOT/tree/master/tests/pkg_fatfs) .Because the running time of f_open and f_close are too long.In some situations, I want to use f_write only.

# Would you mind that I have a question. "If I was not using 'f_close', file would be lost."How can I figure it out?

# As the result , the same taken by sdcard is 1ms when I writing;However the "f_close" exists in fatfs ,when it takes '1ms' in fatfs.

ouyangkan, 2019-11-18:
I built sdf_sync to use f_sync.It can save my write.But it took the system 40ms.Can you give me some advice in f_sync @MichelRottleuthner : 
> 2019-11-18 05:29:40,565 # time sdf_sync is 44763
2019-11-18 05:29:40,826 # time sdf_write is 2593

MichelRottleuthner, 2019-11-18:
Hi @ouyangkan thaks for opening this issue. As you already noticed the speed of the sdcard driver depends on many factors. Speed wasn't the main priority when initially developing this driver and there are some things that can be done more efficiently - but that has other downsides like more RAM usage or worse compatibility to other platforms. 
Regarding your questions I'm not sure if I fully understand, but I'll try my best to provide some answers. 
1) When you write small chunks, the overhead will be quite high relative to the data that is written. So it is preferable to always write multiples of the blocksize at once (512 bytes).
2) Waiting for "not busy" state could, as you say, be moved to before writing data. The problem arising from that is that the calls then loose their "synchronous behaviour" i.e. an operation is not guaranteed to be finished after returning from the call.
3) regarding your remarks on "using f_write only": I'm not sure if I fully understand, but I guess you want to open() the file once, then write() (potentially multiple times) and then only close the file at the end? Yes sure you can do that, but obviously you can not avoid the open and close calls completely as fatfs may can also do some caching internally and only update the physical storage when needed (e.g. on close).
4) You _must_ close the file at the end. How should the file-system otherwise know that you are done with your work and commit everything to the storage?
5) One thing that you can optimize is [this function](https://github.com/RIOT-OS/RIOT/blob/5f4b8ef1310e4df2f09076abef902709f9f7f3b5/drivers/sdcard_spi/sdcard_spi.c#L565). Currently it uses an iteration to write all the bytes of a block. You could get a much better performance by using spi_transfer_bytes instead of spi_transfer_byte. The problem here is, that for initialization a software SPI function is needed to make it work properly on all sdcards which makes it a little bit more complicated that just replacing it.

Feel free to ask more questions I'll be glad to help.

ouyangkan, 2019-11-19:
1. For 1 and 5, I think Fatfs uses disk_write in [mtd_diskio.c.](https://github.com/RIOT-OS/RIOT/blob/76b9642e5f5700c5e369a77dc6ff9353c015c517/pkg/fatfs/fatfs_diskio/mtd/mtd_diskio.c#L122). It packs what I write using a Memory TechnologyDevice and uses disk_write when 512 bytes.
2. I'm also in trouble with 2,3,4. If I only use f_write , I can't complete synchronous brhavious. So I extract F_sync from F_close function to let sdcard be synchronous with stm32. It is successful to save all, but f_sync will cost me 40ms.

MichelRottleuthner, 2019-11-21:
>1. For 1 and 5, I think Fatfs uses disk_write in mtd_diskio.c.. It packs what I write using a Memory TechnologyDevice and uses disk_write when 512 bytes.

Yes. What I meant is that you get best performance if you actually write big chunks from your application level at once, i.e. don't write only "a handful" of bytes on each write.

>2. I'm also in trouble with 2,3,4. If I only use f_write , I can't complete synchronous brhavious. So I extract F_sync from F_close function to let sdcard be synchronous with stm32. It is successful to save all, but f_sync will cost me 40ms.

Yes, of course the sdcard driver can't guarantee that only using f_write will write everything to the storage in a synchronous fashion, because the file system logic sits between the sdcard and your application. This can (and, if configured like that, *will*) buffer writes to the storage medium and only finish all its operation with the f_close/f_sync. That your f_sync takes that much time is probably just related to the fact that "a lot of" interaction with the sdcard needs to happen then.

Other things you could consider:
Look at (5) from my previous comment, this will increase the sdcard write speed.
If possible in your case, consider using [another supported file system](https://github.com/RIOT-OS/RIOT/tree/master/sys/include/fs) like littlefs or spiffs. While fatfs might be nicely interoperable for using it on many systems, I doubt it has the best performance but I didn't benchmark it yet.

ouyangkan, 2019-11-21:
@MichelRottleuthner Thank you for your reply. And I founded one bug in [sdcard_spi](https://github.com/RIOT-OS/RIOT/blob/a05027cdba6a51e8b3f1e76170ce63a621aa1d3f/drivers/sdcard_spi/sdcard_spi.c#L479) when I speeded up. The system would die between `479: r1_resu = sdcard_spi_send_cmd(card, SD_CMD_55, SD_CMD_NO_ARG, 0);` and `481: r1_resu = sdcard_spi_send_cmd(card, sd_cmd_idx, argument, 0);`. Due to they were too close and there was no wait_for_not_busy between them.

MichelRottleuthner, 2019-11-21:
Is this actually a bug? (i.e.: is it also present without any of your modifications?)
What exactly do you mean when you say "when I speeded up"?
What are the changes you applied so that it fails?
I don't think that there should be a `wait_for_not_busy` between those two lines, because within the first `sdcard_spi_send_cmd ` call there is already a wait for the R1 response before it returns and continues with the next command. 
One thing that could be going on is that a timeout occurs while waiting for a valid R1. You can test that by printing the actual R1 value in line 492 to see if it was an error or timeout.
In case it is a timeout you can try to increase the `R1_POLLING_RETRY_CNT` to see if it helps.

ouyangkan, 2019-11-22:
> Is this actually a bug? (i.e.: is it also present without any of your modifications?)
> What exactly do you mean when you say "when I speeded up"?
> What are the changes you applied so that it fails?
> I [...]

This is my changes in sdcard_spi_send_acmd:
> do {
        DEBUG("sdcard_spi_send_acmd: CMD%02d (0x%08" PRIx32 ")(retry %d)\n", sd_cmd_idx, argument, err_cnt);
        printf("sdcard_spi_send_acmd: CMD%02d (0x%08" PRIx32 ")(retry %d)\n", sd_cmd_idx, argument, err_cnt);
        r1_resu = sdcard_spi_send_cmd(card, SD_CMD_55, SD_CMD_NO_ARG, 0);
        if (R1_VALID(r1_resu) && !R1_ERROR(r1_resu)) {
            puts("here will delay some time\n");

and wait_for_not_busy:

> if (_dyn_spi_rxtx_byte(card, SD_CARD_DUMMY_BYTE, &read_byte) == 1) {
            if (read_byte == 0xFF) {
                DEBUG("_wait_for_not_busy: [OK]\n");
                return true;
            }
            else {
                xtimer_usleep(100);
                DEBUG("_wait_for_not_busy: [BUSY]\n");
            }
        }

If I not do that.The system while die in there like this:
> 2019-11-22 01:30:56,336 # SD_INIT_SEND_ACMD41_HCS
2019-11-22 01:30:57,691 # SD_INIT_SEND_ACMD41_HCS
2019-11-22 01:30:58,181 # SD_INIT_SEND_ACMD41_HCS
2019-11-22 01:30:58,822 # SD_INIT_SEND_ACMD41_HCS
2019-11-22 01:30:59,372 # SD_INIT_SEND_ACMD41_HCS

It is always don't working and I can't wakeup it. 
If I add that, I can run normal:

> 2019-11-22 01:30:56,336 # SD_INIT_SEND_ACMD41_HCS
2019-11-22 01:30:57,691 # SD_INIT_SEND_ACMD41_HCS
2019-11-22 01:30:58,181 # SD_INIT_SEND_ACMD41_HCS
2019-11-22 01:30:58,822 # SD_INIT_SEND_ACMD41_HCS
2019-11-22 01:30:59,372 # SD_INIT_SEND_ACMD41_HCS
2019-11-22 01:34:16,215 # SD_INIT_SEND_ACMD41_HCS
2019-11-22 01:34:16,220 # sdcard_spi_send_acmd: CMD41 (0x40000000)(retry 0)
2019-11-22 01:34:16,220 # here will delay some time
2019-11-22 01:34:16,220 #
2019-11-22 01:34:16,231 # sdcard_spi_send_acmd: CMD41 (0x40000000)(retry 0)
2019-11-22 01:34:16,232 # here will delay some time
2019-11-22 01:34:16,233 #
2019-11-22 01:34:16,235 # SD-card spi driver test application
2019-11-22 01:34:16,243 # insert SD-card and use 'init' command to set card to spi mode
2019-11-22 01:34:16,252 # WARNING: using 'write' or 'copy' commands WILL overwrite data on your sd-card and
2019-11-22 01:34:16,255 # almost for sure corrupt existing filesystems, partitions and contained data!
>

**(I changed max of spi speed and the type of wait_for_not_busy.)**

egarrett39, 2019-11-27:
> #### Description
> I use stmf4 and class 10 microSD(10MB/s) card, but when i use [sdcard_spi_write_blocks ](https://github.com/RIOT-OS/RIOT/blob/1018a6fa674591299c25405761ccaa15008258de/drivers/sdcard_spi/sdcard_spi.c#L813).It USES my 3ms.He doesn't have the speed of an sd card.How can I speed him up.
> 
> ### Useful links
> 1.I use [RIOT/tests/driver_sdcard_spi/](https://github.com/RIOT-OS/RIOT/tree/master/tests/driver_sdcard_spi) to test it.
> 
> 2.Add xtimer to calculate run time in [_write:](https://github.com/RIOT-OS/RIOT/blob/1018a6fa674591299c25405761ccaa15008258de/tests/driver_sdcard_spi/main.c#L265)
> 
> > uint32_t now = xtimer_now_usec();
> > sd_rw_response_t state;
> > int chunks_written = sdcard_spi_write_blocks(card, bladdr, write_buffer, SD_HC_BLOCK_SIZE, 1, &state);
> > uint32_t later = xtimer_now_usec();
> 
> 3.This is results(it run 2.75ms):
> 
> > write 0 adsf 4
> > 2019-11-15 07:30:16,946 #  write 0 adsf 4
> > 2019-11-15 07:30:16,952 # will write 'adsf' (4 chars) at start of block 0
> > 2019-11-15 07:30:16,954 # the rest of the block will be filled with zeros
> > 2019-11-15 07:30:16,960 # time1 sdf_creatnew is 2751
> > 2019-11-15 07:30:16,961 # write block 0 [OK]

ouyangkan, 2019-11-27:
> #### Description
> I use stmf4 and class 10 microSD(10MB/s) card, but when i use [sdcard_spi_write_blocks [...]

I change write_block on step two in https://github.com/RIOT-OS/RIOT/issues/12715#issuecomment-554819191 and spi speed. If you do like this , please put all wait for not busy before all functions .(read_block and write_block)

ouyangkan, 2019-11-27:
> > 1. For 1 and 5, I think Fatfs uses disk_write in mtd_diskio.c.. It packs what I write using a Memory TechnologyDevice and uses disk_write when 512 bytes.
> 
> Yes. What I meant is that you get best performance if you actually write big chunks from your application level at once, i.e. don't write only "a handful" of bytes on each write.
> 
> > 1. I'm also in trouble with 2,3,4. If I only use f_write , I can't complete synchronous brhavious. So I extract F_sync from F_close function to let sdcard be synchronous with stm32. It is successful to save all, but f_sync will cost me 40ms.
> 
> Yes, of course the sdcard driver can't guarantee that only using f_write will write everything to the storage in a synchronous fashion, because the file system logic sits between the sdcard and your application. This can (and, if configured like that, _will_) buffer writes to the storage medium and only finish all its operation with the f_close/f_sync. That your f_sync takes that much time is probably just related to the fact that "a lot of" interaction with the sdcard needs to happen then.
> 
> Other things you could consider:
> Look at (5) from my previous comment, this will increase the sdcard write speed.
> If possible in your case, consider using [another supported file system](https://github.com/RIOT-OS/RIOT/tree/master/sys/include/fs) like littlefs or spiffs. While fatfs might be nicely interoperable for using it on many systems, I doubt it has the best performance but I didn't benchmark it yet.

Ok, I test the [vfs_write](https://github.com/RIOT-OS/RIOT/blob/bf676d4728904881849308114d311d8f20f11ff2/tests/pkg_littlefs/main.c#L390) in the littlefs, unfortunately it costs 8ms each time. And it still need vfs_close.

closed 2020-06-30 by stale[bot]