RIOT-OS/RIOT #12736: gcoap observe: Rejection support (and early-options)

chrysn, 2019-11-18:
#### Description

For OSCORE support (#11761; in particular the test cases executed during the plug tests, but also for negative responses to observation requests) it would be convenient to be able to reject an observation, ie. to state a a resource handler that for the given request, no observation is established. (Such a rejection could have several reasons: In OSCORE the most likely is that the inner request is actually unsuccessful (4.xx or 5.xx), which is still transported on the outside as a 2.xx for privacy reasons. Outside OSCORE it could be that a single resource may not support observation as there is no code to detect changes in its value).

The current implementation places the option that accepts the observation right into the message at gcoap_resp_init time, assuming by the presence of a handler that the observation will be accepted. In addition to the application's inability to reject the observation, this also means that no earlier options than Observe (in responses, that's currently only ETag) can be added to an observation.

### Mitigation suggestion

I suggest that rather than adding the option right away, it be added by the option adding / option finishing calls whenever the Observe option threshold is crossed (ie. if last option number was <9 before and is >= 9 after), and only if the observation bits on the message are right. (For compatibility, messages would be set to accept observations by default; in practice I think that asking the handler to acknowledge an observation makes more sense, as it does not assume that for any handler there exists notification producing code).

That would not only cause the option to be produced only on observable resources, but also in passing solve the issue of ETag support on observations.

### Links

* [OSCORE plug test description](https://ericssonresearch.github.io/OSCOAP/test-spec5.html) (see esp. tests 5-7)

kb2ma, 2019-11-26:
If you need to reject an Observe request on a per-message basis, then that will take some work in `_handle_req()` to manage the observe memo buffer. I expect the best approach is to use pdu->observe_value to communicate from the handler callback to _handle_req(). Below is a high level design.

When a request with an Observe registration option arrives:

1. _handle_req() allocates observe memo buffer space and sets pdu->observe_value to COAP_OBS_REGISTER (0).

2. handler calls coap_clear_observe() to cancel Observe before calling `coap_resp_init()`, so the Observe option is not written to the response

3. _handle_req() reads pdu->observe_value as not COAP_OBS_REGISTER and reclaims the observe memo buffer space

This sequence works also if an application simply does not support Observe for a particular resource. The resource handler systematically calls coap_clear_observe(). Actually, #11587 proposed a compile time mechanism to tag a resource as supporting Observe. We did not merge that PR, and instead expect an app to advertise an observable resource by appending an 'obs' to the link format description for `/.well-known/core`. So the work outlined here will be complementary.

 The other requirement to _defer_ addition of an Observe option in a response will require a  little more work. Once again, let's assume a message arrives with an Observe registration request. I won't include the observe memo buffer in this scenario.
 
 1. handler calls coap_clear_observe() *before* calling coap_resp_init()
 
 2. handler calls coap_opt_add_...() to insert an option number < 6
 
 3. handler calls `coap_set_observe(coap_pkt_t*, COAP_OBS_REGISTER)` (a new function) to reset pdu->observe_value, and also `coap_opt_add_observe(coap_pkt_t*)` (a new function) to write the observe option in the response (what coap_resp_init() does now).
 
 4. _handle_req() doesn't take any special action because pdu->observe_value is COAP_OBS_REGISTER, as expected.
 
 I appreciate your suggestion to dynamically determine when to insert the Observe option in this second scenario. This approach effectively adds a new layer to option handling, and I worry that it will become memory/code intensive as we handle corner cases. Inevitably the urge to extend it also will arise. If we had more resources, then the convenience would be worth it.

closed 2020-06-30 by stale[bot]

chrysn, 2020-11-06:
Working with OSCORE observe has shown another tricky point about this: As long as observations are per-resource as gcoap sees the resource, the only-one-observation-per-resource extends to all items in COAP_MATCH_SUBTREE, which severly limits it use.

Keeping this as "closed / stale" because it's probably best fixed by a larger overhaul of observations.