RIOT-OS/RIOT #12761: samr30 xpro doesn't seem to use its radio ok

kmartinez, 2019-11-20:
I can build gnrc_networking applications for samr30-xpro boards - but never see any packets.
The DEFAULT_CHANNEL is 5 for some reason (US) and I have now changed that to 0 (EU 868) - but there must be something else going on - as it is easy with the samr21-xpro.

@cladmi was involved recently - do you know the trick?
Thanks!

biboc, 2019-11-21:
Samr30-xpro uses AT86RF212B transceiver. Depending on the antenna frequency you use, you should configure channel and page to be able to communicate

kmartinez, 2019-11-21:
Thanks! I didn't realise the "page" needed setting - for 868.3MHz it could be page=0
it is set in drivers/include/at86rf2xx.h to IEEE802154_DEFAULT_SUBGHZ_PAGE
defined in  sys/include/net/ieee802154.h as value 2
as far as I can see as long as both nodes are the same they should communicate...  I assume someone has tested these boards? they are more unusual than the 21R

kmartinez, 2019-11-21:
some success! I set PAGE=0 and CHANNEL=0 and listened with a radio scanner

kmartinez, 2019-12-03:
I guess nobody is using these regularly? in the EU we should certainly be setting chan=0 for 868MHz range.. I am not sure how we can make that auto-select on build..

jue89, 2019-12-10:
I had problems with that board, as well. I realized reducing the internal SPI clock from 5MHz to 1MHz solved my problem. But I haven't investigated this problem further ... 5MHz shouldn't be too high according to the datasheet.

Could you try this little patch down below? Does this increase the stability? If so, we should have a deeper looks whats going on here ...

```diff
diff --git a/boards/samr30-xpro/include/board.h b/boards/samr30-xpro/include/board.h
index d6eb0fc9aabb01e906554ef90862fef007aff3ac..9067d5215e47a5203130e0ec44faa7485104db4e 100644
--- a/boards/samr30-xpro/include/board.h
+++ b/boards/samr30-xpro/include/board.h
@@ -36,7 +36,7 @@ extern "C" {
 #define AT86RF2XX_PARAM_INT         GPIO_PIN(PB, 0)
 #define AT86RF2XX_PARAM_SLEEP       GPIO_PIN(PA, 20)
 #define AT86RF2XX_PARAM_RESET       GPIO_PIN(PB, 15)
-#define AT86RF2XX_PARAM_SPI_CLK     SPI_CLK_5MHZ
+#define AT86RF2XX_PARAM_SPI_CLK     SPI_CLK_1MHZ
 /** @}*/
 
 /**
```

jue89, 2020-03-05:
I think #13537 should solve this.

kmartinez, 2020-03-05:
so a question on page/chan defaults - we could put the switch in 
boards/samr30-xpro/include/board.h
or
makefiles/default-radio-settings.inc.mk
as it seems 802.15.4 page numbers and channel nums are generic across boards? (this only appies for sub-GHz)

jue89, 2020-03-06:
I wouldn't change the DEFAULT_CHANNEL and DEFAULT_PAGE in the board.h file since the `samr30-xpro` board can also be used in the US. All used RF components support dual ISM band (EU and US).

But I would rather ensure that it becomes more obvious that one must check and decide which channel to use. (I accidentally used the board with the US frequency here in Germany for some days ;))

Maybe that's a good topic for the Kconfig task force ;)