RIOT-OS/RIOT #12763: [TRACKING] Fixes for automatic tests of ESP32 boards.

gschorcht, 2019-11-21:
#### Description

Automatic tests for ESP32 didn't work because of problems to reset ESP32 boards using `esptool.py`. With PR #12752 it will become possible to reset ESP32 boards and to perform automatic tests.

However, a number of tests are not working due to several open problems. This issue tracks the tests that are not working for ESP32 and their fixes.

#### Open Test Problems

- [] tests/cond_order (PR #12753) 
- [] tests/heap_cmd (PR #12753) 
- [] tests/log_color (PR #12044)
- [] tests/log_printfnoformat
- [] tests/mutex_order (PR #12753) 
- [] tests/netstats_l2 (PR #12756)
- [] tests/periph_timer (PR #12757)
- [] tests/pkg_spiffs (PR #12758)
- [] tests/pkg_littlefs (PR #12758)
- [] tests/posix_semaphore
- [] tests/rmutex_cpp (PR #12753) 
- [] tests/rmutex (PR #12753) 
- [] tests/shell (PR #12753) 
- [] tests/ssp

aabadie, 2020-01-28:
@gschorcht, there's only one item left in this tracking issue. Do you think this is fixable ? Or maybe it's already fixed ?

gschorcht, 2020-01-28:
Unfortunately, the reason is still not really clear. It seems that the system locks when the hardware timer is read while the interrupts are disabled in `BENCHMARK_FUNC`. When I remove the `irq_disable`/`irq_enable` in `BENCHMARK_FUNC`, everything is fine. Maybe it's a race condition between the hardware timer read operation and a timer interrupt that is pending. I tried to debug it with OCD but when the system locks, OCD is not working anymore because interrupts are not handled anymore, even not high priority interrupts.

BTW, when I was investigating the problem I was wondering whether it is really a good idea to disable all interrupts during the benchmark test. There are platforms that require at least the timer interrupts to get benchmark tests working. For example, the results you get with ATmega boards and the `tests/periph_gpio` benchmark tests are completely nonsense because timer is not running during the benchmark tests. I know, ISRs must not affect the benchmark results. But if the timer interrupt is required to keep the timer working, disabling the interrupts is counter productive.

closed 2026-03-06 by gschorcht