RIOT-OS/RIOT #12779: wiki: remove pages which are part of RIOT/*.md

waehlisch, 2019-11-22:
The content of the following wiki pages is documented in https://github.com/RIOT-OS/RIOT/blob/master/CONTRIBUTING.md:

- https://github.com/RIOT-OS/RIOT/wiki/Contributing-to-RIOT 
- https://github.com/RIOT-OS/RIOT/wiki/Guidelines%20for%20Creating%20a%20Good%20Pull%20Request 

I suggest to remove the wiki pages to reduce redundancy and improve maintainability.
In case CONTRIBUTING.md lacks content, we should move content from the wiki pages to this file.

aabadie, 2019-11-22:
+1 for the idea. IMHO contributing.md already contains what's needed to start contributing efficiently. The wiki page contains advanced documentation but I don't find them very useful. So I would just drop the wiki page.

miri64, 2019-11-22:
There is also https://github.com/RIOT-OS/RIOT/wiki/Development-procedures. That one basically is IMHO just a bullet pointed version of CONTRIBUTING.md and also contains some outdated information.

> 4. […] Use Labels to classify your pull request (only maintainers can assign labels). […]

[Members](https://github.com/orgs/RIOT-OS/teams/member) can set labels as well now.

> 5. Comments on a pull request should be added to the request itself, and not to the commit.

The current GitHub interface makes it harder to accidentally do this, but I guess it doesn't harm to be mentioned.

> 6. Only RIOT maintainers have commit rights to the main RIOT repository. However, even they have to create pull requests to submit their changes. A RIOT contributor cannot acknowledge his own pull request. All current RIOT maintainers are listed in the wiki (Maintainers).

Branch protection ensures that no-one can push to master now without passing the CI (via a PR).

> 11. When porting for a new board, please create a wiki page at first if there isn't already one. The wiki page should include a link to an issue in the tracker to coordinate porting tasks. To make sure everybody catches this information, please put it on the developers mailing list once. This should avoid overlaps between the work of multiple developers.

Board doc is now in the documentation.

miri64, 2019-11-22:
> +1 for the idea. IMHO contributing.md already contains what's needed to start contributing efficiently. The wiki page contains advanced documentation but I don't find them very useful. So I would just drop the wiki page.

I think there is some valuable information that could be ported to CONTRIBUTING.md, like the [getting started section](https://github.com/RIOT-OS/RIOT/wiki/Contributing-to-RIOT#getting-started) (which however should rather link to the https://doc.riot-os.org version), or (an updated version of) [help wanted](https://github.com/RIOT-OS/RIOT/wiki/Contributing-to-RIOT#help-wanted). Currently, the `CONTRIBUTING.md` only contains rules how to contribute, but I think its good to also enable contributions via this document.

The git cheatsheet in CONTRIBUTING.md could be amended by how to resolve a merge conflict, as it is documented in the guidelines.

waehlisch, 2019-11-22:
@miri64 a bullet point check list could also be added to CONTRIBUTING.md,

waehlisch, 2019-11-22:
@miri64 @aabadie my point is: one single source is sufficient. if there is valuable information in the wiki page that is not included in CONTRIBUTING.md, we should just extend CONTRIBUTING.md.

miri64, 2020-01-16:
#12786 was merged.

miri64, 2020-01-16:
I also removed the pages in question.

closed 2020-01-16 by miri64