RIOT-OS/RIOT #12780: gcoap: error signaling for gcoap_req_send() insufficient

haukepetersen, 2019-11-22:
#### Description
Currently, `gcoap_req_send()` only has a single return value for all possible error cases. This is not sufficient, as it strips information that is mandatory for certain use cases of the API.

Example: flooding out large amounts of data. When sending a large amount of user data over CoAP as fast as possible (e.g. flood or throughput tests), one would e.g. use code like:
```c
    do {
        res = gcoap_req_send(_txbuf, (hdr_len + len), remote, NULL);
    } while (res == ERROR_CODE_FOR_BUFFER_FULL);
```
Hoever, as `gcoap_req_res()` returns `0` for any kind of error, this code does break when for example exceeding the link layers MTU size with our payload -> `sock_udp_send()` returning `-EMSGSIZE`.

#### Steps to reproduce the issue
Run the code above for large payload sizes (payload exceeding you link capacity, e.g. > 1500 on `native`). The loop is never left.

#### Expected results
`gcoap` should return differentiated errors so the user can react accordingly.

#### Actual results
see above.

#### Versions
any :-)

haukepetersen, 2019-11-22:
Oh, forgot to state: I though about creating a PR to improve this situation, but for many internal errors I can't (quickly) judge, how they can be best grouped. In the end I hope we agree, that we do not want to expose any possible error as unique value, but best group errors to a limited number or return codes, right?

kb2ma, 2019-11-25:
I agree that the return value is the best way to provide this information, and gcoap should take advantage of it. Yes, I agree errors should be grouped, and I would think about identifying groups based on when a user may differentiate their response to the error.

closed 2020-06-28 by stale[bot]