RIOT-OS/RIOT #12796: Test failures in CI

maribu, 2019-11-23:
#### Description

Recently, a two tests on the CI failed without any apparent reason. The failed test were completly unrelated to the PR; one was a doc only PR. In order to not loose the error messages when restarting the build, I post them here. Both were run on `mobi8.inet.haw-hamburg.de`, in both cases a timeout error occurred. Hopefully, there was just to much load on the system and the test would have passed when given more time.

##### PR https://github.com/RIOT-OS/RIOT/pull/12781

Full output: https://ci.riot-os.org/RIOT-OS/RIOT/12781/2c5626b6c0c42d0d77b29162657c62c7fb39def1/output.html
<details><summary>Failure when running `tests/pkg_spiffs` on native</summary>

```
-- running on worker mobi8.inet.haw-hamburg.de thread 15, build number 3858.
make: Entering directory '/tmp/dwq.0.09063174533960616/5756ea57231e45e33bb21b8e2261e6e1/tests/pkg_spiffs'
Building application "tests_pkg_spiffs" for "native" with MCU "native".

rm -Rf /tmp/dwq.0.09063174533960616/5756ea57231e45e33bb21b8e2261e6e1/build/pkg/spiffs
mkdir -p /tmp/dwq.0.09063174533960616/5756ea57231e45e33bb21b8e2261e6e1/build/pkg/spiffs
/tmp/dwq.0.09063174533960616/5756ea57231e45e33bb21b8e2261e6e1/dist/tools/git/git-cache clone "https://github.com/pellepl/spiffs.git" "287148c46587089c4543a21eef2d6e9e14b88364" "/tmp/dwq.0.09063174533960616/5756ea57231e45e33bb21b8e2261e6e1/build/pkg/spiffs"
git-cache: cloning from cache. tag=commit287148c46587089c4543a21eef2d6e9e14b88364-42152
touch /tmp/dwq.0.09063174533960616/5756ea57231e45e33bb21b8e2261e6e1/build/pkg/spiffs/.git-downloaded
if [ 287148c46587089c4543a21eef2d6e9e14b88364 != 287148c46587089c4543a21eef2d6e9e14b88364 ] ; then \
	git -C /tmp/dwq.0.09063174533960616/5756ea57231e45e33bb21b8e2261e6e1/build/pkg/spiffs clean -xdff ; \
	git -C /tmp/dwq.0.09063174533960616/5756ea57231e45e33bb21b8e2261e6e1/build/pkg/spiffs fetch "https://github.com/pellepl/spiffs.git" "287148c46587089c4543a21eef2d6e9e14b88364" ; \
	git -C /tmp/dwq.0.09063174533960616/5756ea57231e45e33bb21b8e2261e6e1/build/pkg/spiffs checkout -f 287148c46587089c4543a21eef2d6e9e14b88364 ; \
	touch /tmp/dwq.0.09063174533960616/5756ea57231e45e33bb21b8e2261e6e1/build/pkg/spiffs/.git-downloaded ; \
fi
git -C /tmp/dwq.0.09063174533960616/5756ea57231e45e33bb21b8e2261e6e1/build/pkg/spiffs checkout -f 287148c46587089c4543a21eef2d6e9e14b88364
HEAD is now at 287148c465 0.3.7 documenting
git -c user.email=buildsystem@riot -c user.name="RIOT buildsystem" -C /tmp/dwq.0.09063174533960616/5756ea57231e45e33bb21b8e2261e6e1/build/pkg/spiffs am --no-gpg-sign --ignore-whitespace "/tmp/dwq.0.09063174533960616/5756ea57231e45e33bb21b8e2261e6e1/pkg/spiffs"/patches/*.patch
Applying: Constify pointer for write operations
touch /tmp/dwq.0.09063174533960616/5756ea57231e45e33bb21b8e2261e6e1/build/pkg/spiffs/.git-patched
"make" -C /tmp/dwq.0.09063174533960616/5756ea57231e45e33bb21b8e2261e6e1/build/pkg/spiffs/src -f /tmp/dwq.0.09063174533960616/5756ea57231e45e33bb21b8e2261e6e1/pkg/spiffs/Makefile.spiffs
sha1sum /tmp/dwq.0.09063174533960616/5756ea57231e45e33bb21b8e2261e6e1/tests/pkg_spiffs/tests/01-run.py /tmp/dwq.0.09063174533960616/5756ea57231e45e33bb21b8e2261e6e1/build/tests_pkg_spiffs.elf > /tmp/dwq.0.09063174533960616/5756ea57231e45e33bb21b8e2261e6e1/build/test-input-hash.sha1
   text	   data	    bss	    dec	    hex	filename
 101854	    932	  49420	 152206	  2528e	/tmp/dwq.0.09063174533960616/5756ea57231e45e33bb21b8e2261e6e1/build/tests_pkg_spiffs.elf
make: Leaving directory '/tmp/dwq.0.09063174533960616/5756ea57231e45e33bb21b8e2261e6e1/tests/pkg_spiffs'
make: Entering directory '/tmp/dwq.0.09063174533960616/5756ea57231e45e33bb21b8e2261e6e1/tests/pkg_spiffs'
/tmp/dwq.0.09063174533960616/5756ea57231e45e33bb21b8e2261e6e1/build/tests_pkg_spiffs.elf  
RIOT native interrupts/signals initialized.
LED_RED_OFF
LED_GREEN_ON
RIOT native board initialized.
RIOT native hardware initialization complete.

main(): This is RIOT! (Version: buildtest)
.......Timeout in expect script at "child.expect(r'OK \(\d+ tests\)')" (tests/pkg_spiffs/tests/01-run.py:14)

/tmp/dwq.0.09063174533960616/5756ea57231e45e33bb21b8e2261e6e1/tests/pkg_spiffs/../../Makefile.include:704: recipe for target 'test' failed
make: *** [test] Error 1
make: Leaving directory '/tmp/dwq.0.09063174533960616/5756ea57231e45e33bb21b8e2261e6e1/tests/pkg_spiffs'
-- build directory size: 2.1M
```

</details>

#### PR https://github.com/RIOT-OS/RIOT/pull/12790

Full output: https://ci.riot-os.org/RIOT-OS/RIOT/12790/4347f720f5e113fca43421f0ef83fe4b166e99e6/output.html

<details><summary>Failure when running `tests/mutex_unlock_and_sleep` on native</summary>

```
-- running on worker mobi8.inet.haw-hamburg.de thread 6, build number 3431.
make: Entering directory '/tmp/dwq.0.09063174533960616/6609c4c8054c57e30d720d4902e98551/tests/mutex_unlock_and_sleep'
Building application "tests_mutex_unlock_and_sleep" for "native" with MCU "native".

sha1sum /tmp/dwq.0.09063174533960616/6609c4c8054c57e30d720d4902e98551/tests/mutex_unlock_and_sleep/tests/01-run.py /tmp/dwq.0.09063174533960616/6609c4c8054c57e30d720d4902e98551/build/tests_mutex_unlock_and_sleep.elf > /tmp/dwq.0.09063174533960616/6609c4c8054c57e30d720d4902e98551/build/test-input-hash.sha1
   text	   data	    bss	    dec	    hex	filename
  24742	    568	  55876	  81186	  13d22	/tmp/dwq.0.09063174533960616/6609c4c8054c57e30d720d4902e98551/build/tests_mutex_unlock_and_sleep.elf
make: Leaving directory '/tmp/dwq.0.09063174533960616/6609c4c8054c57e30d720d4902e98551/tests/mutex_unlock_and_sleep'
make: Entering directory '/tmp/dwq.0.09063174533960616/6609c4c8054c57e30d720d4902e98551/tests/mutex_unlock_and_sleep'
/tmp/dwq.0.09063174533960616/6609c4c8054c57e30d720d4902e98551/build/tests_mutex_unlock_and_sleep.elf  
RIOT native interrupts/signals initialized.
LED_RED_OFF
LED_GREEN_ON
RIOT native board initialized.
RIOT native hardware initialization complete.

main(): This is RIOT! (Version: buildtest)
Timeout in expect script at "child.expect(r"\[ALIVE\] alternated \d+k times.")" (tests/mutex_unlock_and_sleep/tests/01-run.py:15)

/tmp/dwq.0.09063174533960616/6609c4c8054c57e30d720d4902e98551/tests/mutex_unlock_and_sleep/../../Makefile.include:704: recipe for target 'test' failed
make: *** [test] Error 1
make: Leaving directory '/tmp/dwq.0.09063174533960616/6609c4c8054c57e30d720d4902e98551/tests/mutex_unlock_and_sleep'
-- build directory size: 676K

```

</details>

kaspar030, 2019-11-23:
@smlng mobi8 is up to 25sec average build time, up from ~6 before. Can you take a look?

I think I'l pause it for now.

maribu, 2019-11-25:
@kaspar030, @smlng: Feel free to close this issue when you no longer need this.

kaspar030, 2019-11-25:
> @kaspar030, @smlng: Feel free to close this issue when you no longer need this.

Seems like mobi8 is back to its usual performance.

closed 2019-11-25 by kaspar030