RIOT-OS/RIOT #12807: make: rebuild doesn't apply to header files

miri64, 2019-11-25:
#### Description
When rebuilding with a changed value in a header this changed value is not taken into account.

#### Steps to reproduce the issue
I created [a very simple test application](https://gist.github.com/miri64/f5a47100b363fe6b6178652480b659a2). When building and running for `native` in current master (d0d6e53), I first get as expected:

```
$ make clean all term
make: Entering directory '/home/mlenders/Repositories/RIOT-OS/RIOT/tests/header_change'
Building application "tests_" for "native" with MCU "native".

"make" -C /home/mlenders/Repositories/RIOT-OS/RIOT/boards/native
"make" -C /home/mlenders/Repositories/RIOT-OS/RIOT/boards/native/drivers
"make" -C /home/mlenders/Repositories/RIOT-OS/RIOT/core
"make" -C /home/mlenders/Repositories/RIOT-OS/RIOT/cpu/native
"make" -C /home/mlenders/Repositories/RIOT-OS/RIOT/cpu/native/periph
"make" -C /home/mlenders/Repositories/RIOT-OS/RIOT/drivers
"make" -C /home/mlenders/Repositories/RIOT-OS/RIOT/sys
"make" -C /home/mlenders/Repositories/RIOT-OS/RIOT/sys/auto_init
   text	   data	    bss	    dec	    hex	filename
  20251	    568	  47716	  68535	  10bb7	/home/mlenders/Repositories/RIOT-OS/RIOT/tests/header_change/bin/native/tests_.elf
/home/mlenders/Repositories/RIOT-OS/RIOT/tests/header_change/bin/native/tests_.elf  
RIOT native interrupts/signals initialized.
LED_RED_OFF
LED_GREEN_ON
RIOT native board initialized.
RIOT native hardware initialization complete.

main(): This is RIOT! (Version: 2020.01-devel-1051-gd0d6e)
0205
```

If I then change `GNRC_NETAPI_MSG_TYPE_ACK` *in the header* (I incremented it by one to `0x0206`) and then rebuild *without clean* I still get the output `0205`:

```
make: Entering directory '/home/mlenders/Repositories/RIOT-OS/RIOT/tests/header_change'
Building application "tests_header_change" for "native" with MCU "native".

"make" -C /home/mlenders/Repositories/RIOT-OS/RIOT/boards/native
"make" -C /home/mlenders/Repositories/RIOT-OS/RIOT/boards/native/drivers
"make" -C /home/mlenders/Repositories/RIOT-OS/RIOT/core
"make" -C /home/mlenders/Repositories/RIOT-OS/RIOT/cpu/native
"make" -C /home/mlenders/Repositories/RIOT-OS/RIOT/cpu/native/periph
"make" -C /home/mlenders/Repositories/RIOT-OS/RIOT/cpu/native/vfs
"make" -C /home/mlenders/Repositories/RIOT-OS/RIOT/drivers
"make" -C /home/mlenders/Repositories/RIOT-OS/RIOT/drivers/periph_common
"make" -C /home/mlenders/Repositories/RIOT-OS/RIOT/sys
"make" -C /home/mlenders/Repositories/RIOT-OS/RIOT/sys/auto_init
   text	   data	    bss	    dec	    hex	filename
  23303	    568	  47716	  71587	  117a3	/home/mlenders/Repositories/RIOT-OS/RIOT/tests/header_change/bin/native/tests_header_change.elf
/home/mlenders/Repositories/RIOT-OS/RIOT/tests/header_change/bin/native/tests_header_change.elf  
RIOT native interrupts/signals initialized.
LED_RED_OFF
LED_GREEN_ON
RIOT native board initialized.
RIOT native hardware initialization complete.

main(): This is RIOT! (Version: 2020.01-devel-1051-gd0d6e)
0205
```

I confirmed that the same happens on `samr21-xpro` and also with RIOT versions as old as 2017.01.

#### Expected results
After the second build, the app should show `0206` (if `GNRC_NETAPI_MSG_TYPE_ACK` was incremented e.g.).

#### Actual results
After the second build, the app still shows `0205`.

#### Versions
```

Operating System Environment
-----------------------------
       Operating System: "Ubuntu" "19.10 (Eoan Ermine)"
                 Kernel: Linux 5.3.0-23-generic x86_64 x86_64

Installed compiler toolchains
-----------------------------
             native gcc: gcc (Ubuntu 9.2.1-9ubuntu2) 9.2.1 20191008
      arm-none-eabi-gcc: arm-none-eabi-gcc (GNU Tools for Arm Embedded Processors 8-2018-q4-major) 8.2.1 20181213 (release) [gcc-8-branch revision 267074]
                avr-gcc: avr-gcc (GCC) 5.4.0
       mips-mti-elf-gcc: missing
             msp430-gcc: msp430-gcc (GCC) 4.6.3 20120301 (mspgcc LTS 20120406 unpatched)
   riscv-none-embed-gcc: missing
   xtensa-esp32-elf-gcc: missing
   xtensa-lx106-elf-gcc: missing
                  clang: clang version 9.0.0-2 (tags/RELEASE_900/final)

Installed compiler libs
-----------------------
   arm-none-eabi-newlib: "3.0.0"
    mips-mti-elf-newlib: missing
riscv-none-embed-newlib: missing
xtensa-esp32-elf-newlib: missing
xtensa-lx106-elf-newlib: missing
               avr-libc: "2.0.0" ("20150208")

Installed development tools
---------------------------
                  cmake: cmake version 3.13.4
               cppcheck: Cppcheck 1.88
                doxygen: 1.8.13
                    git: git version 2.20.1
                   make: GNU Make 4.2.1
                openocd: Open On-Chip Debugger 0.10.0+dev-00910-g4dbcb1e7 (2019-06-17-16:24)
                 python: Python 2.7.17rc1
                python2: Python 2.7.17rc1
                python3: Python 3.7.5rc1
                 flake8: 3.7.8 (mccabe: 0.6.1, pycodestyle: 2.5.0, pyflakes: 2.1.1) CPython 3.7.5rc1 on Linux
             coccinelle: missing
```

I checked this with RIOT versions d0d6e53ff4, e4fb634e3f (2018.01), and 158463e696 (2017.01).

kaspar030, 2019-11-25:
> When rebuilding with a changed value in a header this changed value is not taken into account.

Is this with ccache enabled? Can you confirm whether this is an issue with ccache disabled (`unset CCACHE; make clean; make; <change header>; make;`?

miri64, 2019-11-25:
Mhhh exactly what I was thinking, after I read https://github.com/RIOT-OS/RIOT/pull/12773#issuecomment-558189376 ^^. I'll have a look tomorrow.

miri64, 2019-11-25:
(btw... `print_toolchain_versions.sh` should print the `ccache` version as well + if `CCACHE` is set in the environment ;-))

leandrolanzieri, 2019-11-25:
I can confirm this. With ccache enabled after the second build I still get the old value. Statistics of ccache only increment the `called for link` and `no input file` every time I call `make`.

When the [depend mode](https://ccache.dev/manual/latest.html#_the_depend_mode) is enabled (`export CCACHE_DEPEND`) the build is updated successfully.

leandrolanzieri, 2019-11-25:
> (btw... `print_toolchain_versions.sh` should print the `ccache` version as well + if `CCACHE` is set in the environment ;-))

+1

Mine: `ccache version 3.6`

miri64, 2019-11-26:
Confirmed that when CCACHE is deactivated (`unset CCACHE`) it works. However when re-activating setting `CCACHE_DEPEND` it does not work. My `ccache` version is  3.7.3 and my environment is configured as follows:

```
CCACHE_MAXSIZE=8G
CCACHE_DIR=/tmp/ccache
CCACHE_COMPRESS=true
CCACHE=ccache
CCACHE_DEPEND=
```

closed 2019-12-02 by cgundogan

miri64, 2019-12-02:
With #12840 merged this issue is confirmed to be fixed now :-).