RIOT-OS/RIOT #12834: make: piping STDOUT updates `riotbuild.h.in` and leads to full rebuild

cgundogan, 2019-11-28:
#### Description
By default, `-fdiagnostics-color` is passed to the `CFLAGS` (if supported) and recorded in the header section of `$(BINDIR)/riotbuild/riotbuild.h.in`.

If the output of `make` is piped, then `-fdiagnostics-color` is not passed to the `CFLAGS` (introduced in #12127). This leads to a modified `riotbuild.h.in` and then to a full rebuild of all modules, since `riotbuild` is a hard dependency of every module.

#### Steps to reproduce the issue
1. `QUIET=0 make all` and directly after that ..
2. `QUIET=0 make all | cat -`

#### Expected results
* The first make invocation should build all modules
* The second invocation should *not* rebuild all modules

#### Actual results
* The second invocation rebuilds all modules

fjmolinas, 2020-01-08:
A workaround is setting `CC_NOLOR=1`.

But do you really want colors when piping? To me it makes sense for them to be deactivated.

cgundogan, 2020-01-09:
>But do you really want colors when piping? To me it makes sense for them to be deactivated.

I am all for disabling colors when piping. Would still be good, if we could convince `make` to not recompile in such cases.

fjmolinas, 2020-01-09:
What about `OPTIONAL_CFLAGS += -fdiagnostics-color=auto`?

Instead of:

```
# ifeq ($(CC_NOCOLOR),0)
OPTIONAL_CFLAGS += -fdiagnostics-color=auto
# endif
```

Seems to work locally

fjmolinas, 2020-01-09:
```
-fdiagnostics-color[=WHEN]
-fno-diagnostics-color

    Use color in diagnostics. WHEN is ‘never’, ‘always’, or ‘auto’. The default depends on how the compiler has been configured, it can be any of the above WHEN options or also ‘never’ if GCC_COLORS environment variable isn’t present in the environment, and ‘auto’ otherwise. ‘auto’ means to use color only when the standard error is a terminal. The forms -fdiagnostics-color and -fno-diagnostics-color are aliases for -fdiagnostics-color=always and -fdiagnostics-color=never, respectively. 
```

cgundogan, 2020-01-09:
@fjmolinas that might be a good solution. Care to PR? (:

fjmolinas, 2020-01-09:
> @fjmolinas that might be a good solution. Care to PR? (:+1: 

It doesn't seem to be supported by all compilers.. not working on msp430 (still trying to figure out if its a viable solution)

fjmolinas, 2020-01-10:
Opened #13071 to address.

closed 2020-01-10 by leandrolanzieri