RIOT-OS/RIOT #12836: udp: receive broken

smlng, 2019-11-28:
#### Description

Receiving network packets seems to be broken, i.e.
UDP server in gnrc_networking does not receive any messages.

#### Steps to reproduce the issue

- Build gnrc_networking for native
- create 2 tap devices
- run 2 instances of RIOT on tap0 and tap1
- run `udp server start 1234` on first instance
- run `udp send <ip of server> 1234 Foobar` on 2nd instance

#### Expected results

UDP server dumps packet.

#### Actual results

no output

#### Versions

RIOT master with commit f4876013cde376cab8233f817fe5e0521bcdb5d0

smlng, 2019-11-28:
nevermind seems to be OS specific issue

miri64, 2019-11-28:
Found this stuck to milhouse (the Apple iMac we intended to use as a test platform for Mac OSX someday (TM)) 

[image]

So I think there is value in keeping this open ;-)

smlng, 2019-11-29:
actually this wasn't about macOS (this time) but CentOS 7, and the problem there was that by default some net filter rules are present for ipv6 which blocks forwarding via the tap bridge.

To enable ipv6 forwarding via the bridge these commands need to be run

```
/sbin/sysctl -w net.bridge.bridge-nf-call-ip6tables=0
/sbin/sysctl -w net.bridge.bridge-nf-call-iptables=0
/sbin/sysctl -w net.bridge.bridge-nf-call-arptables=0
```

fjmolinas, 2020-01-10:
@miri64 @smlng I'm not sure I understand the status here, is it then working on `macOS`?

smlng, 2020-01-13:
@fjmolinas well don't know bc native is broken on macOS since Catalina (no 32bit support) anyway. Also networking with taps and bridge was somewhat limited anyway, bc tun/tap support in macOS is broken (not maintained) in macOS as well.

However, I hastily opened this issue w/o proper description and @miri64 assumed it's related to macOS - as I typically work with that. However, this was actually CentOS 7.

smlng, 2020-01-13:
fixed title, I guess that should rather go into some wiki or similar, bc it's an OS config "problem" - at least if you use a Linux-distro like CentOS 7.

fjmolinas, 2020-01-13:
> fixed title, I guess that should rather go into some wiki or similar, bc it's an OS config "problem" - at least if you use a Linux-distro like CentOS 7.

We could add it to `gnrc_networking/README`

miri64, 2020-01-14:
> > fixed title, I guess that should rather go into some wiki or similar, bc it's an OS config "problem" - at least if you use a Linux-distro like CentOS 7.
> 
> We could add it to `gnrc_networking/README`

I'd rather prefer it to go into https://github.com/RIOT-OS/RIOT/wiki/Family%3A-native. This is really very platform dependent and there is where all the platform dependent native stuff is at. A random application's readme seems to be the completely wrong place to doc this.

miri64, 2020-07-06:
Added a blurb to the wiki for [OSX](https://github.com/RIOT-OS/RIOT/wiki/Family:-native#os-x) and [CentOS](https://github.com/RIOT-OS/RIOT/wiki/Family:-native#centos)

closed 2020-07-06 by miri64