RIOT-OS/RIOT #12858: KW2XRF: Broken network communication without netstats-l2

MatthiasBraeuer, 2019-12-02:
#### Description
When `USEMODULE += netstats_l2` is removed from the Makefile in the `gnrc_networking` example, the packet loss increases to 100% when executing the `ping6` command.

The issue seems to be that the state machine of the `kw2xrf radio driver` gets "confused".

The module `netstats-l2` seems to set/clear a register bit which is needed for correct behavior (see [kw2xrf_getset.c#L411](https://github.com/RIOT-OS/RIOT/blob/master/drivers/kw2xrf/kw2xrf_getset.c#L411)  ).
This bit is only set/cleared when `KW2XRF_OPT_TELL_TX_END` is set, which is only set when the `netstat-l2` module is compiled (see [gnrc_netif.c#L1208](https://github.com/RIOT-OS/RIOT/blob/master/sys/net/gnrc/netif/gnrc_netif.c#L1208) and
[kw2xrf_getset.c#L466](https://github.com/RIOT-OS/RIOT/blob/master/drivers/kw2xrf/kw2xrf_getset.c#L466)).

This issue occurred on a `Phytec phyNODE KW22` 

#### Steps to reproduce the issue
1.  Change into the `gnrc_networking` example and remove following line from the Makefile: `USEMODULE += netstats_l2`
2.  Flash the `gnrc_networking` example on two `Phytec phyNODE KW22` board with following command: `BOARD=pba-d-01-kw2x make all flash term`
3.  Ping one board from the other via `ping6 <ipv6-address> -c 30`
4.  After a while/multiple executions of ping6 the packet loss increases (-> the pinged node does not respond anymore), until it is 100% all the time.

#### Expected results
The ping6 execution has a constant normal packet loss rate < 100%.

#### Actual results
After a few successful pings, the packet loss increases to 100% and is not recovering.

smlng, 2019-12-02:
Thanks for reporting, good analysis!

fjmolinas, 2020-07-03:
@jia200x do you have multiple `pba-d-01-kw2x`? I can't reproduce with only one.

closed 2020-07-06 by PeterKietzmann