RIOT-OS/RIOT #12859: lwip: multiple sock_udp_send() return ENOMEM. (possible memory leak?)

wosym, 2019-12-02:
#### Description

When making multiple sock_udp_send() calls, after n times, -12 (ENOMEM) is returned and sending fails. This happens both on target (nucleo-f207fg), and on **native**.
Also tried using mem_pools, but the behavior remained unchanged. 
Issue is the same with IPv4 and IPv6 lwip.
GNRC does not show this behavior and works fine.

#### Steps to reproduce the issue
in test/lwip shell: _udp send fe80::9333:621c:1f7c:96fd 80 5555 100 100000_
(Or in own code: make a loop and call sock_udp_send())
#### Expected results
Ethernet frame with UDP payload "5555" sent 100 times to the specified IP.

#### Actual results
First n calls: successful, also show up in Wireshark (with n in [5, 20])
After that: "could not send" ( in example), or -12 (ENOMEM) returned from call (when calling it in own code)

#### Versions
OS: Ubuntu 19.10
native gcc: gcc (Ubuntu 9.2.1-9ubuntu2) 9.2.1 20191008
arm-none-eabi-gcc: arm-none-eabi-gcc (GNU Tools for Arm Embedded Processors 9-2019-q4-major) 9.2.1 20191025 (release) [ARM/arm-9-branch revision 277599]

wosym, 2019-12-02:
@miri64

miri64, 2019-12-02:
I would need to debug this myself. From just looking at my part of the code I don't really see where the leak could be.

wosym, 2019-12-02:
I might be able to provide a clue, because I happened to fix the bug by accident, only 20 minutes after I opened this issue. 

Initially I wrote my code more or less parallel to the [Simple UDP client without reply](https://riot-os.org/api/group__net__sock__udp.html). Only a single line of code, with NULL passed as a `sock_udp_t` variable.
After hours of looking for the cause of this behavior in the wrong places, I noticed that it did the same in the examples, and also did this on native. That's when I opened the issue and stopped trying to fix the issue, but instead continued with my work and just ignored the bug for now.
I added code to make the equivalent of the complete Simple UDP echo client. Soon after testing, I noticed that it kept sending without problem. The bug was gone! 

Here's my hypothesis of what's happening: the sock_udp_send() opens a sock and expects it to be closed. This happens when the complete echo-code is implemented. However, when you only do the single line, and pass NULL as the `sock_udp_t` parameter, it is never explicitly closed, because, well... it shouldn't be open in the first place. My guess is that this isn't handled properly somehow. Maybe it allocates memory for a socket, while it shouldn't? 
Anyways, this is pure speculation. I didn't check the relevant lwip source yet. The hypothesis is purely based on the behavior I observed today.

miri64, 2019-12-02:
You might be right about that. There is a `netconn` (lwIP's equivalent to `sock` that `sock` just wraps) created if none is provided

https://github.com/RIOT-OS/RIOT/blob/0151655de5b3fd8ed28899cdd2dd4cb2d3587b28/pkg/lwip/contrib/sock/lwip_sock.c#L505-L510

But no close for that in sight. Even worse: that the `conn` is interpreted as `NULL` pointer as expected, seems to be pure luck oO:

https://github.com/RIOT-OS/RIOT/blob/0151655de5b3fd8ed28899cdd2dd4cb2d3587b28/pkg/lwip/contrib/sock/udp/lwip_sock_udp.c#L131-L132

toonst, 2019-12-11:
> You might be right about that. There is a `netconn` (lwIP's equivalent to `sock` that `sock` just wraps) created if none is provided
> 
Is this the desired behaviour? I would expect `lwip_sock_send` to fail if lwip_sock_create` was not called before.

miri64, 2019-12-12:
> s this the desired behaviour? I would expect `lwip_sock_send` to fail if `lwip_sock_create` was not called before.

No it is specified as such. You *can* send without a sock (e.g. for simple fire and forget schemes).

miri64, 2019-12-12:
See https://doc.riot-os.org/group__net__sock__udp.html#ga89562920ad89fe0e098fc989e3b064ec

closed 2020-01-09 by miri64