RIOT-OS/RIOT #12897: Can't build relic with benchmarks or tests

maksim-ka, 2019-12-06:
I was trying to build riot with the relic library with the BENCH and the TESTS flag set. 
The build failed with the following error when at least one of the flags(BENCH,TEST) is set to a value above zero.
 This is the error it produces

with BENCH flag set:
`                                                                                                                                            
 [ 88%] Linking C executable ../bin/bench_rand                                                                                                                                                                                       
 gcc: error: gcc: No such file or directory                                                                                                                                                                                          
 make[4]: *** [bench/CMakeFiles/bench_rand.dir/build.make:85: bin/bench_rand] Error 1                                                                                                                                                
 make[3]: *** [CMakeFiles/Makefile2:229: bench/CMakeFiles/bench_rand.dir/all] Error 2                                                                                                                                                
 make[2]: *** [Makefile:130: all] Error 2                                                                                                                                                                                            
 make[1]: *** [Makefile:16: all] Error 2                                                                                                                                                                                             
make: *** [/home/maksim/code/RIOT/examples/Relic_Hello-World/../../Makefile.include:614: pkg-build-relic] Error 2                                                                                                                   
[maksim@maksbook ~/code/RIOT/examples/Relic_Hello-World]$
`
with TESTS flag set:
`
 [ 85%] Building C object test/CMakeFiles/test_ec.dir/test_ec.c.obj                                                                                                                                                                  
 [ 86%] Linking C executable ../bin/test_ec                                                                                                                                                                                          
 gcc: error: gcc: No such file or directory                                                                                                                                                                                          
 make[4]: *** [test/CMakeFiles/test_ec.dir/build.make:85: bin/test_ec] Error 1                                                                                                                                                       
 make[3]: *** [CMakeFiles/Makefile2:233: test/CMakeFiles/test_ec.dir/all] Error 2                                                                                                                                                    
 make[2]: *** [Makefile:141: all] Error 2                                                                                                                                                                                            
 make[1]: *** [Makefile:16: all] Error 2                                                                                                                                                                                             
 make: *** [/home/maksim/code/RIOT/examples/Relic_Hello-World/../../Makefile.include:614: pkg-build-relic] Error 2                                                                                                                   │
 [maksim@maksbook ~/code/RIOT/examples/Relic_Hello-World]$
`

To build I copied the hello-world example and changed the Makefile
I just added those two lines: 

`
export RELIC_CONFIG_FLAGS=-DARCH=x86 -DQUIET=off -DWORD=32 -DFP_PRIME=255 -DWITH=ALL -DSEED=ZERO -DBENCH=** -DTESTS=**
`
`USEPKG += relic`

with `-DBENCH` and  `-DTESTS` > 0 

I'm very new to RIOT so please excuse naivety. By any means I'm ready to solve this problem on my own, but I don't know how to approach right now.

aabadie, 2020-01-03:
Sorry for the late reply.

I don't think using BENCH and TESTS > 0 is supported. In the current state they are force to 0.

https://github.com/RIOT-OS/RIOT/blob/e45dc52c7b3d97a111084a20467c2fa5915777a1/pkg/relic/Makefile#L26-L27

Why do need to change these values ?

maksim-ka, 2020-01-22:
> Why do need to change these values ?
I would like to make some benchmarks for performance tests

Is there any chance that it will be supported in near feature?
Would it be hard to implement support? I would try myself but I really don't know how to approach that.

PeterKietzmann, 2020-01-26:
For reference: [https://github.com/relic-toolkit/relic/issues/122](https://github.com/relic-toolkit/relic/issues/122)