RIOT-OS/RIOT #12909: Bug: Unexpected behavior with xtimer_usleep() or xtimer_periodic_wakeup() for large values on Nucleo-f401re

arthur-gauthier, 2019-12-09:
Hello,

#### Description
Cannot use xtimer_periodic_wakeup() with a period greater than 12us on Nucleo f401re.

#### Steps to reproduce the issue
I try the following code with RIOT 2019-10 (coming from a test project):
```
#define INTERVAL (1U * US_PER_SEC)
int main(void) {
    xtimer_ticks32_t last_wakeup = xtimer_now();
    while(1) {
        xtimer_periodic_wakeup(&last_wakeup, INTERVAL);  
        LED0_TOGGLE;
    }
}
```
If I do the same with a period lower or equal to 12us, led is blinking but if period is higher, then I have nothing.

#### Expected results
Having the possibility to use properly xtimer_periodic_wakeup() for any kind of period.

#### Actual results
Only works for short periods lower or equal to 12us.

#### Versions
RIOT-OS 2019-10.
GCC

Thanks in advance for your support.

MichelRottleuthner, 2020-01-27:
Hi @arthur-gauthier thank you for opening this issue.
I tried to reproduce it with your example but I can't see the problem you describe.
Can you actually reproduce the bug with **only** the code you posted above, or is there something else running on your application?
Based on what I can see from your example it is a problem that the pin for LED0 is never initialized.
So with just that code, the toggle should not show any visible effect on the LED - independent of whatever timing `xtimer_periodic_wakeup` may create.
To enable init for that pin either add `gpio_init(LED0_PIN, GPIO_OUT);` at the beginning of your main or compile the application with `CFLAGS="-DAUTO_INIT_LED0" BOARD=nucleo-f401re make all flash term`.

Can you also provide the content of your Makefile to make sure we are running the exact same application?

Also, how did you verify that it works with values lower or equal to 12 µs? Did you measure the pin with a scope? I'm asking because with 12 µs it is hard to see with our eyes if is toggling properly ;)

Note that with the RIOT version you are using there are also known problems in the xtimer implementation and it would be worth trying with the upcoming release 2020.01 (or the current development version).

MichelRottleuthner, 2020-02-25:
@arthur-gauthier did you try the above suggestions?

miri64, 2020-07-06:
ping @arthur-gauthier?

arthur-gauthier, 2020-07-06:
Hello,
Thanks for your feedback and sorry for my very late reply. 
In the meantime I had to switch on another topic.
After I reported the bug report, I had to find a fast workaround and the solution has been to come back to the previous release RIOT 2019-03.
I'm really sorry but I would suggest to put my ticket as invalid as you didn't reproduce it.
I believe I did something wrong if it works on your side.
When I will come back to RIOT topic in my daily tasks I will check it again.
I keep track of your valuable remarks and thank you for your time.
I close that ticket.

closed 2020-07-06 by arthur-gauthier