RIOT-OS/RIOT #12920: kconfiglib: only pull when needed

miri64, 2019-12-11:
#### Description
Currently I'm sitting on a train with a machine that did not build with newest master yet. The consequence: I'm waiting for at least 5 min for `kconfiglib` to be pulled due to the slow WiFi here, even though I don't want to configure anything with kconfig at the moment.

#### Steps to reproduce the issue
Make sure `kconfiglib` is not downloaded yet, start a build e.g. `make -C examples/hello-world`.

#### Expected results
Unless required, `kconfiglib` should not be pulled.

#### Actual results
`kconfiglib` is pulled in any case.

#### Versions
Current RIOT master (f44d9f8)

dylad, 2019-12-11:
Is this somehow related to #11491 ?

miri64, 2019-12-11:
No not really, that one is about fetching (updating) a packet when it is already there. My problem relates to a specific packet being downloaded (cloned) in the first place even if it is not needed.

leandrolanzieri, 2019-12-11:
> My problem relates to a specific packet being downloaded (cloned) in the first place even if it is not needed.

Kconfig runs on every build. Even if no modules get configured, it will read the root `Kconfig` file, it cannot know beforehand that it will produce no configuration values.

One way of solving this problem you just had could be to place the needed files in `dist/tools/kconfiglib` instead of pulling them from the repo.

miri64, 2019-12-12:
> One way of solving this problem you just had could be to place the needed files in `dist/tools/kconfiglib` instead of pulling them from the repo.

How about using a submodule instead?

miri64, 2019-12-16:
Or another alternative: this is (optional!) tooling, so we could also just only use it **if** it is available. Why is this bootstrapped on build anyway. We could just expect the user to install it, as we do with the toolchain, libc, and the build system as well.

leandrolanzieri, 2019-12-16:
> We could just expect the user to install it, as we do with the toolchain, libc, and the build system as well.

@miri64 This seems reasonable, given that the tool can be installed from [PyPI](https://pypi.org/project/kconfiglib/). We could check if it is present before including `makefiles/kconfig.mk`.

kaspar030, 2019-12-16:
> Kconfig runs on every build. Even if no modules get configured, it will read the root `Kconfig` file, it cannot know beforehand that it will produce no configuration values.

As in, use of Kconfig is not optional, as the build is broken if kconfiglib is not available?
Can't Kconfig be guarded, e.g., by (OR'ed)

1. user calls "make menuconfig"
2. there's a config file created by a previous run of "make menuconfig"
3. a Kconfig Kconfig file is present in APPDIR?

ulfalizer, 2019-12-17:
I'm a bit old-school, but I'd just copy the scripts into the project. It's so simple it's really hard for it to break. :)

PR #12977 (2020-01-14): makefiles/kconfig: Only run Kconfig if it could generate configurations
### Contribution description
This adds a check to decide if Kconfig should run on a build (proposed [here](https://github.com/RIOT-OS/RIOT/issues/12920#issuecomment-566142649)). It will run if any of the following conditions is true:
- A file with '.config' extension is present in the application folder
- A 'Kconfig' file is present in the application folder
- A previous configuration file is present (e.g. from a previous call to menuconfig)
- menuconfig is being called

This assumes that Kconfig will not generate any defa [... cut]
closed 2020-01-14 by fjmolinas