RIOT-OS/RIOT #12964: esp32: network devices pull in GNRC dependencies statically

miri64, 2019-12-16:
#### Description
While compile testing #12931 we noticed that the ESP32 won't build `tests/lwip` anymore. The reason: All the device drivers for the SoC network devices pull in GNRC as a static dependency.

#### Steps to reproduce the issue
Build `tests/lwip` for an ESP32 board, e.g.

```sh
BOARD=esp32-wroom-32 make -C tests/lwip
```

#### Expected results
`tests/lwip` should not pull in any GNRC dependencies.

#### Actual results
From the build output in current master:

```
```

#### Versions

miri64, 2019-12-16:
Ah proposed fix from my side would be to not make the GNRC dependencies static but to put it in their own sub-module, as we did e.g. with `xbee`, `cc110x`, etc as well.

gschorcht, 2019-12-16:
Yeah, this is already fixed with the code deduplication PR #12955. Since PR#12955 will take some time before it is merged, I will provide a fix for the current master.

gschorcht, 2019-12-16:
These are simple left overs from the time when I started with the complex building system and wanted to have a working network interface when I compile `tests/shell` with any network device.

PR #12967 (2020-01-31): cpu/esp*: remove dependencies on GNRC for ESP network device drivers
### Contribution description

This PR removes the dependencies of ESP network device drivers on GNRC.

### Testing procedure

The following compilations should still succeed:

- [ ] `esp_eth`: ```make BOARD=esp32-olimex-evb -C tests/shell```
- [ ] `esp_now`: ```make BOARD=esp32-wroom-32 -C tests/lwip```
- [ ] `esp_wifi`: ```USEMODULE=esp_wifi make BOARD=esp32-wroom-32 -C tests/lwip```

### Issues/PRs references

Fixes #12964
Prerequisite for PR #12931
closed 2020-01-31 by benpicco