RIOT-OS/RIOT #12976: Try dtls riot on ubuntu

jabberimaher, 2019-12-18:
Hi,
please i hope to implement dtls with riot os (ubuntu) and measure the energy consumption , the time execution by  node communicate with server.
thanks a lot

miri64, 2019-12-18:
Welcome @jabberimaher! We have two packages already that support DTLS: [`tinydtls`](https://doc.riot-os.org/group__pkg__tinydtls.html) and [`wolfssl`](https://doc.riot-os.org/group__pkg__wolfssl.html). RIOT comes with three examples that might help you get it running for your own application:

- [`examples/dtls-echo`](https://github.com/RIOT-OS/RIOT/tree/master/examples/dtls-echo): a basic `tinydtls` example
- [`examples/dtls-sock`](https://github.com/RIOT-OS/RIOT/tree/master/examples/dtls-sock): usage of the `sock_dtls` API with the `tinydtls` implementation
- [`examples/dtls-wolfssl`](https://github.com/RIOT-OS/RIOT/tree/master/examples/dtls-wolfssl): usage of DTLS with `wolfssl`.

jabberimaher, 2019-12-18:
thank you for your answer,
i tried dtls-sock, i follow steps of the link... but when i launch command "dtlsc <4A:E0:42:98:75:C5> "data" ...it display  error parsing destination adress..it seems that can't read correctly the destination adress... 
best regards

miri64, 2019-12-18:
> i tried dtls-sock, i follow steps of the link... but when i launch command "dtlsc <4A:E0:42:98:75:C5> "data" ...it display error parsing destination adress..it seems that can't read correctly the destination adress...

That looks more like a MAC address than an IP address. Also, the `<>` are just to denote a variable in doc. So have you tried something like `dtlsc fe80::... "data"`?

pokgak, 2019-12-18:
Using the command as stated @miri64 above will solve your problem. I also want to point to documentation of [DTLS support](http://api.riot-os.org/group__net__dtls.html) in RIOT.

miri64, 2019-12-18:
Sorry, hit the wrong button. @jabberimaher does this solve your problem?

jabberimaher, 2019-12-18:
Hi, thank you
with dtls-echo it display:
dtlsc fe80::80be:30ff:febb:8909 "DATA to send under encrypted channel!"
Client: got DTLS Data App -- DATA to send under encrypted channel! --

miri64, 2019-12-18:
So I guess it is solved?

jabberimaher, 2019-12-18:
Now if i want to display ergy onsumption , time execution and occupation memory of node when executing dtls...... so i want to compare between three authentication algorithms (DTLS, MDA,..) by measuring for each algorithm the metrics of performances (time , energy, memory, throughput) , it's possible with riot?...
i will be very grateful for your help

miri64, 2019-12-18:
This should be discussed in #12979 ;-). Closing this issue.

closed 2019-12-18 by miri64