RIOT-OS/RIOT #12979: possibility to display energy consumption, time execution and occuped memory

jabberimaher, 2019-12-18:


jabberimaher, 2019-12-18:
hi, please with riot can i display energy consumption, time execution and occuped memory of node when executing module like dtls
thanks a lot

jabberimaher, 2019-12-18:
Now if i want to display ergy onsumption , time execution and occupation memory of node when executing dtls...... so i want to compare between three authentication algorithms (DTLS, MDA,..) by measuring for each algorithm the metrics of performances (time , energy, memory, throughput) , it's possible with riot?...
i will be very grateful for your help

miri64, 2019-12-18:
The occupied memory can be measured by both looking at the `size` output at compile time and using the `gnrc_pktbuf_cmd` (when using GNRC as a network stack) after the experiment ends. Execution time you could measure by measuring the current time with `xtimer_now_usec()` before and after the execution of the function you want to measure and taking the difference.

jabberimaher, 2019-12-18:
what about energy and throughput? i ask because if it's necessary to by an ESP32 for example and connect it to my computer to carry out these measurements or it is sufficient to use riot os under ubuntu

miri64, 2019-12-18:
Throughput we usually measure either with print outputs at critical points (e.g. before UDP send and before UDP receive) or toggling GPIO pins instead. For energy consumptions I rather let e.g. @PeterKietzmann answer as he has more experience with it.

jabberimaher, 2019-12-18:
thank you so much

miri64, 2019-12-18:
> i ask because if it's necessary to by an ESP32 for example and connect it to my computer to carry out these measurements or it is sufficient to use riot os under ubuntu

For energy consumption measurements or throughput measurements I don't think using `native` (I think that's what you mean by "use riot os under ubuntu"), as in this case RIOT just runs as a process in Linux. I guess you *could* tally up the run time of that process for energy consumption, but I'm not sure how representative this would be for real hardware. Likewise with throughput, Linux' network interface abstraction TAP/TUN is used for communication, so there is virtual no loss, as you would have e.g. over a wireless link or even when communicating over a physical wire.

PeterKietzmann, 2019-12-19:
Measuring these metrics is not a feature of RIOT but rather a general topic and if you are about to write a paper about it, you should not underestimate the efforts needed to set up a proper measurement environment. Some hints anyway:
- Throughput is usually measured by stressing a transmission between networked devices and accumulating transmitted data during a defined interval.
- CPU processing times are usually "short" and I personally use to toggle I/Os before and after one action and measure the time with an oscilloscope.
- Energy is power over time. Power can be depicted (for instance) by applying constant supply voltage and measuring the current flow on dedicated measurement headers. This really depends on your platform. Note that measurement headers usually connect to the MCU itself which ignores other parts (eg sensors) on your development board. Measuring current can by done with almost all multi-meters in "current" mode or with a scope and a shunt resistor. 

I would consider this issue as "solved" withing RIOT. Can we close?

fjmolinas, 2019-12-19:
> Energy is power over time. Power can be depicted (for instance) by applying constant supply voltage and measuring the current flow on dedicated measurement headers. This really depends on your [...]

Just to complement @PeterKietzmann  answer, you could also look into devices like https://www.st.com/en/evaluation-tools/x-nucleo-lpm01a.html. Also on https://www.iot-lab.info/tutorials/monitoring-consumption-m3/ you can measure power consumption on `iotlab-m3` nodes, but only up to a certain precision, you can ask on their platform for more details. If you need high accuracy then consider using a oscilloscope to measure voltage drop across a shunt resistor (many oscilloscopes allow you to log the data measurements and you can the process that externally).

Cheers!

closed 2019-12-19 by jabberimaher