RIOT-OS/RIOT #13007: How to use subdirectories in project workspace

BabyYod4, 2019-12-27:
#### Description

Let say i have a workspace setup in VSCode like this:
```
├── lucidy
│   ├── drivers
│   │   └── rfm95w
│   │       ├── rfm95w.c
│   │       ├── rfm95w.h
│   │       ├── rfm95w_reg.h
│   │       └── rfm95w_settings.h
│   ├── test.c
│   └── test.h
├── main.c
├── Makefile
└── README.md
```

How do i setup my `Makefile` so i can use the code in the `main.c` like this?:
```C++
#include "lucidy/drivers/rfm95w/rfm95w.h"
#include "lucidy/test.h"

int main(void) {
...
};
``` 

I am building an "Second Layer" abstraction "Library" for RIOT OS as many of my clients like to work in this RTOS but they have some difficulties adjusting from a C++ background (so i am kind of hacking in OOP in C on top of RIOT OS). If it would be possible to use this current workspace structure this would be really great for them, as they can more easily port their existing drivers to RIOT OS. 

### Useful links

I have looked online at using make to recursively search for source and header files, but I have no idea where to use it. Do i have to make changes to `($RIOT_BASE)/Makefile.include` or another file. Or can i just do something in my own workspace Makefile? 

You can also contact me on: arsalan.anwari@lucidy.nl if you know a method to this. 
Thanks for the help anyway!

BabyYod4, 2019-12-27:
Found a method! Its a bit cumbersome but doable: I have added a `Makefile` which contains the line: `include $(RIOTBASE)/Makefile.base` to all the different subdirectories used in the project folder so my new workspace looks like this:

```
├── lucidy
│   ├── drivers
│   │   ├── Makefile
│   │   └── rfm95w
│   │       ├── Makefile
│   │       ├── rfm95w.c
│   │       ├── rfm95w.h
│   │       ├── rfm95w_reg.h
│   │       └── rfm95w_settings.h
│   ├── Makefile
│   ├── test.c
│   └── test.h
├── LucidyAPI.include
├── main.c
├── Makefile
└── README.md
``` 

And then in my project directory (folder with `main.c`) i changed:

**Makefile:**
```Makefile
# name of your application
APPLICATION = test

# If no BOARD is found in the environment, use this default:
BOARD ?= arduino-uno
PORT ?= /dev/ttyACM0
TERM_SPEED ?= 9600

TERMFLAGS= -s $(TERM_SPEED) -p $(PORT)

# This has to be the absolute path to the RIOT base directory:
RIOTBASE ?= $(CURDIR)/../RIOT

# User settings
USEMODULE += xtimer
# FEATURES_REQUIRED = periph_gpio

# Include LucidyAPI
include $(CURDIR)/LucidyAPI.include

# Comment this out to disable code in RIOT that does safety checking
# which is not needed in a production environment but helps in the
# development process:
DEVELHELP ?= 1

# Change this to 0 show compiler invocation lines by default:
QUIET ?= 1

include $(RIOTBASE)/Makefile.include
```

**LucidyAPI.include:**
```Makefile
USEMODULE += lucidy
EXTERNAL_MODULE_DIRS += $(CURDIR)/lucidy

USEMODULE += drivers
EXTERNAL_MODULE_DIRS += $(CURDIR)/lucidy/drivers

USEMODULE += rfm95w
EXTERNAL_MODULE_DIRS += $(CURDIR)/lucidy/drivers/rfm95w
```

## Using a program to test it

**lucidy/test.h:**
```C++
#ifndef TEST_H
#define TEST_H

int test(void); //returns 10;

#endif //TEST_
```

**lucidy/drivers/rfm95w/rfm95w.h:**
```C++
#ifndef TEST_H
#define TEST_H

int init(void); //returns 20;

#endif //TEST_
```

**main.c:**
```C++
#include <stdio.h>
#include "xtimer.h"
#include "lucidy/drivers/rfm95w/rfm95w.h"
#include "lucidy/test.h"

int main(void)
{

    xtimer_sleep(5);

    for(int i=0; i<30; i++){
        printf("Hello Test: %i - %i\r\n", init(), test());
        xtimer_sleep(1);
    }
    
    return 0;
}
```

**OUTPUT:**
```bash
main(): This is RIOT! (Version: 2020.01-devel-1481-gd9e9e4)
Hello Test: 20 - 10
Hello Test: 20 - 10
Hello Test: 20 - 10
Hello Test: 20 - 10
Hello Test: 20 - 10

```

BabyYod4, 2019-12-27:
Okay Great, i tested it and i can also use RiotOS completely within my subfolders without having to change the Makefiles. For example in: 

**lucidy/drivers/rfm95w/rfm95w.h:**
```C++
#include "periph/gpio.h"
#include "xtimer.h"

void init(void);
```

**lucidy/drivers/rfm95w/rfm95w.c:**
```C++
#include "rfm95w.h"

void init(void){
    gpio_set(GPIO_PIN(1,0));
    xtimer_usleep(500000);
    gpio_clear(GPIO_PIN(1,0));
    xtimer_usleep(500000);
}
```

fjmolinas, 2019-12-30:
Hi @BabyYod4, first of all welcome to RIOT! I'm sorry for the late answer, last week was not the most active period for us :)

It seems you already found something that is working for you. If you want to use or port external modules to RIOT the best way is to use `EXTERNAL_MODULE_DIRS` and defining an adding a couple of `Makefiles` to your external driver. There is a mention of this on http://riot-os.org/api/creating-modules.html and a a reference to [tests/external_module_dirs](https://github.com/RIOT-OS/RIOT/tree/master/tests/external_module_dirs) which shows you an examples on how to do this. But it seems to me based on the approach you took that you already found these references. If you still have any question let me know. If everything is OK maybe we can close this issue?

> Expect to see 100+ of propitiatory drivers used in Smart Farming to pop up soon

This seems like exiting news!

aabadie, 2020-01-03:
> If everything is OK maybe we can close this issue?

@BabyYod4, are you ok to close this issue ?

BabyYod4, 2020-01-08:
Ahh yes sorry forgot to close

closed 2020-01-08 by BabyYod4

RossComputerGuy, 2020-05-23:
This feels clumbersome, is there a better solution? I don't want a bunch of modules just for subdirectories.