RIOT-OS/RIOT #13008: How to use subdirectory in workspace

BabyYod4, 2019-12-27:
#### Description

Let say i have a workspace setup in VSCode like this:
```
├── lucidy
│   ├── drivers
│   │   └── rfm95w
│   │       ├── rfm95w.c
│   │       ├── rfm95w.h
│   │       ├── rfm95w_reg.h
│   │       └── rfm95w_settings.h
│   ├── test.c
│   └── test.h
├── main.c
├── Makefile
└── README.md
```

How do i setup my `Makefile` so i can use the code in the `main.c` like this?:
```C++
#include "lucidy/drivers/rfm95w/rfm95w.h"
#include "lucidy/test.h"

int main(void) {
...
};
``` 

I am building an "Second Layer" abstraction "Library" for RIOT OS as many of my clients like to work in this RTOS but they have some difficulties adjusting from a C++ background (so i am kind of hacking in OOP in C on top of RIOT OS). If it would be possible to use this current workspace structure this would be really great for them, as they can more easily port their existing drivers to RIOT OS. 

### Useful links

I have looked online at using make to recursively search for source and header files, but I have no idea where to use it. Do i have to make changes to `($RIOT_BASE)/Makefile.include` or another file. Or can i just do something in my own workspace Makefile? 

You can also contact me on: arsalan.anwari@lucidy.nl if you know a method to this. 
Thanks for the help anyway!

BabyYod4, 2019-12-27:
Solution found! See [here](https://github.com/RIOT-OS/RIOT/issues/13007)

aabadie, 2020-01-03:
Same issue as #13007, so closing.

closed 2020-01-03 by aabadie